espressif/esp-idf #13208: invalid header: 0x00000000

heyangxinmei, 2024-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.5

### Espressif SoC revision.

ESP32U4WD

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32U4WD

### Power Supply used.

External 3.3V

### What is the expected behavior?

正常运行的设备，断电重启后出现上面的现象。

### What is the actual behavior?

正常运行的设备，断电重启后出现上面的现象。

### Steps to reproduce.

正常运行的设备，断电重启后出现上面的现象。

### Debug Logs.

```plain
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
invalid header: 0x00000000
ets Jul 29 2019 12:21:46
```

### More Information.

正常运行的设备，断电重启后出现上面的现象。

leeebo, 2024-02-19:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
invalid header: 0x00000000
```

@heyangxinmei It seems like your flash be erased or be broken. please try to flash a `hello_world` example to investigate further.

heyangxinmei, 2024-02-19:
@leeebo re-burned the program and the device worked fine.

I (28) boot: ESP-IDF v4.4.5-dirty 2nd stage bootloader
I (28) boot: compile time 15:32:43
I (28) boot: chip revision: v3.1
I (32) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (91) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (99) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (106) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (114) boot: End of partition table
I (118) boot: No factory image, trying OTA 0
I (123) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e6dch (124636) map
I (177) esp_image: segment 1: paddr=0002e704 vaddr=3ffbdb60 size=01914h (  6420) load
I (180) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7f0b0h (520368) map
I (371) esp_image: segment 3: paddr=000af0d8 vaddr=3ffbf474 size=01e34h (  7732) load
I (374) esp_image: segment 4: paddr=000b0f14 vaddr=40080000 size=1851ch ( 99612) load
I (429) boot: Loaded app from partition at offset 0x10000
I (481) boot: Set actual ota_seq=1 in otadata[0]
I (481) boot: Disabling RNG early entropy source...
I (491) cpu_start: Pro cpu up.
I (491) cpu_start: Single core mode
I (502) cpu_start: Pro cpu start user code
I (502) cpu_start: cpu freq: 160000000
I (502) cpu_start: Application information:
I (507) cpu_start: Project name:     capture_three_slave
I (513) cpu_start: App version:      v4.4.5-dirty
I (518) cpu_start: Compile time:     Jan 26 2024 15:32:10
I (524) cpu_start: ELF file SHA256:  4adaebbca9aa5bd9...
I (530) cpu_start: ESP-IDF:          v4.4.5-dirty
I (535) cpu_start: Min chip rev:     v0.0
I (540) cpu_start: Max chip rev:     v3.99
I (545) cpu_start: Chip rev:         v3.1
I (550) heap_init: Initializing. RAM available for dynamic allocation:
I (557) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (563) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (569) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (575) heap_init: At 3FFC6548 len 00019AB8 (102 KiB): DRAM
I (581) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (588) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (594) heap_init: At 4009851C len 00007AE4 (30 KiB): IRAM
I (600) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (608) spi_flash: detected chip: generic
I (611) spi_flash: flash io: dio
I (616) cpu_start: Starting scheduler on PRO CPU.
I (641) SMART_PAY_NVS: Open namespace done, name "smart_pay"
I (641) system_api: Base MAC address is not set
I (641) system_api: read default base MAC address from EFUSE
I (651) SMART_PAY_NVS: Restore, key "config_data" not exists
I (651) SMART_PAY_NVS: Restore, key "new_data" not exists

heyangxinmei, 2024-02-19:
@leeebo I re-burned the program and the device worked fine.

I (28) boot: ESP-IDF v4.4.5-dirty 2nd stage bootloader
I (28) boot: compile time 15:32:43
I (28) boot: chip revision: v3.1
I (32) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (91) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (99) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (106) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (114) boot: End of partition table
I (118) boot: No factory image, trying OTA 0
I (123) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e6dch (124636) map
I (177) esp_image: segment 1: paddr=0002e704 vaddr=3ffbdb60 size=01914h (  6420) load
I (180) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7f0b0h (520368) map
I (371) esp_image: segment 3: paddr=000af0d8 vaddr=3ffbf474 size=01e34h (  7732) load
I (374) esp_image: segment 4: paddr=000b0f14 vaddr=40080000 size=1851ch ( 99612) load
I (429) boot: Loaded app from partition at offset 0x10000
I (481) boot: Set actual ota_seq=1 in otadata[0]
I (481) boot: Disabling RNG early entropy source...
I (491) cpu_start: Pro cpu up.
I (491) cpu_start: Single core mode
I (502) cpu_start: Pro cpu start user code
I (502) cpu_start: cpu freq: 160000000
I (502) cpu_start: Application information:
I (507) cpu_start: Project name:     capture_three_slave
I (513) cpu_start: App version:      v4.4.5-dirty
I (518) cpu_start: Compile time:     Jan 26 2024 15:32:10
I (524) cpu_start: ELF file SHA256:  4adaebbca9aa5bd9...
I (530) cpu_start: ESP-IDF:          v4.4.5-dirty
I (535) cpu_start: Min chip rev:     v0.0
I (540) cpu_start: Max chip rev:     v3.99
I (545) cpu_start: Chip rev:         v3.1
I (550) heap_init: Initializing. RAM available for dynamic allocation:
I (557) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (563) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (569) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (575) heap_init: At 3FFC6548 len 00019AB8 (102 KiB): DRAM
I (581) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (588) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (594) heap_init: At 4009851C len 00007AE4 (30 KiB): IRAM
I (600) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (608) spi_flash: detected chip: generic
I (611) spi_flash: flash io: dio
I (616) cpu_start: Starting scheduler on PRO CPU.
I (641) SMART_PAY_NVS: Open namespace done, name "smart_pay"
I (641) system_api: Base MAC address is not set
I (641) system_api: read default base MAC address from EFUSE
I (651) SMART_PAY_NVS: Restore, key "config_data" not exists
I (651) SMART_PAY_NVS: Restore, key "new_data" not exists