espressif/esp-idf #13209: nan_console doesn't work with Android

jameskeane, 2024-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

ESP-WROOM-32

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-2432S028R

### Power Supply used.

USB

### What is the expected behavior?

I expected to be able to see a publish or subscribe to a service between esp32 and android using the [WifiNANScan](https://play.google.com/store/apps/details?id=com.google.android.apps.location.rtt.wifinanscan&hl=en_CA&gl=US) test app developed by Google.

### What is the actual behavior?

An error when attempting to subscribe from the esp32.
```
nan> subscribe -n TEST
I (13738) nan_app: Started Subscribing to TEST [Service ID - 5]
nan> E (31008) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (31008) task_wdt:  - IDLE0 (CPU 0)
E (31008) task_wdt: Tasks currently running:
E (31008) task_wdt: CPU 0: wifi
E (31008) task_wdt: CPU 1: IDLE1
E (31008) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400E000B:0x3FFB1E40 0x400E03D0:0x3FFB1E60 0x40083831:0x3FFB1E90 0x40127BE5:0x3FFC00D0 0x4012A521:0x3FFC00F0 0x4012A82F:0x3FFC0110 0x40125DE9:0x3FFC0140 0x40125E71:0x3FFC0180 0x40096965:0x3FFC01A0 0x40094528:0x3FFC01C0 0x4008BD59:0x3FFC01F0
```

And nothing visible when attempting to publish on the esp.

### Steps to reproduce.

1. Create a service in the Android app called 'TEST' and publish it.
1. Build the nan_console example and flash it.
2. On the monitor:
3. `nan -S`
4. `subscribe -n TEST`

### Debug Logs.

```plain
nan> subscribe -n TEST
I (13738) nan_app: Started Subscribing to TEST [Service ID - 5]
nan> E (31008) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (31008) task_wdt:  - IDLE0 (CPU 0)
E (31008) task_wdt: Tasks currently running:
E (31008) task_wdt: CPU 0: wifi
E (31008) task_wdt: CPU 1: IDLE1
E (31008) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400E000B:0x3FFB1E40 0x400E03D0:0x3FFB1E60 0x40083831:0x3FFB1E90 0x40127BE5:0x3FFC00D0 0x4012A521:0x3FFC00F0 0x4012A82F:0x3FFC0110 0x40125DE9:0x3FFC0140 0x40125E71:0x3FFC0180 0x40096965:0x3FFC01A0 0x40094528:0x3FFC01C0 0x4008BD59:0x3FFC01F0
```

ShyamalKhachane, 2024-02-26:
Hi @jameskeane, please provide the corresponding elf file and sniffer capture.
The elf file can be located in the build directory of nan_console example.

Alvin1Zhang, 2025-03-07:
Thanks for reporting, will close due to short of details, feel free to reopen with more updates.

closed 2025-03-07 by Alvin1Zhang