espressif/esp-idf #13210: can not connect to ap  after unsuccessful auth/assoc

hibfh, 2024-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I try to set my esp32 as ap and http server to test ota func . And when i add http server code ,my pc cannot connect to it while my android phone can connect to it.
Here the http server code:
```    
        static httpd_handle_t http_server = NULL;
	httpd_config_t config = HTTPD_DEFAULT_CONFIG();

	if (httpd_start(&http_server, &config) == ESP_OK) {
		httpd_register_uri_handler(http_server, &index_get);
		httpd_register_uri_handler(http_server, &update_post);
	}
```
[https://github.com/Jeija/esp32-softap-ota](url) I refer to this project.

When I try to connect the ap with my PC ,it failed and log like this:
```
I (17334) wifi:removing station <d4:54:8b:a0:10:4a> after unsuccessful auth/assoc, AID = 0
W (17344) wifi:rm mis
I (17354) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
```

My work PC says “Some settings are managed by your organization” in wifi setting， I‘m not sure is it cause the problem.

AxelLin, 2024-02-26:
I hit similar issue on esp32s3, using v5.1.2-695-gd922d4178f.
```
I (70671) wifi:removing station <20:1e:88:5e:2b:3f> after unsuccessful auth/assoc, AID = 0
W (70672) wifi:rm mis
I (70672) wifi:new:<6,0>, old:<6,0>, ap:<6,0>, sta:<0,0>, prof:6
I (92676) wifi:removing station <20:1e:88:5e:2b:3f> after unsuccessful auth/assoc, AID = 0
W (92677) wifi:rm mis
I (92677) wifi:new:<6,0>, old:<6,0>, ap:<6,0>, sta:<0,0>, prof:6
```

hibfh, 2024-02-28:
fine, but I wonder what may cause this problem ? Is there any suggestion to figure it out?

AxelLin, 2024-02-29:
@hmalpani 

Maybe the log is a hint.
What can trigger the log as https://github.com/espressif/esp-idf/issues/13210#issuecomment-1963570530 ?

ShyamalKhachane, 2024-03-04:
Hi @hibfh, @AxelLin,
Please share the debug logs and sniffer capture corresponding to the issue.
(To enable debug logs, run "idf.py menuconfig" in example directory, then navigate to Component config -> Log output -> select default log verbosity to "Debug" and build the project)

nishanth-radja, 2024-03-04:
@hibfh in what security is the AP configured in ? If it is open security, Is your work PC allowed to connect to the open security ?
if it is wpa2 securtiy, can you pls share the ssid and password used, so that we can decode the packets.

hibfh, 2024-03-05:
@ShyamalKhachane @nishanth-radja hi, I don't know why, but my project work well recently, what it only changes is that I add some SPI canfd code. There is no reason the related codes can influence the wifi part. Anyway， The AP work well at last. But when I OTA with the old bin file I saved before. The AP can not to be connected again :P

closed 2024-03-05 by hibfh

gongzi7, 2025-03-01:
Perhaps your host does not support the security protocol set by ESP. The WPA WPA2 WPA3 . You can find this option  in  wifi_config > authmode

lao-zhang-0619, 2025-03-03:
I have this problem too; how can I solve it?

CYK-dot, 2025-07-26:
I'm also encountering the wifi:rm mis issue. I suspect this could be caused by a potential hardware problem with ESP32 chip. Here's why: I have two identical ESP32 modules on hand, both flashed with the same code. The first module shows this rm mis issue intermittently, while the second one does not.