espressif/esp-idf #13212: [ESP32S3] v4.4 WIFI losing connectivity temporarily or permanently without apparent reason

KonssnoK, 2024-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

@zhangyanjiaoesp 

based on 27ec26d2d3f44bbde5da14c7fdfc82226d567874
To reproduce
- Add ping functionality and periodc wifi dumps to main task instead of letting it die:

```
    esp_ping_config_t ping_config = ESP_PING_DEFAULT_CONFIG();
    IP4_ADDR(&ping_config.target_addr, 8,8,8,8);          // target IP address
    ping_config.count = ESP_PING_COUNT_INFINITE;    // ping in infinite mode, esp_ping_stop can stop it
    ping_config.timeout_ms = 3000;

    /* set callback functions */
    esp_ping_callbacks_t cbs;
    cbs.on_ping_success = test_on_ping_success;
    cbs.on_ping_timeout = test_on_ping_timeout;
    cbs.on_ping_end = test_on_ping_end;
    cbs.cb_args = "foo";  // arguments that will feed to all callback functions, can be NULL

    esp_ping_handle_t ping;
    esp_ping_new_session(&ping_config, &cbs, &ping);
    esp_ping_start(ping);

#define DUMP_INTERVAL_MS (1000*60*2)
    int64_t last_dump = 0;
    while (true) {
        if((os_time_get_timestamp() - last_dump) >= DUMP_INTERVAL_MS) {
            last_dump = os_time_get_timestamp();
            esp_wifi_statis_dump(0xFFFF);
        }
        os_thread_sleep(1000);
    }

```
```

static void test_on_ping_success(esp_ping_handle_t hdl, void *args)
{
    // optionally, get callback arguments
    // const char* str = (const char*) args;
    // printf("%s\r\n", str); // "foo"
    uint8_t ttl;
    uint16_t seqno;
    uint32_t elapsed_time, recv_len;
    ip_addr_t target_addr;
    esp_ping_get_profile(hdl, ESP_PING_PROF_SEQNO, &seqno, sizeof(seqno));
    esp_ping_get_profile(hdl, ESP_PING_PROF_TTL, &ttl, sizeof(ttl));
    esp_ping_get_profile(hdl, ESP_PING_PROF_IPADDR, &target_addr, sizeof(target_addr));
    esp_ping_get_profile(hdl, ESP_PING_PROF_SIZE, &recv_len, sizeof(recv_len));
    esp_ping_get_profile(hdl, ESP_PING_PROF_TIMEGAP, &elapsed_time, sizeof(elapsed_time));
    ESP_LOGW(TAG, "%d bytes from %s icmp_seq=%d ttl=%d time=%d ms",
           recv_len, ipaddr_ntoa(&target_addr), seqno, ttl, elapsed_time);
}

static void test_on_ping_timeout(esp_ping_handle_t hdl, void *args)
{
    uint16_t seqno;
    ip_addr_t target_addr;
    esp_ping_get_profile(hdl, ESP_PING_PROF_SEQNO, &seqno, sizeof(seqno));
    esp_ping_get_profile(hdl, ESP_PING_PROF_IPADDR, &target_addr, sizeof(target_addr));
    ESP_LOGW(TAG, "From %s icmp_seq=%d timeout", ipaddr_ntoa(&target_addr), seqno);
}

static void test_on_ping_end(esp_ping_handle_t hdl, void *args)
{
    uint32_t transmitted;
    uint32_t received;
    uint32_t total_time_ms;

    esp_ping_get_profile(hdl, ESP_PING_PROF_REQUEST, &transmitted, sizeof(transmitted));
    esp_ping_get_profile(hdl, ESP_PING_PROF_REPLY, &received, sizeof(received));
    esp_ping_get_profile(hdl, ESP_PING_PROF_DURATION, &total_time_ms, sizeof(total_time_ms));
    ESP_LOGW(TAG, "%d packets transmitted, %d received, time %dms", transmitted, received, total_time_ms);
}
```
- Connect to mobile hostspot wifi
- Put phone simcard to 2G.
- Wait

- The issue happens in the ROOT. Once the root is affected, the issue is propagated to all children.
- The issue is either temporary or permanent. When permanent, the only way to recover is to reboot.

[espressif_wifi_dump_3.txt](https://github.com/espressif/esp-idf/files/14328900/espressif_wifi_dump_3.txt)
[espressif_wifi_dump.txt](https://github.com/espressif/esp-idf/files/14328901/espressif_wifi_dump.txt)
[espressif_wifi_dump_2.txt](https://github.com/espressif/esp-idf/files/14328902/espressif_wifi_dump_2.txt)

Other related issues with similar behavior:
https://github.com/espressif/esp-idf/issues/8953
https://github.com/espressif/esp-idf/issues/10506

AxelLin, 2024-06-07:
@Espressif-liuuuu @zhangyanjiaoesp @nishanth-radja  Do you have any finding in above log?

zhangyanjiaoesp, 2024-06-07:
@KonssnoK Are you using the `ip_internal_network` example? In your logs, it show the ping timeout, but it can't confirm the wifi connection is down. Can you capture packets for this ?

KonssnoK, 2024-06-07:
> @KonssnoK Are you using the `ip_internal_network` example? In your logs, it show the ping timeout, but it can't confirm the wifi connection is down. Can you capture packets for this ?

@zhangyanjiaoesp yes the base is the ip_internal_network example.
how would you check if the wifi connection is down? apart from seeing no packets are sent/receive.

Also, how would you get the packets ? Wireshark connected to a sniffer?

KonssnoK, 2024-06-10:
So @zhangyanjiaoesp   i was able to generate one strange behavior, even if it's not exacly the one reported in this issue.

With the same code  (v4.4 top of c0e0af03d153d2c157d1d420831ab33d48888768 )

you can apply patches 1 2 3, which enable monitoring and pinging

[03_ip_internal.patch](https://github.com/user-attachments/files/15759038/03_ip_internal.patch)
[02_if_dumps.patch](https://github.com/user-attachments/files/15759039/02_if_dumps.patch)
[01_packets_dump.patch](https://github.com/user-attachments/files/15759040/01_packets_dump.patch)

by randomly detaching/attaching the layer 2 device i was able to reach this state, where the L2 device is never able to communicate with L1
[timeout_l2.txt](https://github.com/user-attachments/files/15759054/timeout_l2.txt)

I got an extract of L1 too (i would say MESH_EVENT_CHILD_CONNECTED )

[ESP-IDF_test.txt](https://github.com/user-attachments/files/15759063/ESP-IDF_test.txt)

KonssnoK, 2024-06-10:
interestingly enough to recover the L2 device i had to reboot both devices

KonssnoK, 2024-06-11:
@zhangyanjiaoesp 
again by simply resetting the 2 devices in different ways, i was able to trigger another case in which one device does not work anymore until reboot.
to be noted: once this device is failing, rebooting the other device makes it fail too.

[timeout_reset1.txt](https://github.com/user-attachments/files/15786044/timeout_reset1.txt)
[timeout_reset2.txt](https://github.com/user-attachments/files/15786046/timeout_reset2.txt)

zhangyanjiaoesp, 2024-06-13:
@KonssnoK please provide your sdkconfig file, and you are using PSRAM, right?

KonssnoK, 2024-06-13:
[sdkconfig.txt](https://github.com/user-attachments/files/15816181/sdkconfig.txt)
@zhangyanjiaoesp here it is

KonssnoK, 2024-06-13:
@zhangyanjiaoesp i reproduced with a Google Pixel8.

not getting the IP - strange, would mean the IP service is stuck 🤔

KonssnoK, 2024-06-18:
@zhangyanjiaoesp i moved to 3 devices and trying to replicate but for now without success..

KonssnoK, 2024-06-18:
and as soon as i wrote that, something strange happened again:
dev3 is not able to connect

[dev2.txt](https://github.com/user-attachments/files/15883196/dev2.txt)
[disconnected_dev3.txt](https://github.com/user-attachments/files/15883197/disconnected_dev3.txt)
[dev1.txt](https://github.com/user-attachments/files/15883200/dev1.txt)

KonssnoK, 2024-06-18:
one hour in:
`W (6444006) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:5334`

KonssnoK, 2024-06-18:
`W (13189216) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:10955`

KonssnoK, 2024-06-18:
this is instead the log of device1 getting stuck and not trying to connect to the mesh anymore

[dev1_stuck.txt](https://github.com/user-attachments/files/15886993/dev1_stuck.txt)

zhangyanjiaoesp, 2024-06-19:
> and as soon as i wrote that, something strange happened again: dev3 is not able to connect
> 
> [dev2.txt](https://github.com/user-attachments/files/15883196/dev2.txt) 
> [disconnected_dev3.txt](https://github.com/user-attachments/files/15883197/disconnected_dev3.txt)
>  [dev1.txt](https://github.com/user-attachments/files/15883200/dev1.txt)
> 
> ```
> I (42466) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
> W (43196) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:0
> W (44396) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:1
> W (45596) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:2
> W (46796) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:3
> W (47996) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:4
> W (49196) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:5
> ```
> 
> (devices are 30cm apart fom each other)
> 
> it seems it goes on forever `W (1983596) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:1617`

@KonssnoK In the log, I see that initially communication among the three devices was normal, and then you restarted device2 and device3？ 
And in the logs of device2 and device3, there are logs showing `I (42466) wifi:state: run -> init (2c0)`, this means the wifi connection is disconnected. The wifi disconnection will cause the `W (6444006) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:43, timeout_count:5334`. But I can't find why the wifi disconnect.  Can you use wireshark to capture packets of the device1/2/3, and send the logs and captures to me? In the log, please display the absolute time.

And I have using the Google Pixel5 mobile to test, but I didn't reproduce the problem.

KonssnoK, 2024-06-19:
> @KonssnoK In the log, I see that initially communication among the three devices was normal, and then you restarted device2 and device3？ And in the logs of device2 and device3, there are logs [...]

@zhangyanjiaoesp 
today i have no way to use wireshark.
the only way to recover that device was to reboot it again.
Yes, to trigger the issue i simply randomly restarted devices.
i'm not sure on how to display absolute time considering the logs are in close sourced files.
I understand the wifi is disconnected, but i would expect it to retry a connection once disconnected 🤔

zhangyanjiaoesp, 2024-06-19:
> Yes, to trigger the issue i simply randomly restarted devices

Ok, I will try to restart the device and test again.

KonssnoK, 2024-06-19:
@zhangyanjiaoesp in my experience slow data rates help achieving the issues.
Please put your phone cell technology to 2G or go to "developer options" networking "network download rate limit" and put the minimum.

KonssnoK, 2024-06-19:
[240619dev3.txt](https://github.com/user-attachments/files/15897804/240619dev3.txt)
[240619dev1.txt](https://github.com/user-attachments/files/15897805/240619dev1.txt)
[240619dev2.txt](https://github.com/user-attachments/files/15897806/240619dev2.txt)

@zhangyanjiaoesp i rebooed the root device and it went offline.

After a while device 3 managed to change status and directly connect as the root.
the other 2 remain disconnected

[240619dev3_2.txt](https://github.com/user-attachments/files/15897829/240619dev3_2.txt)
[240619dev1_2.txt](https://github.com/user-attachments/files/15897830/240619dev1_2.txt)
[240619dev2_2.txt](https://github.com/user-attachments/files/15897831/240619dev2_2.txt)

KonssnoK, 2024-06-20:
@zhangyanjiaoesp i tried also today to replicate, to verify if this is consistent:
- phone on rate limiter 128kbps
- keep devices off until dev1 root
- dev 2 3 as children
- reboot dev 1
- dev 1 blocked , dev 2 3 blocked 

it's quite easy to create issues in this configuration, please let me know if you manage.

[240620dev3.txt](https://github.com/user-attachments/files/15914480/240620dev3.txt)
[240620dev1.txt](https://github.com/user-attachments/files/15914482/240620dev1.txt)
[240620dev2.txt](https://github.com/user-attachments/files/15914483/240620dev2.txt)

zhangyanjiaoesp, 2024-06-21:
@KonssnoK I'm sorry, I have an urgent task recently. I will test your issue next week.

KonssnoK, 2024-06-21:
@zhangyanjiaoesp sure, i'll concentrate on another issue meanwhile

zhangyanjiaoesp, 2024-06-25:
> and as soon as i wrote that, something strange happened again: dev3 is not able to connect
>
> [dev2.txt](https://github.com/user-attachments/files/15883196/dev2.txt) [...]

@KonssnoK I have reproduced this issue by rebooting the root device, and I have found the root cause, the following wifi libs can solve the problem. Please replace the wifi libs and test again.
[wifi_lib_s3_0625.zip](https://github.com/user-attachments/files/15967100/wifi_lib_s3_0625.zip) (wifi firmware version: f736b07)

For the other issues, I still can't reproduce them although I randomly reboot the device2/3.

AxelLin, 2024-06-25:
> @KonssnoK I have reproduced this issue by rebooting the root device, and I have found the root cause

What's the root cause?

KonssnoK, 2024-06-25:
> @KonssnoK I have reproduced this issue by rebooting the root device, and I have found the root cause, the following wifi libs can solve the problem. Please replace the wifi libs and test again. [...]

@zhangyanjiaoesp sure I will try them.
Should this unblock devices from avoid reconnection?
Because we are currently seeing this kind of issues a bit everywhere on the field.
Thanks

KonssnoK, 2024-06-25:
@zhangyanjiaoesp i changed the library but i see no difference in the behavior of the children 🤔 
[240625dev1_1.txt](https://github.com/user-attachments/files/15967890/240625dev1_1.txt)
[240625dev3_1.txt](https://github.com/user-attachments/files/15967892/240625dev3_1.txt)
[240625dev2_1.txt](https://github.com/user-attachments/files/15967893/240625dev2_1.txt)

zhangyanjiaoesp, 2024-06-25:
> @zhangyanjiaoesp i changed the library but i see no difference in the behavior of the children 🤔 [240625dev1_1.txt](https://github.com/user-attachments/files/15967890/240625dev1_1.txt) [240625dev3_1.txt](https://github.com/user-attachments/files/15967892/240625dev3_1.txt) [240625dev2_1.txt](https://github.com/user-attachments/files/15967893/240625dev2_1.txt)
> 
> EDIT: i think this might have been related to the fact that the 4th device, which was acting as root, was not updated yet with latest libraries. I will now retry in the nominal configuration

It's wired. Here is device2 log on my side when the root rebooting. 
[device2.txt](https://github.com/user-attachments/files/15968213/device2.txt)

zhangyanjiaoesp, 2024-06-25:
> So @zhangyanjiaoesp i was able to generate one strange behavior, even if it's not exacly the one reported in this issue.
> 
> With the same code (v4.4 top of [c0e0af0](https://github.com/espressif/esp-idf/commit/c0e0af03d153d2c157d1d420831ab33d48888768) )
> 
> you can apply patches 1 2 3, which enable monitoring and pinging
> 
> [03_ip_internal.patch](https://github.com/user-attachments/files/15759038/03_ip_internal.patch) [02_if_dumps.patch](https://github.com/user-attachments/files/15759039/02_if_dumps.patch) [01_packets_dump.patch](https://github.com/user-attachments/files/15759040/01_packets_dump.patch)

@KonssnoK My test is based on the `ip_internal_network` example and added the above patch. And I noticed this print `W (106224) mesh_hand: Triggering DYNAMIC MESH handover` in your logs. You used a different test code, right?

KonssnoK, 2024-06-25:
> @KonssnoK My test is based on the `ip_internal_network` example and added the above patch. And I noticed this print `W (106224) mesh_hand: Triggering DYNAMIC MESH handover` in your logs. You used a [...]

yes sorry it's an evolution of your code.
I will put back the old one and reproduce again

KonssnoK, 2024-06-25:
dev1 is now blocked in a 
`I (135890) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2` loop

[240625dev2_3.txt](https://github.com/user-attachments/files/15969572/240625dev2_3.txt)
[240625dev1_3.txt](https://github.com/user-attachments/files/15969573/240625dev1_3.txt)
[240625dev3_3.txt](https://github.com/user-attachments/files/15969575/240625dev3_3.txt)

KonssnoK, 2024-06-25:
with the same procedure another issue appeared:
- generation of 2 networks that do not merge back together
- dev1 and dev3 are both ROOT and do not try to merge together

[240625dev1_5.txt](https://github.com/user-attachments/files/15969871/240625dev1_5.txt)
[240625dev2_5.txt](https://github.com/user-attachments/files/15969872/240625dev2_5.txt)
[240625dev3_5.txt](https://github.com/user-attachments/files/15969870/240625dev3_5.txt)

KonssnoK, 2024-06-25:
another case of dev1 getting stuck after reset.
the reset was hold for some seconds before releasing (i don't always just press/release)

[240625dev2_6.txt](https://github.com/user-attachments/files/15969971/240625dev2_6.txt)
[240625dev1_6.txt](https://github.com/user-attachments/files/15969973/240625dev1_6.txt)
[240625dev3_6.txt](https://github.com/user-attachments/files/15969975/240625dev3_6.txt)

KonssnoK, 2024-06-25:
@zhangyanjiaoesp 
last example of this issue:

[240625dev3_7.txt](https://github.com/user-attachments/files/15972456/240625dev3_7.txt)
[240625dev1_7.txt](https://github.com/user-attachments/files/15972457/240625dev1_7.txt)
[240625dev2_7.txt](https://github.com/user-attachments/files/15972458/240625dev2_7.txt)

now i'll go back to the other one while you fix this

zhangyanjiaoesp, 2024-06-26:
> dev1 is now blocked in a `I (135890) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2` loop
>
> [240625dev2_3.txt](https://github.com/user-attachments/files/15969572/240625dev2_3.txt) [...]

When the root device is connecting to the router, and the disconnect reason is 2 (auth expire), then the root will continue to reconnect to the router. So you see the `I (135890) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2` log loop. User should handle this case in the application layer. 
By the way, `I (50960) wifi:state: auth -> init (200)` this log indicates the device sends auth request, but the router doesn't reply auth response. It's strange why the router doesn't reply auth response. Could it have something to do with the hotspot used? Do you have this problem if you use another hotspot or router？

KonssnoK, 2024-06-26:
> When the root device is connecting to the router, and the disconnect reason is 2 (auth expire), then the root will continue to reconnect to the router. So you see the `I (135890) mesh_main: [...]

how should the user handle this error?
and why is it reported continuously alternated to the error 205?

zhangyanjiaoesp, 2024-06-26:
> with the same procedure another issue appeared:
>
> * generation of 2 networks that do not merge back together
> * dev1 and dev3 are both ROOT and do not try to merge together [...]

By default, it allows more than one root existing in one mesh network. Please call `esp_mesh_allow_root_conflicts(false)` to disable it.

zhangyanjiaoesp, 2024-06-26:
> how should the user handle this error?

For example, if the auth fails because the router moved its position, user can call `esp_mesh_waive_root()` to change a better root. 
> and why is it reported continuously alternated to the error 205?

When connects fail,  sta will add this AP to a blacklist, and reason code 205 indicates scan fail due to the AP is in blacklist, after this the AP will be removed from the blacklist. So you will see the reason code is 2/205 loop.

KonssnoK, 2024-06-26:
> By default, it allows more than one root existing in one mesh network. Please call `esp_mesh_allow_root_conflicts(false)` to disable it.

@zhangyanjiaoesp should this be called on all devices or only on root? when should it be called ? before mesh_start?

zhangyanjiaoesp, 2024-06-26:
> @zhangyanjiaoesp should this be called on all devices or only on root? when should it be called ? before mesh_start?

Call it before mesh start on all devices.

zhangyanjiaoesp, 2024-06-26:
> Is it the STA blacklisting the AP or or the AP blacklisting the STA?

The STA add the AP to STA's blacklist.

> something like this for the waiving?
> 
> ```
>     case MESH_EVENT_PARENT_DISCONNECTED: {
>         mesh_event_disconnected_t *disconnected = (mesh_event_disconnected_t *)event_data;
>         ESP_LOGI(MESH_TAG,
>                  "<MESH_EVENT_PARENT_DISCONNECTED>reason:%d",
>                  disconnected->reason);
>         mesh_layer = esp_mesh_get_layer();
>         mesh_netifs_stop();
> 
>         if (esp_mesh_is_root() && disconnected->reason == WIFI_REASON_CONNECTION_FAIL){
>             esp_mesh_waive_root();
>         }
>     }
> ```

yes

KonssnoK, 2024-06-26:
@zhangyanjiaoesp calling `esp_mesh_waive_root` always returns error

 ESP_ERR_MESH_DISCARD

and the device remains stuck.

Patch for latest code 
[04_waive_root.patch](https://github.com/user-attachments/files/15988041/04_waive_root.patch)

zhangyanjiaoesp, 2024-06-27:
@KonssnoK 

`ESP_ERR_MESH_DISCARD` this error indicates that the softAP doesn't have children. In your auth failure scenario, I think we should first confirm why the hotspot does not reply auth response. Have you tested on another hotspot or router?

KonssnoK, 2024-06-27:
> @KonssnoK
> 
> `ESP_ERR_MESH_DISCARD` this error indicates that the softAP doesn't have children. In your auth failure scenario, I think we should first confirm why the hotspot does not reply auth response. Have you tested on another hotspot or router?

No I updated the code and retried with the changes. I will go back to testing with the Samsung phone.
In any case, it should work any router/phone 🤔,
A different behavior in the AP should be handled in any case correctly by the STA

KonssnoK, 2024-06-27:
with samsung phone:
- device2 gets stuck

`W (435697) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:26, no_wnd_count:0, timeout_count:244
`
[240627dev1_1.txt](https://github.com/user-attachments/files/16011127/240627dev1_1.txt)
[240627dev3_1.txt](https://github.com/user-attachments/files/16011129/240627dev3_1.txt)
[240627dev2_1.txt](https://github.com/user-attachments/files/16011130/240627dev2_1.txt)

KonssnoK, 2024-06-27:
@zhangyanjiaoesp 
the failed auth does not happen on the samsung phone but dev2 apparently gets stuck as a child.

[240627dev3_2.txt](https://github.com/user-attachments/files/16011513/240627dev3_2.txt)
[240627dev1_2.txt](https://github.com/user-attachments/files/16011515/240627dev1_2.txt)
[240627dev2_2.txt](https://github.com/user-attachments/files/16011516/240627dev2_2.txt)

but on the Google phone what should be done when the root gets into a 2 205 error loop? Of course it doesn't have children because the rest of the network reshape by itself. What should be done to make it recover and search for other devices?

zhangyanjiaoesp, 2024-06-27:
@KonssnoK 
Please use this wifi lib to see whether the 2/205 error loop can be solved, thanks.
[wifi_lib_s3_0627.zip](https://github.com/user-attachments/files/16012882/wifi_lib_s3_0627.zip) (wifi firmware version: ace69bc)

KonssnoK, 2024-06-27:
@zhangyanjiaoesp i confirm i'm currently not able to reproduce the 2/205 infinite loop :)
Did you find another error ?

I will continue to try to reproduce

zhangyanjiaoesp, 2024-06-28:
> @zhangyanjiaoesp i confirm i'm currently not able to reproduce the 2/205 infinite loop on the pixel phone :) Did you find another error ?
> 
yes, I change the wifi code. For some routers, if the STA is powered off and restarted, the router still records the connection status of the STA, so when the STA is connected again to send the auth request, the router will not reply the auth response. In this case, we first send a deauth packet before the STA sends the auth request, so that the router can clear the previous connection state, and then the STA can send the auth request to the normal connection.

zhangyanjiaoesp, 2024-06-28:
> @zhangyanjiaoesp currently the only bug I am able to reproduce is a child device becoming stuck on `W (155190) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:0, timeout_count:123`
> 
> On the root node i cannot trigger much with the new library
> 
> * dev3 root
> * dev1 stuck
> 
> [240627dev2_3.txt](https://github.com/user-attachments/files/16015087/240627dev2_3.txt) [240627dev1_3.txt](https://github.com/user-attachments/files/16015089/240627dev1_3.txt) [240627dev3_3.txt](https://github.com/user-attachments/files/16015090/240627dev3_3.txt)
> 
> for reference, it seems that it's not that easy to trigger this error on the child as it was before the 0625 library, so it could be that the library fix of 0625 is not complete

OK, I will follow up to solve this problem.

KonssnoK, 2024-06-28:
ok thanks, meanwhile i will continue some testing to see if something else appears.
- For now i saw that sometimes a device gets completely stuck (like a deadlock) but I have no idea how to replicate, and from the logs it's impossible to understand why
- I will also move to the other ticket because fixed root seemed to trigger the same issue, but it could be once again that  the current fix is not final

KonssnoK, 2024-06-28:
apparently i cannot continue on the other issues until this is fixed.

@zhangyanjiaoesp do you have any way to check when this issue was introduced?

I would like to understand if this was always there (and means slow rate is a trigger we didn't check) or if instead it was introduced by a commit, and we could rollback devices in the field

zhangyanjiaoesp, 2024-07-01:
@KonssnoK 
I have added some debug log for the stuck issue, please use the new wifi lib to test, thanks.
[wifi_lib_s3_debug_0701.zip](https://github.com/user-attachments/files/16049434/wifi_lib_s3_debug_0701.zip)
wifi firmware version: fbf47d3

KonssnoK, 2024-07-01:
@zhangyanjiaoesp dev1 is stuck here without printing anything but
`>>>mesh_nwk_task_main,5572, nwk.state = 21`

[240701dev1_1.txt](https://github.com/user-attachments/files/16049808/240701dev1_1.txt)
[240701dev2_1.txt](https://github.com/user-attachments/files/16049812/240701dev2_1.txt)
[240701dev3_1.txt](https://github.com/user-attachments/files/16049809/240701dev3_1.txt)

KonssnoK, 2024-07-01:
one thing i don't understand is that when i press reset on dev2 or dev3 (with dev1 root), also the other 2 devices stop sending data completely for some time. Is this expected? It seems like the timeout of the device i'm resetting is blocking the parent, which for 3 devices isn't much, but if i think at 6 devices per node, means a huge lag for no reason...

zhangyanjiaoesp, 2024-07-01:
> @zhangyanjiaoesp dev1 is stuck here without printing anything but `>>>mesh_nwk_task_main,5572, nwk.state = 21`
> 
> [240701dev1_1.txt](https://github.com/user-attachments/files/16049808/240701dev1_1.txt) [240701dev2_1.txt](https://github.com/user-attachments/files/16049812/240701dev2_1.txt) [240701dev3_1.txt](https://github.com/user-attachments/files/16049809/240701dev3_1.txt)
> 
> more logs of dev1 once stuck
> 
> ```
> I (157009) wifi:station: 7c:df:a1:e2:7b:84 leave, AID = 1, bss_flags is 658531, bss:0x3fcb8f24
> I (157009) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,2>, prof:11
> I (186309) mesh: >>>8415[nwk]pending[0], state:MESH_NWK_TIMER
> >>>mesh_nwk_task_main,3768, nwk.state = 21
> >>>mesh_nwk_task_main,5572, nwk.state = 21
> I (216309) mesh: >>>8415[nwk]pending[0], state:MESH_NWK_TIMER
> >>>mesh_nwk_task_main,3768, nwk.state = 21
> >>>mesh_nwk_task_main,5572, nwk.state = 21
> I (246309) mesh: >>>8415[nwk]pending[0], state:MESH_NWK_TIMER
> >>>mesh_nwk_task_main,3768, nwk.state = 21
> >>>mesh_nwk_task_main,5572, nwk.state = 21
> I (276309) mesh: >>>8415[nwk]pending[0], state:MESH_NWK_TIMER
> >>>mesh_nwk_task_main,3768, nwk.state = 21
> >>>mesh_nwk_task_main,5572, nwk.state = 21
> ```

You didn't reproduce the stuck issue on `W (155190) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:0`,  but stuck on a new case, right? It seems that the wifi task stop running.

> for reference, it seems that it's not that easy to trigger this error on the child as it was before the 0625 library, so it could be that the library fix of 0625 is not complete

Can you confirm that this problem would not occurred without the change in 0625 library?

KonssnoK, 2024-07-01:
Since we have more logs in this lib I tried it also with our code that uses both fixed root and dynamic one.
There seem to be serious issues.
[240701dev1_3_alt.txt](https://github.com/user-attachments/files/16050578/240701dev1_3_alt.txt)
[240701dev2_3_alt.txt](https://github.com/user-attachments/files/16050580/240701dev2_3_alt.txt)

yes,
 i did not reproduce the `W (155190) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:0` yet, i'll resume it now

no, also before 0625 i had devices getting stuck without printing anything. The way to recover was rebooting them.

KonssnoK, 2024-07-01:
@zhangyanjiaoesp here is the infinite loop `W (155190) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:1, no_wnd_count:0` with library 0701

[240701dev1_4.txt](https://github.com/user-attachments/files/16050816/240701dev1_4.txt)
[240701dev3_4.txt](https://github.com/user-attachments/files/16050817/240701dev3_4.txt)
[240701dev2_4.txt](https://github.com/user-attachments/files/16050818/240701dev2_4.txt)

zhangyanjiaoesp, 2024-07-01:
@KonssnoK Can you capture the wifi packets via the Wireshark?

zhangyanjiaoesp, 2024-07-01:
@KonssnoK 
add more debug log in this wifi lib:
[wifi_lib_s3_0701_2.zip](https://github.com/user-attachments/files/16052331/wifi_lib_s3_0701_2.zip)
wifi firmware version: a93bcab

KonssnoK, 2024-07-01:
@zhangyanjiaoesp 
extension `.pcapng`
[test1.txt](https://github.com/user-attachments/files/16052346/test1.txt)

[98 more replies]

closed 2024-07-17 by espressif-bot