espressif/esp-idf #13213: i2c_master_probe returning ESP_OK for all addresses

puppy112358, 2024-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect to get the address of the device when I make it connected to ESP32S3 and address is supposed to be 0x68 .

### What is the actual behavior?

i2c_master_probe() returning ESP_OK for all addresses I have put in it. 

### Steps to reproduce.

1. Step   
    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_config_handle1,
                                       &iic_config_handle[0]));

    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_config_handle2,
                                       &iic_config_handle[1]));

2.    for (uint8_t i = 0x01; i < 0x90; i ++) {
            esp_err_t ret1 = i2c_master_probe(iic_config_handle[0], i, I2C_TOOL_TIMEOUT_VALUE_MS);
            if (ret1 == ESP_OK) {
                ESP_LOGI(TAG,"address: %x",i);
               device_number++;
                continue;
            } else{
                ESP_LOGE(TAG,"address %x failed",i);
            }
            esp_err_t ret2 = i2c_master_probe(iic_config_handle[1], 0x68, I2C_TOOL_TIMEOUT_VALUE_MS);
            if (ret2 == ESP_OK) {
                ESP_LOGI(TAG,"address: %x",i);
                device_number++;
                continue;
            }
    }

### More Information.

I have used i2ctools and it works normally...

mythbuster5, 2024-03-14:
@puppy112358 Any update? May I know the pins you are using. Or any picture related to logic analyzer? Because I cannot reproduce this issue on my master with my board. So it would be helpful if you give me more clue. I tried with following code.
```
    i2c_master_bus_config_t i2c_mst_config_1 = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .i2c_port = 0,
        .scl_io_num = 9,
        .sda_io_num = 10,
        .glitch_ignore_cnt = 7,
        .flags.enable_internal_pullup = true,
    };
    i2c_master_bus_handle_t bus_handle;

    TEST_ESP_OK(i2c_new_master_bus(&i2c_mst_config_1, &bus_handle));
    for (uint8_t addr = 0; addr < 127; addr++) {
        esp_err_t error_status = i2c_master_probe(bus_handle, addr, -1);

            // If we received the ACK bit, we found something! Print its address.
            if (error_status == ESP_OK) {
                printf("Found device at address %d (0x%02X)\n", addr, addr);

                uint8_t write_address = (addr << 1) | 0;
                uint8_t read_address = (addr << 1) | 1;

                printf("Write address: %d (0x%02X)\n", write_address, write_address);
                printf("Read address: %d (0x%02X)\n", read_address, read_address);
                
            }
    }
```
If there is no further update, I'm going to conclude this issue~. Thank you.

closed 2024-03-20 by espressif-bot

RCapolungo, 2025-02-26:
Hello @mythbuster5,
I'd like to reopen this issue as i have the same behavior explained by @puppy112358.

I'm using ESP-IDF v5.4 stable release commit: 473771bc14b7f76f9f7721e71b7ee16a37713f26.
What i'm doing is trying to probe an EEPROM module (that is not present) to address 0x50, and if the probe return success i'm calling the i2c_master_bus_add_device and writing some byte for testing purpose, otherwise i'm trying to probe address 0x52.
What i'm seeing is that the probe function returns always success, even if the module is not connected.
Here the communication between the master and the slave (that is not responding):

[image]

This are my bus config:
```
const i2c_master_bus_config_t i2c_bus_config_port1 =
{
    .i2c_port = hal_i2c_port1,
    .sda_io_num = halGpioPin20,
    .scl_io_num = halGpioPin21,
    .clk_source = I2C_CLK_SRC_DEFAULT,
    .glitch_ignore_cnt = 7,
    .intr_priority  = 0,
    .trans_queue_depth = 0,
    .flags = {
        .enable_internal_pullup = 1
        .allow_pd =               0
    }
};
```
And device config

```
i2c_device_config_t config = 
{
    .device_address = 0, //updated when checking the correct address
    .dev_addr_length = hal_i2c_add_len_7bit,
    .scl_speed_hz = hal_i2c_100_khz,
};
```
where `config.device_address` is set when the probe is successful.

mythbuster5, 2025-02-27:
@RCapolungo Which chip are you using?

RCapolungo, 2025-02-27:
@mythbuster5 sorry I completely forgot to add it, ESP32P4

mythbuster5, 2025-02-27:
That is very strange because I make a very quick test on my side. My code is very easy

```
i2c_master_bus_config_t i2c_mst_config_1 = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .i2c_port = 1,
        .scl_io_num = 21,
        .sda_io_num = 20,
        .flags.enable_internal_pullup = true,
    };
    i2c_master_bus_handle_t bus_handle;

    i2c_new_master_bus(&i2c_mst_config_1, &bus_handle);

    esp_err_t ret = i2c_master_probe(bus_handle, 0x50, -1);
    if (ret == ESP_OK) {
        printf("the device is on\n");
    } else {
        printf("the device is off\n");
    }
    i2c_del_master_bus(bus_handle);
```

And I tried EEPROM with address 0x50, first time the device is connected to P4
```
Running I2C master probe device test...
the device is on
MALLOC_CAP_8BIT usage: Free memory delta: -104 Leak threshold: -400
MALLOC_CAP_8BIT potential leak: Before 588380 bytes free, After 588276 bytes free (delta 104)
MALLOC_CAP_32BIT usage: Free memory delta: -104 Leak threshold: -400
MALLOC_CAP_32BIT potential leak: Before 588380 bytes free, After 588276 bytes free (delta 104)
/IDF/components/esp_driver_i2c/test_apps/i2c_test_apps/main/test_i2c_common.c:185:I2C master probe device test:PASS
Test ran in 45ms

-----------------------
1 Tests 0 Failures 0 Ignored
```

```
Running I2C master probe device test...
the device is off
MALLOC_CAP_8BIT usage: Free memory delta: -104 Leak threshold: -400
MALLOC_CAP_8BIT potential leak: Before 588380 bytes free, After 588276 bytes free (delta 104)
MALLOC_CAP_32BIT usage: Free memory delta: -104 Leak threshold: -400
MALLOC_CAP_32BIT potential leak: Before 588380 bytes free, After 588276 bytes free (delta 104)
/IDF/components/esp_driver_i2c/test_apps/i2c_test_apps/main/test_i2c_common.c:185:I2C master probe device test:PASS
Test ran in 45ms

-----------------------
1 Tests 0 Failures 0 Ignored
```
@RCapolungo I'm sorry becasue I don't have more clue so far. But I can explain the basic logic to you and maybe you can find sth. This logic is very ease.

here https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2c/i2c_master.c#L1290 we send the command and there is a queue wait https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2c/i2c_master.c#L519. The interrupt will give the event. If your device is not connected to the bus, P4 will trigger a nack interrupt and give that to the queue https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2c/i2c_master.c#L701. So may you please check whether the interrupt is triggered (with esp_rom_printf) and if it's triggered, why task does not get the event. Thanks!

RCapolungo, 2025-02-27:
I'm gonna check in the next days, thanks

RCapolungo, 2025-02-27:
Hi @mythbuster5, thank you for your assistance!
I found there was an issue on my end in parsing the esp_err_t returned by the i2c_master_probe