espressif/esp-idf #13216: ESP32C6 throws a ninja error

jpitz31, 2024-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2, 5.2

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

In building the Blink sample using esp32S3 or esp32C6 project builds just fine. for either v5.1.2 or v5.2
If I then load tusb_serial _device, sample using either 5.1.2 or 5.2 I can build the project just fine using esp32S3
but when I attempt to build the project using an ESP32C6 I get the following errors:

### What is the actual behavior?

Here is the error that is thrown:

CMake Error at /home/binshift/esp/v5.1.2/esp-idf/tools/cmake/build.cmake:540 (message):
  ERROR: Cannot find versions of "espressif/tinyusb" satisfying ">=0.14.2"
  for the current target "esp32c6".

Call Stack (most recent call first):
  /home/binshift/esp/v5.1.2/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:6 (project)

### Steps to reproduce.

1. Load Visual Studio, Load project tusb_serial_device
2. Configure extension for either 5.1.2 or 5.2
3. select the Port and device esp32S3
4. Build the project,  Project builds just fine
5. Select port and device esp32C6
6. build the project
7. above exception is thrown
8. 
...

igrr, 2024-02-19:
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/device/tusb_serial_device/README.md lists the chips supported by this example — ESP32-S2, ESP32-S3, ESP32-P4.

ESP32-C6 doesn't have a USB_OTG peripheral, so TinyUSB library doesn't work there.

jpitz31, 2024-02-19:
Thanks Ivan

-- 
http://joestechblog.com

*//
--------------------------------------------------------------------------------------*
*Old C programmers never die, they just get a new function pointer.*

*..."It began upon the following occasion: It is allowed on all hands, that
the primitive way of breaking eggs before we eat them, was upon the larger
end: but his present Majesty's grandfather, while he was a boy, going to
eat an egg, and breaking it according to the ancient practice, happened to
cut one of his fingers. Whereupon the Emperor his father published an
edict, commanding all his subjects, upon great penalties, to break the
smaller end of their eggs."  Jonathan Swift*

closed 2024-02-20 by tore-espressif