espressif/esp-idf #13217: Application scheduler not starting, RTC watchdog triggers

bnowicki-ce, 2024-02-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-231-ga7b62bbcaf

### Espressif SoC revision.

ESP32-D0WD-V3

### Operating System used.

macOS

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROVER-IE

### Power Supply used.

Battery

### What is the expected behavior?

Application is starting as it was starting when ESP-IDF v4.4 was used.

### What is the actual behavior?

Application scheduler is not starting, RTC watchdog gets triggered after 9 seconds.

### Steps to reproduce.

1. Have a working ESP-IDF v4.4 application.
2. Port it to ESP-IDF v5.1.
3. Erase chip completely and flash it with ported software.

`log_idf44` - boot log from working module when using ESP-IDF v4.4, notice `I (0) cpu_start: Starting scheduler on APP CPU.` at the end.

```
I (27) boot: compile time 10:10:31
I (28) boot: chip revision: 3
I (28) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (29) qio_mode: Enabling default flash chip QIO
I (29) boot.esp32: SPI Speed      : 40MHz
I (30) boot.esp32: SPI Mode       : QIO
I (31) boot.esp32: SPI Flash Size : 16MB
I (31) boot: Enabling RNG early entropy source...
I (32) boot: Partition Table:
I (32) boot: ## Label            Usage          Type ST Offset   Length
I (33) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (34) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (35) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (36) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (37) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (38) boot:  5 storage          Unknown data     01 82 00340000 00060000
I (39) boot: End of partition table
I (39) boot_comm: chip revision: 3, min. application chip revision: 0
I (40) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=40654h (263764) map
[... 181 lines cut ...]
I (1304) cpu_start: Application information:
I (1305) cpu_start: Project name:     ***REDACTED***
I (1306) cpu_start: App version:      t1-148-gd886b06-dirty
I (1306) cpu_start: Compile time:     Feb 19 2024 21:50:55
I (1307) cpu_start: ELF file SHA256:  9822f593b9be4ecc...
I (1308) cpu_start: ESP-IDF:          v5.1-231-ga7b62bbcaf
I (1309) cpu_start: Min chip rev:     v0.0
I (1309) cpu_start: Max chip rev:     v3.99
I (1310) cpu_start: Chip rev:         v3.0
I (1311) heap_init: Initializing. RAM available for dynamic allocation:
I (1311) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1312) heap_init: At 3FFBB300 len 00024D00 (147 KiB): DRAM
I (1313) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1314) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1315) heap_init: At 40094AF8 len 0000B508 (45 KiB): IRAM
I (1316) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (1317) spi_flash: detected chip: generic
I (1317) spi_flash: flash io: qio
```
[sdkconfig_v44.txt](https://github.com/espressif/esp-idf/files/14336877/sdkconfig_v44.txt)
[sdkconfig_v51.txt](https://github.com/espressif/esp-idf/files/14336878/sdkconfig_v51.txt)

### More Information.

Example application `hello_world` from ESP-IDF v5.1 works fine, I suppose the problem is located somewhere in `sdkconfig`, but I don't know where to look. Issue persists in ESP-IDF v5.2

nopnop2002, 2024-02-20:
>Example application hello_world from ESP-IDF v5.1 works fine

Try removing sdkconfig.default from your project.

ESP-Marius, 2024-02-21:
@bnowicki-ce I didnt manage to reproduce this, but I'd guess it might have to do with the flash settings (since the last output is related to flash configuration)

Can you try:
* Enabling debug level for logging to better see what the app is doing before crashing
* Try to set some of the flash related configs to their default values

bnowicki-ce, 2024-02-21:
@ESP-Marius I've changed Serial Flasher config to its defaults:

* enabled `Disable download stub`
* changed `Flash SPI mode` to `DIO`
* disabled `Detect flash size when flashing bootloader`

Please find verbose logs before and after change attached.
[log_idf51_verbose.log](https://github.com/espressif/esp-idf/files/14357796/log_idf51_verbose.log)
[log_idf51_verbose_flash_defaults.log](https://github.com/espressif/esp-idf/files/14357797/log_idf51_verbose_flash_defaults.log)

ESP-Marius, 2024-02-22:
From the log it seems like your app is calling a lot of constructor functions, most likely it is one of these that are doing something they shouldn't and causing issues.

`D (1608) cpu_start: calling init function: 0x400d86f0`, what is this function at `0x400d86f0` doing?

If you are able to connect a JTAG and see what is is happening or share an example that can reproduce this issue then that would be even more convenient.

bnowicki-ce, 2024-02-22:
It is a bit problematic for me to attach JTAG, but you gave me a hint where to search. Code executions stops at WiFi constructor:

```cpp
  WiFi::WiFi() {
      esp_netif_init();
      esp_event_loop_create_default();
      esp_event_handler_instance_register(WIFI_EVENT,
                                          ESP_EVENT_ANY_ID,
                                          &WiFi::wifi_event_callback,
                                          this,
                                          &m_instance_wifi_event);

      esp_event_handler_instance_register(IP_EVENT,
                                          ESP_EVENT_ANY_ID,
                                          &WiFi::wifi_event_callback,
                                          this,
                                          &m_instance_ip_event);
  }
```
It stops precisely at `esp_netif_init()`, because when I swap this function with `esp_event_loop_create_default()` I have got additional two prints:

```
D (1270) event: created task for loop 0x3ffafeb8
D (1271) event: created event loop 0x3ffafeb8
```

Is there a new way of using WiFi in this version of IDF?

ESP-Marius, 2024-02-23:
Not sure if `esp_netif_init()` changed a lot between 4.4 and 5.1, maybe @david-cermak could give some suggestions what could be causing it to get stuck here.

In general though you should be a bit careful with what you do in your cpp constructors. These are called before all the esp-idf system services are running. So there might be some actions that can work in a normal app, but will not behave correctly if called from a constructor.

You instantiate `WiFi` statically dont you? Try just doing it after `app_main` starts instead and I suspect you will be able to bypass this issue.

david-cermak, 2024-02-23:
> Not sure if esp_netif_init() changed a lot between 4.4 and 5.1

One potential problem could be that we're now *blocking* until the TCP/IP task is up and running in `esp_netif_init()`. (This was backported to `v4.4` in 29b8a00c0a1ca7d7112a3b03c3e87b44c06ac5bf though, but maybe it was some older v4.4 without the fix)
To check if this is the case you can try to replace these two lines

https://github.com/espressif/esp-idf/blob/5f4249357372f209fdd57288265741aaba21a2b1/components/esp_netif/lwip/esp_netif_lwip.c#L543-L544

with
```cpp
tcpip_init(NULL, NULL);
```

But in general, calling the `esp_netif_init()` in `app_main()` would fix your issues.

bnowicki-ce, 2024-03-01:
@david-cermak @ESP-Marius 

This indeed solved the issue, thank you very much!

closed 2024-03-01 by bnowicki-ce