espressif/esp-idf #13219: Coredump failed on overflowed tasks due to excess stack usage of panic handler

andylinpersonal, 2024-02-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2 7380f96 | v5.2 93ea06f | v5.3 4d90eed

### Espressif SoC revision.

esp32s3: v0.1 | esp32c3: v0.4 | esp32c6: v0.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32c3: LuatOS CORE-ESP32-C3 | esp32c6: esp32-c6-devkitc-1-n8 | esp32s3: esp32-s3-devkitc-1-n32r8v

### Power Supply used.

USB

### What is the expected behavior?

Core dump should be reliably performed, even for the overflowed tasks.

Following is the normal coredump flow:
1. [Exception occurred]
2. Run the exception handler on the current stack.
3. Fatal exception -> invoke panic handler.
4. If coredump is enabled, invoke the coredump function.
5. If coredump stack is enabled, swap the stack to the preallocated one.
6. Perform real coredump stuff.
7. Swap back the stack to the panic handler's one.
8. Reset.

### What is the actual behavior?

* `espcoredump` crashed randomly when the original task was overflowed.
* After reserving a large space before the stack, most configurations can perform coredump successfully.
  * To proof and check the memory region polluted by the overflowed panic handler and the coredump function, I have placed a second canary watchpoint below the first one. It seems that reserving some space under the end of stack could be possible a workaround as the second watchpoint will not be triggered duing coredump if the space is sufficiently large.

  ```
  Original stack:

  |  ---  stack top   ---
  |  ====================
  |  <-   stack body   ->
  |
  |    [1st canary wp]
  |  ====================
  V  ---  stack end   ---

  Reserve some space under the stack end as workaround:

  G  ---  stack top   ---
  R  ====================
  O  <-   stack body   ->
[... 382 lines cut ...]
sdkconfig.defaults
```Kconfig
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y

CONFIG_ESP32_ENABLE_COREDUMP_TO_FLASH=y

# Optionally enable this to print more details.
# CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y

# -O0
CONFIG_COMPILER_OPTIMIZATION_NONE=y
# -Og (Default)
# CONFIG_COMPILER_OPTIMIZATION_DEBUG=y
# CONFIG_COMPILER_OPTIMIZATION_DEFAULT=y
# -Os
# CONFIG_COMPILER_OPTIMIZATION_SIZE=y

# CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_ENABLE=y

# Enable and allocate a dedicated stack for coredump code to handle the overflowed stacks safely.
CONFIG_ESP_COREDUMP_STACK_SIZE=1536
CONFIG_ESP32_CORE_DUMP_STACK_SIZE=1536

```

andylinpersonal, 2024-02-20:
Extra log files:
[logs.tar.gz](https://github.com/espressif/esp-idf/files/14338741/logs.tar.gz)

andylinpersonal, 2024-02-25:
# Additional note for ESP32-S3 with -O0:
1. Coredump cannot even continue with enlarged stack
  * With a bit of `esp_rom_printf`, it seems like S3 hanged near `esp_core_dump_replace_sp()` and `xthal_window_spill()`. At least `esp_rom_printf` cannot print anything anymore after `xthal_window_spill()`.

andylinpersonal, 2024-03-09:
* Update some log files of v5.3 to c460e1cd7e20063ef600107fb58555f597e61b7e.
* Add another examination method.

[test.tar.gz](https://github.com/espressif/esp-idf/files/14546016/test.tar.gz)

andylinpersonal, 2024-03-30:
Any update? Thanks.

erhankur, 2024-03-30:
@andylinpersonal Thanks for reporting the issue. We have some findings regarding stack replacement on both xtensa and riscv chips. I will wrap up what we did and explain here next week. 

Regarding overflow scenarios, I am sorry, I couldn't find a time to investigate yet. Will check it soon.

AxelLin, 2024-06-06:
@erhankur Any follow up of https://github.com/espressif/esp-idf/issues/13219#issuecomment-2027991578 ?