espressif/esp-idf #13220: Core dump occurs "ValueError: invalid literal for int() with base 16: ''"

yizi15, 2024-02-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

application Core dump can be decode.

### What is the actual behavior?

ValueError: invalid literal for int() with base 16: ''

### Steps to reproduce.

1. enable the core dump 
(Top) → Component config → Core dump
                                                                                                             Espressif IoT Development Framework Configuration
    Data destination (UART)  --->
    Core dump data format (ELF format)  --->
    Core dump data integrity check (Use CRC32 for integrity verification)  --->
[*] Enable coredump logs for debugging
(64) Maximum number of tasks
(0) Delay before print to UART
(2048) Reserved stack size
    Handling of UART core dumps in IDF Monitor (Decode and show summary (info_corefile))  --->
2. 
Add 'volatile int i = 1/0;' at the end of app()

### Build or installation Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  0 register dump:
PC      : 0x42009e24  PS      : 0x00060430  A0      : 0x82080892  A1      : 0x3fca7340  
0x42009e24: app_main at D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/wifi/power_save/main/power_save.c:119 (discriminator 2)

A2      : 0x00000003  A3      : 0x3c094484  A4      : 0x3c094504  A5      : 0x000001a1  
A6      : 0x3c094484  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x00000000
A10     : 0x00001388  A11     : 0x3fca7288  A12     : 0x00000001  A13     : 0x00000001
A14     : 0x3fca7330  A15     : 0x0000000c  SAR     : 0x0000001a  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
0x40056f5c: memcpy in ROM

0x40056f72: memcpy in ROM

Backtrace: 0x42009e21:0x3fca7340 0x4208088f:0x3fca7370 0x40380a89:0x3fca73a0
0x42009e21: app_main at D:/Espressif5/frameworks/esp-idf-v5.1.2/examples/wifi/power_save/main/power_save.c:119 (discriminator 2)

0x4208088f: main_task at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/freertos/app_startup.c:208 (discriminator 13)

0x40380a89: vPortTaskWrapper at D:/Espressif5/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 810ed8da05f7c9ee

Initiating core dump!
I (827) esp_core_dump_uart: Press Enter to print core dump to UART...
I (834) esp_core_dump_uart: Print core dump to uart...
I (840) esp_core_dump_common: Backing up stack @ 0x3fca7120 and use core dump stack @ 0x3fca00c0
Core dump started (further output muted)
Received  20 kB...
Core dump finished!

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\__main__.py", line 4, in <module>
    main()
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\idf_monitor.py", line 432, in main
    monitor.main_loop()
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\idf_monitor.py", line 178, in main_loop
    self._main_loop()
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\idf_monitor.py", line 294, in _main_loop
    super()._main_loop()
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\idf_monitor.py", line 226, in _main_loop
    self.serial_handler.handle_serial_input(data, self.console_parser, self.coredump,  # type: ignore
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\serial_handler.py", line 114, in handle_serial_input
    with coredump.check(line_strip):
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\coredump.py", line 128, in check
    self._check_coredump_trigger_before_print(line)
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\coredump.py", line 106, in _check_coredump_trigger_before_print
    self._process_coredump()
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_idf_monitor\base\coredump.py", line 79, in _process_coredump
    coredump.info_corefile()
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_coredump\coredump.py", line 566, in info_corefile
    self.print_threads_info(task_info)
  File "D:\Espressif5\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_coredump\coredump.py", line 355, in print_threads_info
    pxEndOfStack = int(self.gdb_esp.parse_tcb_variable(tcb_addr, 'pxEndOfStack'), 16)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 16: ''
```

### More Information.

I want save the core dump in the sdcard(copy from flash). When i decode the dump.elf, the error occured.  Then I use the uart decode.

erhankur, 2024-02-21:
@yizi15 I would like to reproduce the issue. How did you create the panic error? Can you send your sdkconfig also?

yizi15, 2024-02-22:
> @yizi15 I would like to reproduce the issue. How did you create the panic error? Can you send your sdkconfig also?

the example code 
[power_save.zip](https://github.com/espressif/esp-idf/files/14368892/power_save.zip)

erhankur, 2024-02-22:
It worked on my end. I tested on MacOS with esp32s3, idf v5.1.2 and esp-coredump-1.10.0 

Will be testing on Windows. 

```
Initiating core dump!
I (5874) esp_core_dump_uart: Press Enter to print core dump to UART...
I (5882) esp_core_dump_uart: Print core dump to uart...
I (5887) esp_core_dump_common: Backing up stack @ 0x3fca8680 and use core dump stack @ 0x3fca1560
Core dump started (further output muted)
Received  21 kB...
Core dump finished!
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3fca8a68, name: 'main', GDB name: 'process 1070238312'
Crashed task is not in the interrupt context

================== CURRENT THREAD REGISTERS ===================
exccause       0x6 (IntegerDivideByZeroCause)
excvaddr       0x0
epc1           0x4037ba7b
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x42009e60          0x42009e60 <app_main+140>
lbeg           0x40056f5c          1074098012
lend           0x40056f72          1074098034
lcount         0xffffffff          4294967295
sar            0x1a                26
ps             0x60820             395296
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x820811ba          -2113400390
a1             0x3fca88a0          1070237856
a2             0x3                 3
a3             0x3c094584          1007240580
a4             0x3c094604          1007240708
a5             0x1e4               484
a6             0x3c094584          1007240580
a7             0x0                 0
a8             0x1                 1
a9             0x0                 0
a10            0x1388              5000
a11            0x3fca87e8          1070237672
a12            0xa                 10
a13            0x4                 4
a14            0x3fca8890          1070237840
a15            0xc                 12

==================== CURRENT THREAD STACK =====================
#0  0x42009e60 in app_main () at /Users/erhan/esp/workspace/power_save/main/power_save.c:119
#1  0x420811ba in main_task (args=<optimized out>) at /Users/erhan/dev/esp-idf/components/freertos/app_startup.c:208
#2  0x40380f58 in vPortTaskWrapper (pxCode=0x42081110 <main_task>, pvParameters=0x0) at /Users/erhan/dev/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

======================== THREADS INFO =========================
  Id   Target Id          Frame 
* 1    process 1070238312 0x42009e60 in app_main () at /Users/erhan/esp/workspace/power_save/main/power_save.c:119
  2    process 1070240196 0x4037b95a in esp_cpu_wait_for_intr () at /Users/erhan/dev/esp-idf/components/esp_hw_support/cpu.c:121
  3    process 1070242080 0x4037b95a in esp_cpu_wait_for_intr () at /Users/erhan/dev/esp-idf/components/esp_hw_support/cpu.c:121
  4    process 1070248708 0x400559e0 in ?? ()
  5    process 1070227660 0x400559e0 in ?? ()
  6    process 1070229288 0x400559e0 in ?? ()
  7    process 1070252676 0x400559e0 in ?? ()
  8    process 1070263808 0x400559e0 in ?? ()
  9    process 1070233732 0x400559e0 in ?? ()
  
  .....
  
  ```

ztefanjo, 2024-02-28:
I get a similar error on ESP32-S3 after changing to v5.1.2 when parsing some coredumps:
`  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\espcoredump.py", line 77, in <module>
    main()
  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\espcoredump.py", line 62, in main
    temp_core_files = espcoredump.info_corefile()
  File "C:\Data\Git\mower\Ext\python_env\idf5.1_py3.9_env\lib\site-packages\esp_coredump\coredump.py", line 566, in info_corefile
    self.print_threads_info(task_info)
  File "C:\Data\Git\mower\Ext\python_env\idf5.1_py3.9_env\lib\site-packages\esp_coredump\coredump.py", line 355, in print_threads_info
    pxEndOfStack = int(self.gdb_esp.parse
_tcb_variable(tcb_addr, 'pxEndOfStack'), 16)
ValueError: invalid literal for int() with base 16: ''`
Attaching sdkconfig.
[sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/14434265/sdkconfig.defaults.txt)

yizi15, 2024-02-29:
> Can you select don't decode option and send the base64 output with your app elf file? <img alt="image" width="700" src="https://private-user-images.githubusercontent.com/43336369/308574516-833e5ca5-b36b-4dbb-8805-e21edb1b703b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNzI5NDcsIm5iZiI6MTcwOTE3MjY0NywicGF0aCI6Ii80MzMzNjM2OS8zMDg1NzQ1MTYtODMzZTVjYTUtYjM2Yi00ZGJiLTg4MDUtZTIxZWRiMWI3MDNiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI5VDAyMTA0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA4MGQ5NWQ0YWNiMDk4ZWU3ZDgxMzU5ZjQwMTc3OGUwZGM0OWRmZTg1NmRlYTlkZjcxOGVhMzE3NDZkYjg3YWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4xiCcD3x9ej1zAwW6Hir6PHHGBGmPHrNJgUhw8dJbaQ">
> 
> gdb couldn't find `pxEndOfStack` variable. It should be there if `configRECORD_STACK_HIGH_ADDRESS` is set to 1. https://github.com/espressif/esp-idf/blob/release/v5.1/components/freertos/FreeRTOS-Kernel/tasks.c#L359C23-L359C35

[Uploading core_dump.zip…]()

erhankur, 2024-02-29:
I couldn't see `================= CORE DUMP END =================`  line in your coredump-elf file.  Can you send full lines between start and end?

yizi15, 2024-03-01:
> I couldn't see `================= CORE DUMP END =================`  line in your coredump-elf file. Can you send full lines between start and end?

I dont Konw what happend, the new log has ================= CORE DUMP END =================.
[core_dump.zip](https://github.com/espressif/esp-idf/files/14456591/core_dump.zip)

erhankur, 2024-03-01:
Unfortunately, your base64 file looks corrupted and can not be parsable. Did you test this with another board or OS? I still can not reproduce the issue on Windows as well. 

Was it working with idf 5.0 & esp-coredump 1.2.0?

Another thing we can try is; you can save the coredump file to the flash and send me the coredump partition content in a file.

And just to understand if the other part of the data is ok, you can edit `C:\Data\Git\mower\Ext\python_env\idf5.1_py3.9_env\lib\site-packages\esp_coredump\coredump.py` file like below.

 `pxEndOfStack = 0 # int(self.gdb_esp.parse _tcb_variable(tcb_addr, 'pxEndOfStack'), 16)`

I wonder if will you see the rest of the output.

closed 2025-02-12 by yizi15