espressif/esp-idf #13224: [release/v5.1] "Re-entered core dump! Exception happened during core dump!"

chipweinberger, 2024-02-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1 (d922d4178) (Feb 20, 2024)

### Espressif SoC revision.

ESP32-S3 (v0.1)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

I'm simulating a crash using ESP_ERROR_CHECK, I expect a coredump to happen

### What is the actual behavior?

The coredump crashes. I think it crashes here

```
    ESP_COREDUMP_LOG_PROCESS("============== Data size = %d bytes ============", tot_len);
    
    // Prepare write elf
    if (write_cfg->prepare) {
        err = write_cfg->prepare(write_cfg->priv, (uint32_t*)&tot_len);  // crashes here I think
```

### Steps to reproduce.

force a crash by calling _esp_error_check_failed(ESP_ERR_INVALID_RESPONSE, __FILE__, __LINE__,   __ASSERT_FUNC, #x);    

### Debug Logs.

```plain
I (43951) pd console handlers: /api/debug/force-system-crash called 
E (44361) pd api debug: force system crash
ESP_ERROR_CHECK failed: esp_err_t 0x108 (ESP_ERR_INVALID_RESPONSE) at 0x420712f1
0x420712f1: pd_api_debug_force_system_crash(pd_error_t*) at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jcommon/api/core/pd_api_debug.cpp:39

file: "/Volumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jcommon/api/core/pd_api_debug.cpp" line 39
func: cJSON* pd_api_debug_force_system_crash(pd_error_t*)
expression: false

abort() was called at PC 0x40384b03 on core 0
0x40384b03: _esp_error_check_failed at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_system/esp_err.c:50

Backtrace: 0x40376057:0x3fcc7170 0x40384b1d:0x3fcc71a0 0x4038d817:0x3fcc71d0 0x40384b03:0x3fcc7250 0x420712f1:0x3fcc7290 0x42014c3f:0x3fcc72c0 0x420ba01d:0x3fcc7300 0x42015400:0x3fcc7330 0x40385a8e:0x3fcc7370
[... 294 lines cut ...]

CONFIG_HEAP_POISONING_COMPREHENSIVE=y
CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y

CONFIG_LWIP_MAX_SOCKETS=16
CONFIG_LWIP_MAX_ACTIVE_TCP=32
CONFIG_LWIP_MAX_LISTENING_TCP=32
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=6500
CONFIG_LWIP_TCP_WND_DEFAULT=6500
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=4096

CONFIG_MBEDTLS_HARDWARE_AES=n
CONFIG_MBEDTLS_MPI_USE_INTERRUPT=n

CONFIG_PTHREAD_TASK_STACK_SIZE_DEFAULT=4096

CONFIG_MDNS_TASK_STACK_SIZE=3072
CONFIG_MONITOR_BAUD=460800

CONFIG_CXX_EXCEPTIONS=y
CONFIG_CXX_EXCEPTIONS_EMG_POOL_SIZE=2048
CONFIG_STACK_CHECK_ALL=y
CONFIG_STACK_CHECK=y
CONFIG_WARN_WRITE_STRINGS=y
```

chipweinberger, 2024-02-21:
I'll be debugging this further, give me 30 mins

chipweinberger, 2024-02-21:
`write_cfg 0xa5a5a5a5`

uh oh. the `write_cfg` struct is corrupt memory

chipweinberger, 2024-02-21:
So i added a bunch of logs:

`ESP_COREDUMP_LOG_PROCESS("write_cfg %p", write_cfg);`

and this time it succeeded. either the logs solved it somehow, or it is a race condition perhaps, etc.

chipweinberger, 2024-02-21:
specificaly, it is these 3 logs that somehow fix the coredump crash. When they're there, the coredump works. When they're not there, it crashes. 

```
inline void esp_core_dump_write(panic_info_t *info, core_dump_write_config_t *write_cfg)
{
    ESP_COREDUMP_LOG_PROCESS("write_cfg199 %p", write_cfg);

#ifndef CONFIG_ESP_COREDUMP_ENABLE_TO_NONE
    esp_err_t err = ESP_ERR_NOT_SUPPORTED;
    s_exc_frame = (void*) info->frame;

    esp_core_dump_setup_stack();

ESP_COREDUMP_LOG_PROCESS("write_cfg198%p", write_cfg);

    esp_core_dump_port_init(info);

ESP_COREDUMP_LOG_PROCESS("write_cfg197 %p", write_cfg);

#if CONFIG_ESP_COREDUMP_DATA_FORMAT_BIN
    err = esp_core_dump_write_binary(write_cfg);
#elif CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF
    err = esp_core_dump_write_elf(write_cfg);
#endif
    if (err != ESP_OK) {
        ESP_COREDUMP_LOGE("Core dump write binary failed with error=%d", err);
    }
    esp_core_dump_report_stack_usage();
#endif
}
```

chipweinberger, 2024-02-21:
I wont be debugging this further:
1. it's kind of out of my expertise. I think I'd take a look at the generated assembly next
2. the `ESP_COREDUMP_LOG_PROCESS` workaround is sufficient for my needs

But this is clearly a pretty severe bug somewhere.

andylinpersonal, 2024-02-21:
Maybe related to this issue?
https://github.com/espressif/esp-idf/issues/13219

chipweinberger, 2024-02-21:
True, but I've no idea why adding print statements reliably fixes things

`ESP_COREDUMP_LOG_PROCESS("write_cfg199 %p", write_cfg);`

erhankur, 2024-02-21:
@chipweinberger Thanks for reporting. I will check it.

chipweinberger, 2024-05-29:
fixed here? https://github.com/espressif/esp-idf/commit/9046bf1645da1a8d49d7927b9836a6c380ec3e3c

closed 2024-05-29 by chipweinberger