espressif/esp-idf #13225: Why there is no xtensa-esp32-elf-ldd file in xtensa-esp32-elf compile tool?

robbinlu-ayla, 2024-02-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Why there is no xtensa-esp32-elf-ldd file in xtensa-esp32-elf compile tool?
ls -l ./.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin
xtensa-esp32-elf-addr2line
xtensa-esp32-elf-ar
xtensa-esp32-elf-as
xtensa-esp32-elf-c++
xtensa-esp32-elf-cc -> xtensa-esp32-elf-gcc
xtensa-esp32-elf-c++filt
xtensa-esp32-elf-cpp
xtensa-esp32-elf-ct-ng.config
xtensa-esp32-elf-elfedit
xtensa-esp32-elf-g++
xtensa-esp32-elf-gcc
xtensa-esp32-elf-gcc-8.4.0
xtensa-esp32-elf-gcc-ar
xtensa-esp32-elf-gcc-nm
xtensa-esp32-elf-gcc-ranlib
xtensa-esp32-elf-gcov
xtensa-esp32-elf-gcov-dump
xtensa-esp32-elf-gcov-tool
xtensa-esp32-elf-gdb
xtensa-esp32-elf-gdb-add-index
xtensa-esp32-elf-gprof
xtensa-esp32-elf-ld
xtensa-esp32-elf-ld.bfd
xtensa-esp32-elf-nm
xtensa-esp32-elf-objcopy
xtensa-esp32-elf-objdump
xtensa-esp32-elf-ranlib
xtensa-esp32-elf-readelf
xtensa-esp32-elf-size
xtensa-esp32-elf-strings
xtensa-esp32-elf-strip

antmak, 2024-02-21:
Hi! There is no much sense in having *-ldd, because the toolchain usually does not build shared objects. In general, the lack of ldd is a kind of default in the binutils build for our embedded purposes.

BTW, `ldd /usr/bin/okular` is equal to `LD_TRACE_LOADED_OBJECTS=1 /usr/bin/okular` - just an operation system feature if the OS supports shared objects. (okular is just an example of linux binary)

If you are interesting in looking inside object files, I'd recommend to use `xtensa-esp32-elf-objdump -x <obj file>` or something else.

BTW, `objdump -x /usr/bin/okular | grep NEED` would show you the required shared objects if they were there. But an IDF app doesn't has them.

robbinlu-ayla, 2024-02-24:
Thanks your response.
Is there an easy way to know what libraries and components included in an  ESP app image?

igrr, 2024-02-24:
For binary size analysis you can use https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/size.html#idf-py-size.

The other option is to generate an SBOM to get the list of libraries: https://github.com/espressif/esp-idf-sbom.

closed 2026-01-14 by gerekon