espressif/esp-idf #13229: exFAT on esp32s3 support feature request

alex04130, 2024-02-22:
### Is your feature request related to a problem?

I wanted to have a exfat support on my esp32 board with ESP-IDF. And I found a way to let it work.

### Describe the solution you'd like.

# There is already a solution that can be applied to this question, which relies on IDF's FatFs library. 
| in ffconf.h
set FF_LBA64 to 1
set FF_FS_EXFAT to 1
* because LFN must be enabled when enabling exFAT
Long filename support on menu-config should not be set to  "No long filenames".

* because when using spi sd host and exFAT format, the driver will throughout an ESP_ERR_INVALID_RESPONSE error in the trim function
 the FF_USE_TRIM in ffconf.h should be set to 0

The solution I'd like is to put these in the menu-config since I tested it and it worked.

igrr, 2024-02-22:
We have intentionally not added an easy way to enable exFAT support via menuconfig. This was because in the development stage, people may enable this option and forget about it. However, usage of exFAT implementations in many cases requires a [license](https://www.microsoft.com/en-us/legal/intellectualproperty/tech-licensing/programs?activetab=pivot1%3aprimaryr2) from Microsoft.

On balance, we decided that if you are going through the trouble of obtaining this license, using a patched version of ffconf.h isn't the biggest of the issues. And for everyone else, this prevents potential legal issues.

alex04130, 2024-02-23:
t

alex04130, 2024-02-23:
Thank you very much

closed 2024-02-23 by alex04130

Red-Owl, 2024-05-25:
exFAT as part of the Linux kernel

In August 2019 Microsoft announced its support for integrating exFAT into the Linux kernel. exFAT is now part of the Linux kernel 5.7 and devices using this code are covered under the GPLv2 license.

translate: your software is open source or pay license to microsoft

Trackhe, 2025-04-16:
Is there any update on this?