espressif/esp-idf #13230: restful_server example not working

krupis, 2024-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to run the restful server example to test the webserver using framework such as vue.js

I have followed the instructions to see how to build the firmware:
[image]

When trying to install ```npm``` packages, I am getting ton of different warnings:

```
PS C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo> npm install
npm WARN deprecated @babel/plugin-proposal-unicode-property-regex@7.18.6: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-unicode-property-regex instead.
npm WARN deprecated @babel/plugin-proposal-optional-catch-binding@7.18.6: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-optional-catch-binding instead.
npm WARN deprecated @babel/plugin-proposal-json-strings@7.18.6: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-json-strings instead.
npm WARN deprecated @babel/plugin-proposal-class-properties@7.18.6: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-class-properties instead.
npm WARN deprecated stable@0.1.8: Modern JS already guarantees Array#sort() is a stable sort, so this library is deprecated. See the compatibility table on MDN: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array/sort#browser_compatibility
npm WARN deprecated source-map-url@0.4.1: See https://github.com/lydell/source-map-url#deprecated
npm WARN deprecated figgy-pudding@3.5.2: This module is no longer supported.
npm WARN deprecated request-promise-native@1.0.9: request-promise-native has been deprecated because it extends the now deprecated request package, see https://github.com/request/request/issues/3142
npm WARN deprecated @hapi/bourne@1.3.2: This version has been deprecated and is no longer supported or maintained
npm WARN deprecated @hapi/topo@3.1.6: This version has been deprecated and is no longer supported or maintained
npm WARN deprecated eslint-loader@2.2.1: This loader has been deprecated. Please use eslint-webpack-plugin
npm WARN deprecated consolidate@0.15.1: Please upgrade to consolidate v1.0.0+ as it has been modernized with several long-awaited fixes implemented. Maintenance is supported by Forward Email 
at https://forwardemail.net ; follow/watch https://github.com/ladjs/consolidate for updates and release changelog
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated @babel/plugin-proposal-object-rest-spread@7.20.7: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use @babel/plugin-transform-object-rest-spread instead.
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm WARN deprecated @babel/plugin-proposal-async-generator-functions@7.20.7: This proposal has been merged to the ECMAScript standard and thus this plugin is no longer maintained. Please use 
@babel/plugin-transform-async-generator-functions instead.
npm WARN deprecated source-map-resolve@0.5.3: See https://github.com/lydell/source-map-resolve#deprecated
npm WARN deprecated chokidar@2.1.8: Chokidar 2 does not receive security updates since 2019. Upgrade to chokidar 3 with 15x fewer dependencies
npm WARN deprecated chokidar@2.1.8: Chokidar 2 does not receive security updates since 2019. Upgrade to chokidar 3 with 15x fewer dependencies
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm WARN deprecated circular-json@0.3.3: CircularJSON is in maintenance only, flatted is its successor.
npm WARN deprecated html-webpack-plugin@3.2.0: 3.x is no longer supported
npm WARN deprecated @hapi/address@2.1.4: Moved to 'npm install @sideway/address'
npm WARN deprecated babel-eslint@10.1.0: babel-eslint is now @babel/eslint-parser. This package will no longer receive updates.
npm WARN deprecated webpack-chain@4.12.1: Package no longer supported. Contact Support at https://www.npmjs.com/support for more info.
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated @hapi/hoek@8.5.1: This version has been deprecated and is no longer supported or maintained
npm WARN deprecated @hapi/joi@15.1.1: Switch to 'npm install joi'
npm WARN deprecated svgo@1.3.2: This SVGO version is no longer supported. Upgrade to v2.x.x.
npm WARN deprecated vue@2.7.16: Vue 2 has reached EOL and is no longer actively maintained. See https://v2.vuejs.org/eol/ for more details.
npm WARN deprecated core-js@2.6.12: core-js@<3.23.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Some versions have web compatibility issues. Please, upgrade your dependencies to the actual version of core-js.

added 1493 packages, and audited 1494 packages in 1m

123 packages are looking for funding
  run `npm fund` for details

82 vulnerabilities (1 low, 53 moderate, 23 high, 5 critical)

To address issues that do not require attention, run:       
  npm audit fix

To address all issues (including breaking changes), run:    
  npm audit fix --force

Run `npm audit` for details.
PS C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo> 
```

Regardless of all the issues above, I have tried to run command ```npm run build``` but it does not seem to go through properly:
[... 51 lines cut ...]
    at AsyncSeriesHook.lazyCompileHook (C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\tapable\lib\Hook.js:154:20)
    at C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\webpack\lib\Compilation.js:1409:36
    at AsyncSeriesHook.eval [as callAsync] (eval at create (C:\Users\petrikas.lu\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:6:1)
    at AsyncSeriesHook.lazyCompileHook (C:\Users\petrikas.lu\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\tapable\lib\Hook.js:154:20)
    at C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\webpack\lib\Compilation.js:1405:32
    at AsyncSeriesHook.eval [as callAsync] (eval at create (C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:6:1)
    at AsyncSeriesHook.lazyCompileHook (C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\tapable\lib\Hook.js:154:20)
    at Compilation.seal (C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\webpack\lib\Compilation.js:1342:27)
    at C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\webpack\lib\Compiler.js:675:18
    at C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\webpack\lib\Compilation.js:1261:4
    at AsyncSeriesHook.eval [as callAsync] (eval at create (C:\Users\petrikas.lu\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:24:1)
[... cut ...]
    at C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo\node_modules\webpack\lib\Compilation.js:1097:9
    at process.processTicksAndRejections (node:internal/process/task_queues:77:11)

 ERROR  Build failed with errors.
PS C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo> 
```

After that, I have tried tu run ```npm audit``` and ```npm audit --force``` to see if it makes any difference but I still cannot seem to get it built. I would highly appreciate any help on how to get this setup and running.

LucianoBenedetti, 2024-02-25:
I had the same problem, apparently is something about the dependence version. I fixed like that:
- I used Node Js v20.11.1 [NodeJs](https://nodejs.org/download/release/v20.11.1/) and I changed 
**_pacakge.json_** the scrip field from:
```
"scripts": {
    "serve": "vue-cli-service"
    "build": "vue-cli-service"
    "lint":  "vue-cli-service"
},
```

To:
```
scripts": {
    "serve": "set NODE_OPTIONS=--openssl-legacy-provider && vue-cli-service serve",
    "build": "set NODE_OPTIONS=--openssl-legacy-provider && vue-cli-service build",
    "lint": "set NODE_OPTIONS=--openssl-legacy-provider && vue-cli-service lint"
},
```

- Then I runner the command as readme said:
```
cd path_to_this_example/front/web-demo
npm install
npm run build
```
it install the dependence but **_still some errors_**. I didn’t fix it with audit fix or --force, just I leaved them  because in other way it don’t build.
[https://stackoverflow.com/questions/69665222/node-js-17-0-1-gatsby-error-digital-envelope-routinesunsupported-err-os](https://stackoverflow.com/questions/69665222/node-js-17-0-1-gatsby-error-digital-envelope-routinesunsupported-err-os)

hmalpani, 2024-02-26:
Hello @krupis 
Please refer to the following comment and check if it helps.
https://github.com/espressif/esp-idf/issues/10344#issuecomment-1347782776

krupis, 2024-02-26:
> Hello @krupis Please refer to the following comment and check if it helps. [#10344 (comment)](https://github.com/espressif/esp-idf/issues/10344#issuecomment-1347782776)

I have installed node version 10.19.0 as suggested using the nvm with following command:
```
C:\Users\my_user>nvm install 10.19.0
Downloading node.js version 10.19.0 (64-bit)...
Complete
Creating C:\Users\my_user\AppData\Roaming\nvm\temp

Downloading npm version 6.13.4... Complete
Installing npm v6.13.4...

Installation complete. If you want to use this version, type

nvm use 10.19.0
```

Then I did:
```nvm use 10.19.0``` to ensure the correct node version is being used to compile the files.

I have tried to run ```npm run build``` again and it worked:

```
PS C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo> npm run build 

> web-demo@0.1.0 build C:\Users\my_user\esp5\esp-idf\examples\protocols\http_server\restful_server\front\web-demo
> vue-cli-service build

|  Building for production...

 WARNING  Compiled with 3 warnings                                                                                                                               11:32:18 AM

 warning  

asset size limit: The following asset(s) exceed the recommended size limit (244 KiB).
This can impact web performance.
Assets: 
  js/chunk-vendors.d1061d71.js (271 KiB)

 warning

entrypoint size limit: The following entrypoint(s) combined asset size exceeds the recommended limit (244 KiB). This can impact web performance.
Entrypoints:
  app (430 KiB)
      css/chunk-vendors.f77e647c.css
      js/chunk-vendors.d1061d71.js
      js/app.1b14630b.js

 warning

webpack performance recommendations:
You can limit the size of your bundles by using import() or require.ensure to lazy load some parts of your application.
For more info visit https://webpack.js.org/guides/code-splitting/

  File                                   Size              Gzipped

  dist\js\chunk-vendors.d1061d71.js      271.28 KiB        85.83 KiB
  dist\js\app.1b14630b.js                7.04 KiB          2.66 KiB
  dist\css\chunk-vendors.f77e647c.css    151.68 KiB        19.56 KiB

  Images and other types of assets omitted.

 DONE  Build complete. The dist directory is ready to be deployed.
 INFO  Check out deployment instructions at https://cli.vuejs.org/guide/deployment.html
```

Thank you very much for help! Additionally, I have a few questions:

1. Why the web files need to be compiled? Is there any advantage? 
I have previously tried to embed the index.html, styles.css and scripts.js files as following:

```
extern const char index_start[] asm("_binary_index_html_start");
extern const char index_end[] asm("_binary_index_html_end");
extern const char styles_start[] asm("_binary_styles_css_start");
extern const char styles_end[] asm("_binary_styles_css_end");
extern const char scripts_start[] asm("_binary_scripts_js_start");
extern const char scripts_end[] asm("_binary_scripts_js_end");

static esp_err_t web_root_hdlr(httpd_req_t* req) {
    const uint32_t len = index_end - index_start;
    TraceInf("ROOT");
    httpd_resp_set_type(req, TYPE_TXT_HTML);
    httpd_resp_send(req, index_start, len);
    return ESP_OK;
}

static esp_err_t web_styles_hdlr(httpd_req_t* req) {
    const uint32_t len = styles_end - styles_start;
    TraceInf("STYLES");
    httpd_resp_set_type(req, TYPE_TXT_CSS);
    httpd_resp_send(req, styles_start, len);
    return ESP_OK;
}

static esp_err_t web_scripts_hdlr(httpd_req_t* req) {
    const uint32_t len = scripts_end - scripts_start;
    TraceInf("SCRIPTS");
    httpd_resp_set_type(req, TYPE_TXT_JSCRIPT);
    httpd_resp_send(req, scripts_start, len);
    return ESP_OK;
}

static const httpd_uri_t web_handlers[] = {
    {.uri = "/",                .method = HTTP_GET,  .handler = web_root_hdlr,             .user_ctx = &web_context},
    { .uri = "/styles.css",             .method = HTTP_GET,  .handler = web_styles_hdlr,           .user_ctx = &web_context},
    { .uri = "/scripts.js",             .method = HTTP_GET,  .handler = web_scripts_hdlr,          .user_ctx = &web_context},
    };

static void web_register_handlers(httpd_handle_t server) {
    uint8_t cnt = ELEM_IN_ARRAY(web_handlers);
    TraceInf("Registering handlers. Total count %u", cnt);
    for (uint8_t i = 0; i < cnt; i++) {
        esp_err_t err = httpd_register_uri_handler(server, &web_handlers[i]);
        if ((err != ESP_OK) && (err != ESP_ERR_HTTPD_HANDLER_EXISTS)) {
            TraceWrn("Failed to register %u|%s handler, reason [%s]", i, web_handlers[i].uri, esp_err_to_name(err));
            break;
        }
    }
    httpd_register_err_handler(server, HTTPD_404_NOT_FOUND, http_404_error_handler);
    TraceInf("Handlers registered successfully");
}

```

And then in CMakeLists.txt I can do the following:
```
set(embed_files index.html styles.css scripts.js)
idf_component_register(EMBED_FILES ${embed_files})
```

How does exactly compiling web files work and why is it used here? Is it purely for convenience purposes?  Would using the above method work for framework such as Vue.js where I embed each?

hmalpani, 2024-02-27:
Thanks for the update. Closing this issue as it seems to be resolved. Please feel free to reopen in case it still persists.

closed 2024-02-27 by hmalpani

krupis, 2024-02-27:
I think it would be wise to update the example README to include the dependency on the old node version as there seems to be quite a confusion. There are multiple other threads regarding the same issue and everything could be fixed by adding a single line to say that this example needs version 10.19.0

hmalpani, 2024-02-28:
Yes indeed. I have created an MR internally for adding the note in README file.