espressif/esp-idf #13231: BLE mesh: Cannot send the message from server model, in BLE mesh provisonor example

ankit-thealchemist, 2024-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP idf , tag v5.1.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Device should send the message from the server model without any error.

### What is the actual behavior?

I am getting the error message, `BLE_MESH: Invalid NetKeyIndex 0x0000`

### Steps to reproduce.

1. Step: Open the provisionor example
2. Step: Add any server model in the root model. 
3. Step: Try to broadcast the messge, by applying the remote add = 0xFFFF
4. Device should not report the net key index error. 
5. Attached the full code 
`
/* main.c - Application main entry point */

/*
 * SPDX-FileCopyrightText: 2017 Intel Corporation
 * SPDX-FileContributor: 2018-2021 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#include "esp_log.h"
#include "nvs_flash.h"

#include "esp_ble_mesh_defs.h"
#include "esp_ble_mesh_common_api.h"
#include "esp_ble_mesh_provisioning_api.h"
#include "esp_ble_mesh_networking_api.h"
#include "esp_ble_mesh_config_model_api.h"
#include "esp_ble_mesh_generic_model_api.h"

#include "ble_mesh_example_init.h"

#define TAG "EXAMPLE"

#define LED_OFF             0x0
#define LED_ON              0x1

#define CID_ESP             0x02E5

[... 716 lines cut ...]
I (397) spi_flash: flash io: dio
W (401) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (414) sleep: Configure to isolate all GPIO pins in sleep state
I (421) sleep: Enable automatic switching of GPIO sleep configuration
I (428) coexist: coex firmware version: b6d5e8c
I (434) coexist: coexist rom version 9387209
I (439) app_start: Starting scheduler on CPU0
I (443) main_task: Started on CPU0
I (443) main_task: Calling app_main()
I (443) EXAMPLE: Initializing...
I (453) BLE_INIT: BT controller compile version [59725b5]
I (453) BLE_INIT: Bluetooth MAC: 7c:df:a1:61:c8:1e

I (463) phy_init: phy_version 1130,b4e4b80,Sep  5 2023,11:09:30
I (573) EXAMPLE: ESP_BLE_MESH_PROVISIONER_SET_DEV_UUID_MATCH_COMP_EVT, err_code 0
I (583) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_ENABLE_COMP_EVT, err_code 0
I (583) EXAMPLE: ESP_BLE_MESH_PROVISIONER_ADD_LOCAL_APP_KEY_COMP_EVT, err_code 0
I (593) EXAMPLE: ESP_BLE_MESH_PROVISIONER_BIND_APP_KEY_TO_MODEL_COMP_EVT, err_code 0
I (603) EXAMPLE: ESP_BLE_MESH_PROVISIONER_BIND_APP_KEY_TO_MODEL_COMP_EVT, err_code 0
I (603) EXAMPLE: BLE Mesh Provisioner initialized
I (10613) EXAMPLE: Sending the data to all nodes
E (10613) BLE_MESH: Invalid NetKeyIndex 0x0000
I (20613) EXAMPLE: Sending the data to all nodes
E (20613) BLE_MESH: Invalid NetKeyIndex 0x0000
```

ankit-thealchemist, 2024-02-22:
My question is could we use server model on provisionor? if not then why?  As far as I know about the ble mesh, there is no limitaion on the provisioner to do not use the server mode.

lebrislo, 2024-02-26:
Hi,

I'm facing the same issue. Did you find any solution ? 

Regards

ankit-thealchemist, 2024-02-26:
> Hi,
> 
> I'm facing the same issue. Did you find any solution ?
> 
> Regards

@lebrislo no, not for the problem. I cannot register the server model on the provisionor, and its quite strange. Suppose, provisionor is just a part of mesh with the sensor, then the whole functionallity became halted. 

I solved my problem, as I am able to find message which needed to transmitted in the client model. So I replace the server model with the client model, hence my problem solved. 
Let me know what you need to be transmitted, may be I can help.

lebrislo, 2024-02-26:
I have two devices working as light controler, they both support generic on/off servel model. On each device I have a button that toggle the on/off state and send `ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_STATUS`.

The first one device is a simple node and the second is the mesh provisioner. When I press the button on the simple node device I get the expected behavior. But when I perform the same action on the provisioner I get `BLE_MESH: Invalid NetKeyIndex 0x0000`

To send the status I'm using this code :

---

#define PRIMARY_NET_KEY_INDEX 0x0000
  #define PRIMARY_APP_KEY_INDEX   0x0000
  
  static void setCommonMessage(esp_ble_mesh_msg_ctx_t *ctx, uint16_t unicast_dest)
  {
      ctx->addr = unicast_dest;
      ctx->net_idx = PRIMARY_NET_KEY_INDEX;
      ctx->app_idx = PRIMARY_APP_KEY_INDEX;
      ctx->send_rel = false;
      ctx->send_ttl = 7;
  }
  
  setCommonMessage(&ctx, ESP_BLE_MESH_ADDR_ALL_NODES);
          err = esp_ble_mesh_server_model_send_msg(onOffServer0.model, &ctx,
                                                   ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_STATUS,
                                                   sizeof(p_onOff), (uint8_t *)&p_onOff);
          if (err != ESP_OK)
          {
              LOG_ERROR("Failed to send generic onoff status message %s", esp_err_to_name(err));
          }

---

I saw in the esp-idf this function : 

---
  uint8_t bt_mesh_get_device_role(struct bt_mesh_model *model, bool srv_send)
  {
      bt_mesh_client_user_data_t *client = NULL;
  
    if (srv_send) {
        BT_DBG("Message is sent by a server model");
        return NODE;
    }

    if (!model || !model->user_data) {
        BT_ERR("%s, Invalid parameter", __func__);
        return ROLE_NVAL;
    }
  
      client = (bt_mesh_client_user_data_t *)model->user_data;
  
      return client->msg_role;
  }
---

I think this is the one that cause my problem. Since I use the generic on/off server model, this function automatically set the device role as `ROLE_NODE`. And so, the function `bt_mesh_tx_netkey_get` cannot retreive my NetKey

ankit-thealchemist, 2024-02-26:
@lebrislo Yeah! the problem is provisioner cannot have the server model in their element and its quite strange. 
What I didn't try is, my server model was in the root model, try implement the server model in the other element, may it can recognize.