espressif/esp-idf #13232: menuconfig Access violation with Python 3.12

Addenri, 2024-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

run menuconfig and setup project

### What is the actual behavior?

get "Access violation" error while trying build "menuconfig" target.

### Steps to reproduce.

1. install git, Python 3.12
2. clone git repo ESP-IDF (branch master or v5.2)
3. run install script (install.ps1)
4. export environment variables (export.ps1)
5. copy example project
6. run "idf.py menuconfig" in project's directory

### Build or installation Logs.

```plain
Executing action: menuconfig
Running ninja in directory C:\Users\Andrii\esp\hello_world\build
Executing "ninja menuconfig"...
[0/1] Re-running CMake...-- ccache will be used for faster recompilation
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/Andrii/esp/hello_world/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- App "hello_world" version: 1
-- Adding linker script C:/Users/Andrii/esp/hello_world/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/Andrii/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Users/Andrii/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/Andrii/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/Andrii/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/Andrii/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/Andrii/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/Andrii/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/Andrii/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_h
ttp_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs f
reertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/Andrii/esp/esp-idf/components/app_trace C:/Users/Andrii/esp/esp-idf/components/app_update C:/Users/Andrii/esp/esp-idf/components/bootloader C:/Users/Andrii/esp/esp-idf/components/bootloader_support C:/Users/Andrii/esp/esp-idf/components/bt C:/Users/Andrii/esp/esp-idf/components/cmock C:/Users/Andrii/esp/esp-idf/components/console C:/Users/Andrii/esp/esp-idf/components/cxx C:/Users/Andrii/esp/esp-idf/components/driver C:/Users/Andri
i/esp/esp-idf/components/efuse C:/Users/Andrii/esp/esp-idf/components/esp-tls C:/Users/Andrii/esp/esp-idf/components/esp_adc C:/Users/Andrii/esp/esp-idf/components/esp_app_format C:/Users/Andrii/esp/esp-idf/components/esp_bootloader
_format C:/Users/Andrii/esp/esp-idf/components/esp_coex C:/Users/Andrii/esp/esp-idf/components/esp_common C:/Users/Andrii/esp/esp-idf/components/esp_eth C:/Users/Andrii/esp/esp-idf/components/esp_event C:/Users/Andrii/esp/esp-idf/co
mponents/esp_gdbstub C:/Users/Andrii/esp/esp-idf/components/esp_hid C:/Users/Andrii/esp/esp-idf/components/esp_http_client C:/Users/Andrii/esp/esp-idf/components/esp_http_server C:/Users/Andrii/esp/esp-idf/components/esp_https_ota C
:/Users/Andrii/esp/esp-idf/components/esp_https_server C:/Users/Andrii/esp/esp-idf/components/esp_hw_support C:/Users/Andrii/esp/esp-idf/components/esp_lcd C:/Users/Andrii/esp/esp-idf/components/esp_local_ctrl C:/Users/Andrii/esp/es
p-idf/components/esp_mm C:/Users/Andrii/esp/esp-idf/components/esp_netif C:/Users/Andrii/esp/esp-idf/components/esp_netif_stack C:/Users/Andrii/esp/esp-idf/components/esp_partition C:/Users/Andrii/esp/esp-idf/components/esp_phy C:/U
[... 16 lines cut ...]

FAILED: CMakeFiles/menuconfig C:/Users/Andrii/esp/hello_world/build/CMakeFiles/menuconfig
cmd.exe /C "cd /D C:\Users\Andrii\esp\hello_world\build && C:\Users\Andrii\.espressif\python_env\idf5.2_py3.12_env\Scripts\python.exe C:/Users/Andrii/esp/esp-idf/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon 
--env-file C:/Users/Andrii/esp/hello_world/build/config.env && C:\Users\Andrii\.espressif\python_env\idf5.2_py3.12_env\Scripts\python.exe -m kconfgen --list-separator=semicolon --kconfig C:/Users/Andrii/esp/esp-idf/Kconfig --sdkconf
ig-rename C:/Users/Andrii/esp/esp-idf/sdkconfig.rename --config C:/Users/Andrii/esp/hello_world/sdkconfig --env-file C:/Users/Andrii/esp/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_TOOLCHAIN=gcc --env IDF_ENV_FPGA=
 --env IDF_INIT_VERSION=5.2.0 --dont-write-deprecated --output config C:/Users/Andrii/esp/hello_world/sdkconfig && C:\Users\Andrii\.espressif\python_env\idf5.2_py3.12_env\Scripts\python.exe C:/Users/Andrii/esp/esp-idf/tools/check_te
rm.py && C:\Users\Andrii\.espressif\tools\cmake\3.24.0\bin\cmake.exe -E env COMPONENT_KCONFIGS_SOURCE_FILE=C:/Users/Andrii/esp/hello_world/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=C:/Users/Andrii/esp/hello_world/bu
ild/kconfigs_projbuild.in KCONFIG_CONFIG=C:/Users/Andrii/esp/hello_world/sdkconfig IDF_TARGET=esp32 IDF_TOOLCHAIN=gcc IDF_ENV_FPGA= IDF_INIT_VERSION=5.2.0 C:/Users/Andrii/.espressif/python_env/idf5.2_py3.12_env/Scripts/python.exe -m
 menuconfig C:/Users/Andrii/esp/esp-idf/Kconfig && C:\Users\Andrii\.espressif\python_env\idf5.2_py3.12_env\Scripts\python.exe -m kconfgen --list-separator=semicolon --kconfig C:/Users/Andrii/esp/esp-idf/Kconfig --sdkconfig-rename C:
/Users/Andrii/esp/esp-idf/sdkconfig.rename --config C:/Users/Andrii/esp/hello_world/sdkconfig --env-file C:/Users/Andrii/esp/hello_world/build/config.env --env IDF_TARGET=esp32 --env IDF_TOOLCHAIN=gcc --env IDF_ENV_FPGA= --env IDF_INIT_VERSION=5.2.0 --output config C:/Users/Andrii/esp/hello_world/sdkconfig"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information.

Reproduced on Windows 10 and 11 with Python 3.12.2. Can't reproduce with older Python versions, but need to keep it updated. Can be reproduced not only with "hello_world" example, same as others. If try to build "menuconfig" target in CLion, got:
`====================[ Build | menuconfig | Debug ]==============================
"C:\Program Files\JetBrains\CLion 2023.2\bin\cmake\win\x64\bin\cmake.exe" --build C:\Users\Andrii\esp\hello_world\cmake-build-debug --target menuconfig -j 14
[0/1] cmd.exe /C "cd /D C:\Users\Andrii\esp\hello_world\cmake-build-debug && python C:/Users/Andrii/esp/esp-idf/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file C:/Users/Andrii/esp/hello_world/cmake-build-debug/config.env && python -m kconfgen --list-separator=semicolon --kconfig C:/Users/Andrii/esp/esp-idf/Kconfig --sdkconfig-rename C:/Users/Andrii/esp/esp-idf/sdkconfig.rename --config C:/Users/Andrii/esp/hello_world/sdkconfig --env-file C:/Users/Andrii/esp/hello_world/cmake-build-debug/config.env --env IDF_TARGET=esp32 --env IDF_TOOLCHAIN=gcc --env IDF_ENV_FPGA= --env IDF_INIT_VERSION=5.2.0 --dont-write-deprecated --output config C:/Users/Andrii/esp/hello_world/sdkconfig && python C:/Users/Andrii/esp/esp-idf/tools/check_term.py && "C:\Program Files\JetBrains\CLion 2023.2\bin\cmake\win\x64\bin\cmake.exe" -E env COMPONENT_KCONFIGS_SOURCE_FILE=C:/Users/Andrii/esp/hello_world/cmake-build-debug/kconfigs.in COMPONENT_KCONFIGS_PROJBU [... cut ...] i/esp/esp-idf/Kconfig && python -m kconfgen --list-separator=semicolon --kconfig C:/Users/Andrii/esp/esp-idf/Kconfig --sdkconfig-rename C:/Users/Andrii/esp/esp-idf/sdkconfig.rename --config C:/Users/Andrii/esp/hello_world/sdkconfig --env-file C:/Users/Andrii/esp/hello_world/cmake-build-debug/config.env --env IDF_TARGET=esp32 --env IDF_TOOLCHAIN=gcc --env IDF_ENV_FPGA= --env IDF_INIT_VERSION=5.2.0 --output config C:/Users/Andrii/esp/hello_world/sdkconfig"
ninja: build stopped: subcommand failed.`

Honza0297, 2024-02-23:
Hello @Addenri, thank you for the report! 
I was able to reproduce your issue with ESP-IDF 5.2 and Python 3.12.2. 
Quick workaround is to remove Python 3.12 from PATH and use older version (3.11.7 is working). 
I will dig deeper into the problem and come up with better solution next week.

Honza0297, 2024-02-26:
After a bit of poking things around, I found the culprit, windows curses, used by kconfiglib/menuconfig: https://github.com/zephyrproject-rtos/windows-curses/issues/48
Unfortunately, there is nothing we can do now as neither kconfiglib nor windows-curses are packages maintained by us.  I would suggest sticking to older versions of Python (=<3.11)

Honza0297, 2024-03-15:
Now, it should work in versions 5.2 and below, where the fix is present directly in ESP-IDF.
Master branch needs the latest version of [esp-idf-kconfig](https://github.com/espressif/esp-idf-kconfig) in order to work properly for now. It can be installed as a local package from the linked git repository.
This issue will be closed shortly. 
@Addenri Thanks again for reporting and feel free to reopen if the problem prevails.

closed 2024-03-15 by dobairoland

dobairoland, 2024-03-15:
> Master branch needs the latest version of [esp-idf-kconfig](https://github.com/espressif/esp-idf-kconfig) in order to work properly for now. It can be installed as a local package from the linked git repository.

It is enough to re-run the install scripts now and it will install `esp-idf-kconfig>=2.2.0` which contains the fix.

ARNik, 2024-03-22:
> Quick workaround is to remove Python 3.12 from PATH and use older version (3.11.7 is working).

Hello!
How can I implement this workaround? Now install scripts (even old versions) use 3.12 Python as default. And menuconfig works in this isolated invironment. It does not depend on python version installed in the system.
Do I need to make changes in install scripts somewere?

Honza0297, 2024-03-25:
Hello @ARNik ! 
Have you installed your ESP-IDF via the installer or from github repository?

ARNik, 2024-03-25:
> Hello @ARNik ! Have you installed your ESP-IDF via the installer or from github repository?

I cloned the repo and checked out different versions usibn "git worktree".  Then I tryed install.bat script from different esp-idf versions deleting .espressif folder in between.

dobairoland, 2024-03-25:
You can see the linked fixes were published a week ago so they couldn't make into any of the releases released earlier.  You can use the release branches (release/v5.*) until the next release. Or if you insist on using a tagged version then downgrade your system Python to 3.11.

ARNik, 2024-03-25:
> You can use the release branches (release/v5.*) until the next release.
Yes, I can. But I think something is completely wrong that I can't build none of the previous releases.