espressif/esp-idf #13233: `iram0_0_seg' overflowed by 10488 bytes after adding TCP server code in existing code.

shameem1423, 2024-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP IDF 5.0

### Espressif SoC revision.

ESP32-PICO-MINI-02

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

Battery

### What is the expected behavior?

Build successfully and run as normal. 

### What is the actual behavior?

I was getting an error message as :- 
    c:/espressif3/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: SenseH_FW_Esp32.elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/espressif3/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/espressif3/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 10448 bytes

After receiving this error changed the iram0_0_seg len from 0x20000 to 0x25000

/* IRAM for PRO cpu. Not sure if happy with this, this is MMU area... */
  iram0_0_seg (RX) : org = 0x40080000, len = 0x20000
  **TO**
/* IRAM for PRO cpu. Not sure if happy with this, this is MMU area... */
  iram0_0_seg (RX) : org = 0x40080000, len = 0x25000

After this change its code builded successfully after flashing getting run time error (device is booting in loop) 
[image]

### Steps to reproduce.

Due limitations can't attach the code.

### Debug Logs.

```plain
I (28) boot: ESP-IDF -128-NOTFOUND 2nd stage bootloader
I (28) boot: compile time 17:13:17
I (28) boot: Multicore bootloader
I (32) boot: chip revision: v3.0
I (36) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 8MB
[... 110 lines cut ...]
E (641) esp_image: Segment 5 0x400879a8-0x400a28d0 invalid: bad load address range
0x400879a8: r_lld_evt_schedule_next_instant_hack at ??:?

E (646) boot: Factory app partition is not bootable
E (652) esp_image: image at 0x170000 has invalid magic byte (nothing flashed here?)
E (660) boot: OTA app partition slot 0 is not bootable
E (666) esp_image: image at 0x2d0000 has invalid magic byte (nothing flashed here?)
E (674) boot: OTA app partition slot 1 is not bootable
E (680) boot: No bootable app partitions in the partition table
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6992
load:0x40078000,len:15512
load:0x40080400,len:3864
entry 0x40080650
0x40080400: _init at ??:?
```

### More Information.

The one was developed the code in his system its working fine. I have tried to do it in 3 systems and all device showing same error.

shameem1423, 2024-02-28:
euripedesrocha do you have any thought on this.

ESP-Marius, 2024-03-06:
Hi,

You cannot simply just change the linkerscript like that unless you are 100% sure you know what you are doing. Those numbers have been chosen for a reason.

For info on how to optimize your application for IRAM usage I suggest taking a look at https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32/api-guides/performance/ram-usage.html#optimize-iram-usage

closed 2024-04-02 by euripedesrocha