espressif/esp-idf #13234: Documentation for usage of opt_val param in esp_netif_dhcps_option function

otaxhu, 2024-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

What struct should I use if I want to obtain for example ESP_NETIF_DOMAIN_NAME_SERVER? or ESP_NETIF_ROUTER_SOLICITATION_ADDRESS?:
```c
esp_err_t esp_netif_dhcps_option(esp_netif_t *esp_netif,
                                 esp_netif_dhcp_option_mode_t opt_op,
                                 esp_netif_dhcp_option_id_t opt_id,
                                 void *opt_val, // What struct or type should I use for dereference this pointer?
                                 uint32_t opt_len);
```
I'm just asking for a example in the documentation, if there is already one, please comment the link :)

david-cermak, 2024-03-11:
Hi @otaxhu 

I'm sorry, this is not properly documented. You can check the usage in the examples here:

 https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/examples/mesh/ip_internal_network/main/mesh_netif.c#L68-L70

Most of the dhcp server's types used are defined here:

https://github.com/espressif/esp-idf/blob/60a2bf6a68867fd468a250966230a067f8d97ea2/components/lwip/include/apps/dhcpserver/dhcpserver.h#L79-L84

Please keep this open, I'll update the docs and close this issue.

david-cermak, 2025-01-24:
Documentation updated:
* [about DHCP options](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif_programming.html#configure-dhcp-options)
* about [esp_netif_dhcpc_option](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif_programming.html#_CPPv422esp_netif_dhcpc_optionP11esp_netif_t28esp_netif_dhcp_option_mode_t26esp_netif_dhcp_option_id_tPv8uint32_t)
* about [esp_netif_dhcps_option](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif_programming.html#_CPPv422esp_netif_dhcps_optionP11esp_netif_t28esp_netif_dhcp_option_mode_t26esp_netif_dhcp_option_id_tPv8uint32_t)

Please reopen if you think it should be documented better.

closed 2025-01-24 by david-cermak