espressif/esp-idf #13235: assert failed: esp_openthread_task_queue_post esp_openthread_task_queue.c (ret == sizeof(val)) on heavy openthread activity

rretanubun, 2024-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Setup
esp-idf: release 4.4.6
openthread component: https://github.com/espressif/esp-idf/commit/68ce4f1404632fa775a97f555dc3daab98bd67d3 - back when everything was open sourced.

### Hardware
ESP32 as Host Processor, EFR32 as Radio CoProcessor, running in Thread Border Router Mode

### Trigger for issue
Under heavy thread activity for example
1. A TBR with 25+ SRP clients where 24+ are sleepy devices
2. A TBR with a lot of on-mesh joiners joining and leaving

### Analysis done so far
- The `assert` fires because the `write` call returns -1 with `errno` set to 9 (EBADF)
- The `s_task_queue_event_fd` itself is valid, and when trying to get the fd flags using `int fdFlags = fcntl(s_task_queue_event_fd, F_GETFL);` this always returns `-1` even when things are working, so I don't think this is the root cause.
- I am aware of the renames https://github.com/espressif/esp-idf/commits/8672467d3628a30a03c7b9bc964fcd05e335d289/components/openthread/port/esp_openthread_task_queue.c?browsing_rename_history=true&new_path=components/openthread/src/esp_openthread_task_queue.c&original_branch=4d90eedb6ef32ff949c6f298a3d845f9f07bac92 when the same function resurfaces as open source, however the file has been reworked significantly and I am not sure I can transplant the new code into my codebase.

### Question
- Is such an issue seen elsewhere under different triggers?
- Is this fixed already? if so, any hint of a small fix I can backport to my codebase? I am not sure if I can just pickup openthread from current esp-idf since that seems to already be targeted to esp-idf-5 and I am still on esp-idf-4

Thank you for everyone's time

chshu, 2024-02-23:
@rretanubun The Thread BR implementation in IDF v4.4 is not ready for production, and out of maintenance for more than 2 years. OpenThread is also an old version there, which includes a critical security issue [(link)](https://github.com/openthread/openthread/security/advisories/GHSA-vr3r-363g-72j9), all vendors need to upgrade the OT to later versions for production.

Thread and BR features are officially supported since IDF v5.1, could you switch your project to IDF v5.1.2 version? 

You may also check the [esp-thread-br](https://github.com/espressif/esp-thread-br) SDK, it supports more production-ready features like RCP update and coexistence.

chshu, 2024-02-24:
@rretanubun As long as the [RCP API version](https://github.com/openthread/openthread/blob/f0b6fcea6ef77c9a54ab11767593f9a8798e3662/src/lib/spinel/radio_spinel.cpp#L363) matches, the ESP Thread BR SDK (Networking features) could work with any standard OT RCP, no matter the SoC vendors.

Only some platform specific features like RCP update and coexistence don't work with `non-espressif` RCP, since these depend on HW features.

chshu, 2024-02-27:
@rretanubun Regarding 25+ Thread devices, what kind of info are you looking for? I don't see any issue to host 25+ Thread devices in a Thread network under ESP Thread BR. We do have some internal testbeds testing large scale topology.

rretanubun, 2024-03-06:
@chshu : Apologies for the delay in responding. I am looking for advice on how to tune the system to:
1. Given a device mix of 1xTBR and 25+ non-TBR devices (2x TME and 24+ MTD)
2. Have the TBR Maintain SRP client registrations & services for 25+ devices
3. Have the TBR Maintain mDNS client entry for the same devices to publish via OT-BR ADPROXY
3. Have the TBR Process and relay CoAP messages from thread end device to a CoAP client on the Lan side (say once-per-second-per-device)
4. Have the TBR do this while sustaining GET request on http every 15-20 ms over ethernet (or wifi)

Given the limited number of internal and external memory available on esp32

Out of curiosity, how large is your `internal testbed for testing large scale topology` and what devices are used?
I am asking because if some of them are publicly available `devkits` we may consider adding them to our testbed for `device diversification`.

rretanubun, 2024-03-08:
Thanks for the guidance @chshu! I have shared this with the team and will be in touch if we have further questions.

chshu, 2024-03-08:
@rretanubun Sure, closing the current issue.

closed 2024-03-08 by chshu