espressif/esp-idf #13237: Bug in IRAM in the new SDK v5.2 whem using BLE even with the NIMBLE at libbtdm_app.a using 30981 bytes

filzek, 2024-02-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-37-g93ea06fe3e

### Espressif SoC revision.

esp32 3.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

wrover ie external memory psram enable

### Power Supply used.

USB

### What is the expected behavior?

Bug in IRAM in the new SDK v5.2 whem using BLE even with the NIMBLE at libbtdm_app.a using 30981 bytes

It is expect to use less than 15k that is a lot of IRAM, so whem using the bluetooth it uses ALL The IRAM by itself and we cant simple build a little more compelx firmware.

it shall reduce this extreme BLE/Wifi usage.

### What is the actual behavior?

Uses extreme IRAM size
libbtdm_app.a        30981

### Steps to reproduce.

libbtdm_app.a        30981

filzek, 2024-02-22:
Way to add more IRAM is to add the DRAM part as IRAM at menuconfig
CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM=y

filzek, 2024-02-22:
Its not necessary to use the Bootloader to manage the memory to bood the application, this somehow is not viable, so, the firmware build could use the same signature to be BOOTABLE and add an wrapper to later add the memory to the pool, its simple by doing a post memory pool setup in the RUNNING APP environment. 

How could Espressif do it?

Create the memory pool to use the IRAM place in a ADD way besides doind it at BOOT.

In this way the memory REMAP is possible in the fly.

Weijian-Espressif, 2024-02-26:
@filzek  We have a modification that can reduce IRAM by about 10K bytes in BLE only scenarios, the code has not been synchronized to github, please wait for a while

filzek, 2024-02-27:
> @filzek We have a modification that can reduce IRAM by about 10K bytes in BLE only scenarios, the code has not been synchronized to github, please wait for a while

This adjustment should address the primary concerns faced by users of ESP32 and other MCUs, as memory optimization is currently a major issue.

We are pleased to report that in SDK v5.2, there are no issues with Nimble BLE and WiFi registration. Our comprehensive testing failed to replicate any of the bugs found in SDK v4. The team has made significant efforts to resolve the predominant issues with ESP32, especially those related to register malfunctions during system halts, and these efforts have been highly successful!

Weijian-Espressif, 2024-02-29:
@filzek Code has been synced to github, https://github.com/espressif/esp-idf/commit/e265484084081f280322f51f3f3b4dd61afc30ee

filzek, 2024-03-01:
so, we can use the Wifi and Nimble BLE with 10k extra IRAM using the v5.2 brunch?

AxelLin, 2024-03-05:
> so, we can use the Wifi and Nimble BLE with 10k extra IRAM using the v5.2 brunch?

This fix is not yet in v5.2 branch. @Weijian-Espressif

Weijian-Espressif, 2024-03-05:
@AxelLin @filzek v4.4, v5.0, v5.1 and the master branch support this feature, v5.2 code is not yet synchronized

filzek, 2024-03-08:
fixed in 5.2.1

closed 2024-03-08 by filzek