espressif/esp-idf #13238: Esp-idf not working in my laptop

moaz-azam, 2024-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

PS D:\esp_idf workplace\project-namebgfbgfnhgnh> idf.py build
Executing action: all (aliases: build)
Running ninja in directory 'D:\esp_idf workplace\project-namebgfbgfnhgnh\build'
Executing "ninja all"...
[1/6] Linking CXX executable project-namebgfbgfnhgnh.elfFAILED: project-namebgfbgfnhgnh.elf
cmd.exe /C "cd . && C:\esp\tools\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  -g -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 "-Wl,--Map=D:/esp_idf workplace/project-namebgfbgfnhgnh/build/project-namebgfbgfnhgnh.map" -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\project-namebgfbgfnhgnh.elf.rsp -o project-namebgfbgfnhgnh.elf  && cd ."
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lkernel32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -luser32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lgdi32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lwinspool: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lshell32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lole32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -loleaut32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -luuid: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lcomdlg32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -ladvapi32: No such file or directory
collect2.exe: error: ld returned 1 exit status
[1/1] cmd.exe /C "cd /D "D:\esp_idf workplace\project-name.../project-namebgfbgfnhgnh/build/bootloader/bootloader.bin""Bootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the D:\esp_idf workplace\project-namebgfbgfnhgnh\build\log\idf_py_stderr_output_8944 and D:\esp_idf workplace\project-namebgfbgfnhgnh\build\log\idf_py_stdout_output_8944

### What is the actual behavior?

I install vs code extension. configure it for 5.2 version it could not find environment variable. build failed due to it can't find files which are associated with environmental variable.
then i tried online installer installation failed again with error code 101 than i tried offline installer it installs correctly but now when i build project it is showing above error tools not found.   

### Steps to reproduce.

step 1) Install vs code extension environmental variables not found project build failed uninstall that extension than tried different version same error.
step 2) online installation failed due to some error.
step 3) offline installation successful but build fail due to some tools not found. 

### Build or installation Logs.

```plain
[1/6] Linking CXX executable project-namebgfbgfnhgnh.elfFAILED: project-namebgfbgfnhgnh.elf
cmd.exe /C "cd . && C:\esp\tools\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  -g -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 "-Wl,--Map=D:/esp_idf workplace/project-namebgfbgfnhgnh/build/project-namebgfbgfnhgnh.map" -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\project-namebgfbgfnhgnh.elf.rsp -o project-namebgfbgfnhgnh.elf  && cd ."
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lkernel32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -luser32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lgdi32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lwinspool: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lshell32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lole32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -loleaut32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -luuid: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lcomdlg32: No such file or directory
c:/esp/tools/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -ladvapi32: No such file or directory
collect2.exe: error: ld returned 1 exit status
[1/1] cmd.exe /C "cd /D "D:\esp_idf workplace\project-name.../project-namebgfbgfnhgnh/build/bootloader/bootloader.bin""Bootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the D:\esp_idf workplace\project-namebgfbgfnhgnh\build\log\idf_py_stderr_output_8944 and D:\esp_idf workplace\project-namebgfbgfnhgnh\build\log\idf_py_stdout_output_8944
```

moaz-azam, 2024-02-23:
[image]

brianignacio5, 2024-02-23:
About vscode extension, did you follow the [install tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) ?

dobairoland, 2024-02-23:
"esp_idf workplace" has a space in it. I'm not sure about it but that might be causing the problem. It would be worth to try another directory without a space.

moaz-azam, 2024-02-23:
vs code extension problem resolved by removing space in workspace folder.

in offline installer problem was to be maybe due to i created projects using 5.2 version from vs code extension and tried to built them with 5.1.2 version 
thanks @dobairoland

closed 2024-02-23 by moaz-azam