espressif/esp-idf #13239: 关于export.bat脚本工具建议

NewBlueshi, 2024-02-23:
### Is your feature request related to a problem?

在windwos 环境下 ESP-IDF 使用CLion 进行环境配置开发
使用export.bat 对环境变量进行配置时会很麻烦，因为bin 文件处于不同文件夹，要配置很多条PATH 变量
建议在export.bat脚本中直接生成 tools/bin目录，把其他的可执行文件都cp 到这个目录下
这样在配置环境变量或者其他操作都很方便，不用一直去执行export.bat文件就可以工作了

dobairoland, 2024-02-23:
The export.bat is for terminal use of ESP-IDF.

If you are interested in CLion the could give a shot for https://github.com/espressif/idf-clion-plugin.

There are other options as  well:
- https://github.com/espressif/idf-eclipse-plugin
- https://github.com/espressif/vscode-esp-idf-extension

NewBlueshi, 2024-02-23:
Thinks

closed 2024-02-23 by Alvin1Zhang