espressif/esp-idf #13240: ESP32S3 HID and CDC class both not working

fenilGhoghari, 2024-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I am using ESP32S3 and i want to run USB CDC and HID class both at a same time some of more details here,

I am using HID class to interface Keyboard and CDC class used for Firmware updating have you any idea how can I do this?

Regards,
Fenil

fenilGhoghari, 2024-02-23:
@songruo can you please help me?

fenilGhoghari, 2024-02-28:
@Dazza0  Thanks for the reply,
I am using 5.2 ESP-IDF.

- I couldn't found any error log as of now but CDC class is working
- Yes, i have USB Host stack interface with two device classes HID and CDC

I am using CDC for Firmware Update and HID to interface Keyboard

Dazza0, 2024-03-06:
@fenilGhoghari Closing this issue as I'm not understanding what is the problem you are encountering. Leaving some related links for your reference:

- If you need to the ESP32S3 to act as Host and talk to 2 device simultaneously, then this will require hub support (see https://github.com/espressif/esp-idf/issues/12554)
- Both HID and CDC Host classes are already supported (see the [USB Host examples](https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/api-reference/peripherals/usb_host.html#examples) for more details)

> I am using CDC for Firmware Update and HID to interface Keyboard

- USB has a separate [Device Firmware Update (DFU)](https://www.usb.org/sites/default/files/DFU_1.1.pdf) class. You may want to consider using this instead.

ProPablo, 2024-04-10:
I think I am encountering a similiar situation, my scenario is I am trying to emulate a HID device while also being able to print to console (CDC) for debugging. So the esp is acting as a client for both HID and Serial. Basically I am trying to follow [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb_device.html)
and implanting some of that code into [this template for HID](https://github.com/espressif/esp-idf/blob/5a40bb8746633477c07ff9a3e90016c37fa0dc0c/examples/peripherals/usb/device/tusb_hid/main/tusb_hid_example_main.c). Using [this]( https://github.com/espressif/esp-idf/blob/5a40bb8746633477c07ff9a3e90016c37fa0dc0c/examples/peripherals/usb/device/tusb_console/main/tusb_console_main.c#L57) for help. 

```c

#define TUSB_DESC_TOTAL_LEN (TUD_CONFIG_DESC_LEN + CFG_TUD_HID * TUD_HID_DESC_LEN)

/**
 * @brief HID report descriptor
 *
 * In this example we implement Keyboard + Mouse HID device,
 * so we must define both report descriptors
 */
const uint8_t hid_report_descriptor[] = {
    TUD_HID_REPORT_DESC_KEYBOARD(HID_REPORT_ID(HID_ITF_PROTOCOL_KEYBOARD)),
    TUD_HID_REPORT_DESC_MOUSE(HID_REPORT_ID(HID_ITF_PROTOCOL_MOUSE))};

/**
 * @brief String descriptor
 */
const char* hid_string_descriptor[5] = {
    // array of pointer to string descriptors
    (char[]){0x09, 0x04},  // 0: is supported language is English (0x0409)
    "TinyUSB",             // 1: Manufacturer
    "TinyUSB Device",      // 2: Product
    "123456",              // 3: Serials, should use chip ID
    "Example HID interface",  // 4: HID
};

/**
 * @brief Configuration descriptor
 *
 * This is a simple configuration descriptor that defines 1 configuration and 1 HID interface
 */
static const uint8_t hid_configuration_descriptor[] = {
    // Configuration number, interface count, string index, total length, attribute, power in mA
    TUD_CONFIG_DESCRIPTOR(1, 1, 0, TUSB_DESC_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),

    // Interface number, string index, boot protocol, report descriptor len, EP In address, size & polling interval
    TUD_HID_DESCRIPTOR(0, 4, false, sizeof(hid_report_descriptor), 0x81, 16, 10),
};

/*** Keyboard Initialisation ****/
void initialise_keyboard() {

    ESP_LOGI(TAG, "USB initialization");
    const tinyusb_config_t tusb_cfg = {
        .device_descriptor = NULL,
        .string_descriptor = hid_string_descriptor,
        .string_descriptor_count = sizeof(hid_string_descriptor) / sizeof(hid_string_descriptor[0]),
        .external_phy = false,
        .configuration_descriptor = hid_configuration_descriptor,
    };

    ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));
    ESP_LOGI(TAG, "USB initialization DONE");

}

void app_main(void)
{
    // Initialize button that will trigger HID reports
    const gpio_config_t boot_button_config = {
        .pin_bit_mask = BIT64(APP_BUTTON),
        .mode = GPIO_MODE_INPUT,
        .intr_type = GPIO_INTR_DISABLE,
        .pull_up_en = true,
        .pull_down_en = false,
    };
    ESP_ERROR_CHECK(gpio_config(&boot_button_config));

    initialise_keyboard();

    const tinyusb_config_cdcacm_t acm_cfg = {
        .usb_dev = TINYUSB_USBDEV_0,
        .cdc_port = TINYUSB_CDC_ACM_0,
        .rx_unread_buf_sz = 64,
        //.callback_rx = &tinyusb_cdc_rx_callback,
         .callback_rx = NULL,
        .callback_rx_wanted_char = NULL,
        .callback_line_state_changed = NULL,
        .callback_line_coding_changed = NULL};
    tusb_cdc_acm_init(&acm_cfg);
      esp_tusb_init_console(TINYUSB_CDC_ACM_0);
// And so on...
    }
```

I'm pretty sure this is possible (see [this youtube video](https://youtu.be/iOD-NOzgj7o?si=7BEktlSjJ2iYAFMe&t=250)), just need some solid pointers/ need docs on composite HID and CDC. 

Big thanks :green_heart:

roma-jam, 2024-04-10:
Hi @ProPablo,

For your goal, you may refer to the esp-idf usb device example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_composite_msc_serialdevice

The one thing you should do is to replace MSC interface with HID.

ProPablo, 2024-04-11:
Hi, 
Ive managed to solve this issue counterintuitively by sleeping on it, looking deeper at what you meant by `replace MSC interface with HID` and also looking more at [this](https://github.com/brendena/pico_device_and_host/blob/main/usb_descriptors.c). 

Ive had a look at that example already, my understanding is every member in `tinyusb_config_t` that is null is filled in by the framework. In the [tusb_composite_msc_serialdevice](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_composite_msc_serialdevice) example all members are null so I wasnt sure how to add cdc to my list of interfaces. 

Turns out in order to keep the HID interface definition, the configuration object needs to be specified as such:
```c
#define TUSB_DESC_TOTAL_LEN (TUD_CONFIG_DESC_LEN + TUD_HID_DESC_LEN + TUD_CDC_DESC_LEN)
const char* hid_string_descriptor[5] = {
    // array of pointer to string descriptors
    (char[]){0x09, 0x04},  // 0: is supported language is English (0x0409)
    "TinyUSB",             // 1: Manufacturer
    "TinyUSB Device",      // 2: Product
    "123456",              // 3: Serials, should use chip ID
    "Example HID interface",  // 4: HID
"My debugging CDC interface",  // 5: CDC
};

#define EPNUM_CDC_NOTIF   0x81
#define EPNUM_CDC_OUT     0x02
#define EPNUM_CDC_IN      0x82
#define EPNUM_HID   0x83

/************ THIS IS THE IMPORTANT PART *************/
/**
 * @brief Configuration descriptor
 *
 * This is a simple configuration descriptor that defines 1 configuration and 1 HID interface
 */
static const uint8_t hid_configuration_descriptor[] = {
    // Configuration number, interface count, string index, total length, attribute, power in mA
    TUD_CONFIG_DESCRIPTOR(1, 1, 0, TUSB_DESC_TOTAL_LEN, TUSB_DESC_CONFIG_ATT_REMOTE_WAKEUP, 100),

    // Interface number, string index, boot protocol, report descriptor len, EP In address, size & polling interval
    TUD_HID_DESCRIPTOR(0, 4, false, sizeof(hid_report_descriptor), EPNUM_HID, 16, 10),

    // Interface number, string index, EP notification address and size, EP data address (out, in) and size.
    TUD_CDC_DESCRIPTOR(1, 5, EPNUM_CDC_NOTIF, 8, EPNUM_CDC_OUT, EPNUM_CDC_IN, 64),

};

void initialise_keyboard()
{

    ESP_LOGI(TAG, "USB initialization");
    const tinyusb_config_t tusb_cfg = {
        .device_descriptor = NULL,
        .string_descriptor = hid_string_descriptor,
        .string_descriptor_count = sizeof(hid_string_descriptor) / sizeof(hid_string_descriptor[0]),
        .external_phy = false,
        .configuration_descriptor = hid_configuration_descriptor,
    };

    ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));
    ESP_LOGI(TAG, "USB initialization DONE");
}

```

Thank you for the help 💚

cnfatal, 2024-05-13:
There is a my project which use HID and CDC composite device: https://github.com/cnfatal/esp32-cdc-keyboard