espressif/esp-idf #13241: BLE MESH: Unable to send the time server status message to client due to the structure misalignment of esp_ble_mesh_state_change_time_status_t

ankit-thealchemist, 2024-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.5.1.2

### Espressif SoC revision.

ESP32 C3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3 devkit m1 

### Power Supply used.

USB

### What is the expected behavior?

I have setup of the 2 devices, one is time client on its element and other device has  time server model implemented on the element.
I want to know the time on the time on the server element. For this I send the ' ESP_BLE_MESH_MODEL_OP_TIME_GET ' from the client to the server. This will trigger the ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_GET_MSG_EVT on the server callback( as I set the response control to be setted as  respose by app). Now I need to send the time status message to client. For this I write down the small function to do that:
```c
ESP_LOGI(TAG, "ESP_BLE_MESH_TIME_SCENE_SERVER_RECV_GET_MSG_EVT");
ESP_LOGI(TAG, "Received addredd is %04x", param->ctx.addr);
esp_ble_mesh_msg_ctx_t ctx;
memset(&ctx,0,sizeof(ctx));
ctx.net_idx = 0x0000;
ctx.app_idx = 0x0000;
ctx.addr = param->ctx.addr;
ctx.send_ttl = ESP_BLE_MESH_TTL_DEFAULT;
struct timeval tv;
gettimeofday(&tv,NULL);
ESP_LOGI(TAG,"Sending the epoch time %lld",tv.tv_sec);;
esp_ble_mesh_state_change_time_status_t time_status;
memcpy(&time_status.tai_seconds,&tv.tv_sec,sizeof(tv.tv_sec));
ESP_LOGI(TAG,"status size is %d",sizeof(time_status));
 // need to send the status message
ESP_ERROR_CHECK_WITHOUT_ABORT(esp_ble_mesh_server_model_send_msg(parammodel,&ctx,ESP_BLE_MESH_MODEL_OP_TIME_STATUS,sizeof(time_status),(uint8_t*)&time_status));
```

On the client side I am getting the following error:

[image]

How ever on further inpecton I found out that the size of the esp_ble_mesh_state_change_time_status_t is 12 bytes, which is causing the problem, while the expected receive size should be 10 bytes. 
How ever I change the structure to this:

```c
typedef struct __attribute__((packed)) {
    uint8_t  tai_seconds[5];        /*!< The current TAI time in seconds */
    uint8_t  subsecond;             /*!< The sub-second time in units of 1/256 second */
    uint8_t  uncertainty;           /*!< The estimated uncertainty in 10-millisecond steps */
    uint8_t time_authority : 1;    /*!< 0 = No Time Authority, 1 = Time Authority */
    uint16_t tai_utc_delta_curr : 15;   /*!< Current difference between TAI and UTC in seconds */
    uint8_t time_zone_offset_curr;     /*!< The local time zone offset in 15-minute increments */
} time_status_t;
```
now my messages is correctly parsed by the client, as its size is 10 bytes. 

### What is the actual behavior?

The server should send the message using the `esp_ble_mesh_state_change_time_status_t`  structure.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

chegewara, 2025-02-24:
Hi,
today i hit the same wall (i forgot about this issue being opened) when i am sending time from client model.
```
typedef struct {
    uint8_t  tai_seconds[5];        /*!< The current TAI time in seconds */
    uint8_t  sub_second;            /*!< The sub-second time in units of 1/256 second */
    uint8_t  uncertainty;           /*!< The estimated uncertainty in 10-millisecond steps */
    uint16_t time_authority : 1;    /*!< 0 = No Time Authority, 1 = Time Authority */
    uint16_t tai_utc_delta : 15;    /*!< Current difference between TAI and UTC in seconds */
    uint8_t  time_zone_offset;      /*!< The local time zone offset in 15-minute increments */
} __attribute__((__packed__)) esp_ble_mesh_time_set_t;
```
Command:
```
esp_ble_mesh_client_model_send_msg(model, ctx, ESP_BLE_MESH_MODEL_OP_TIME_SET, sizeof(esp_ble_mesh_time_set_t), (uint8_t*)data, 0, false, ROLE_NODE);
```
Packed fixed the problem.

Thanks

sworisbreathing, 2026-10-30:
hey @forx157 is there an update on this one? I'm hitting it on esp-idf v6.1.

I was hoping to open a PR to fix but having dug into the code a bit, I'm getting a bit confused about how the different ble mesh components fit together (like `esp_ble_mesh...` vs `btc_ble_mesh...`) so I'm not 100% clear on which structs should be packed vs which should not, and also how to write regression tests to ensure the structs are packed and unpacked properly.

Would you be able to provide some guidance here?

forx157, 2026-10-30:
Hi, @chegewara  @sworisbreathing 
Sorry for the late reply. We have confirmed this problem and submitted the fix internally. We are currently testing it. If you are more urgent, I can send you the internal patch in advance after the test passes.

closed 2026-11-02 by espressif-bot

chegewara, 2026-11-02:
Hi
any chance to have it backported to 5.5.4? Not really important, but may be useful for some people.

Thanks

chegewara, 2026-11-03:
Sadly those are not the only structs affected by missing packed attribute:

```
'BLE_MESH_WIRE_SIZE'
   36 | BLE_MESH_WIRE_SIZE(esp_ble_mesh_light_hsl_range_status_cb_t, 9);
'BLE_MESH_WIRE_SIZE'
   31 | BLE_MESH_WIRE_SIZE(esp_ble_mesh_light_ctl_temperature_range_status_cb_t, 5);
'BLE_MESH_WIRE_SIZE'
   34 | BLE_MESH_WIRE_SIZE(esp_ble_mesh_light_lightness_range_status_cb_t, 5);
macro 'BLE_MESH_WIRE_SIZE'
   29 | BLE_MESH_WIRE_SIZE(esp_ble_mesh_gen_power_range_status_cb_t, 5);
'BLE_MESH_WIRE_SIZE'
   32 | BLE_MESH_WIRE_SIZE(esp_ble_mesh_light_xyl_range_status_cb_t, 9);
macro 'BLE_MESH_WIRE_SIZE'
   28 | BLE_MESH_WIRE_SIZE(esp_ble_mesh_gen_loc_global_status_cb_t, 10);
macro 'BLE_MESH_WIRE_SIZE'
   31 | BLE_MESH_WIRE_SIZE(esp_ble_mesh_gen_loc_local_status_cb_t, 9);
```

all in master branch pulled today.