espressif/esp-idf #13242: ble_config_file_path_update() only works with filename of exactly 15 chars

sergiutigu, 2024-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ble_config_file_path_update() doesn't work with files name less than 15 characters.

Function is inside the file:
../components/bt/host/bluedroid/btc/core/btc_config.c

Bug is  at line 270:
memcpy(CONFIG_FILE_PATH, file_path, strlen(file_path));

this should be replaced by the following line:
strncpy(CONFIG_FILE_PATH, file_path, NVS_NS_NAME_MAX_SIZE);

I have already tested this fix and it is working.

Explanation of the bug:
The issue with original code is that you initialize the CONFIG_FILE_PATH to:
static char CONFIG_FILE_PATH[NVS_NS_NAME_MAX_SIZE] = "bt_config.conf"; 
which is 15 characters long.
Now you are using  strlen(file_path) inside memcpy, which means if that the new path is less than 15 characters you only copy the part of the source, whiteout the '\0'. 
In our case this resulted in a filename called #####.conff . The last 'f' was from previous value. 
With this the delete of the file was not working.

Reproduction of the issue:
This error is very easy to reproduce, just attempt to configure BT with a path less than 15 characters then attempt to delete that file using nvs_open() and nvs_erase_all().
This is working with a name only of exactly 15 characters.

Please fix this bug.

Weijian-Espressif, 2024-02-26:
Please use esp_bt_config_file_path_update() in ESP_IDF

Alvin1Zhang, 2024-03-10:
Thanks for reporting, feel free to reopen.

closed 2024-03-10 by Alvin1Zhang

esp-zhp, 2024-08-07:
1-The esp_bt_config_file_path_update API can only be called before the host initialization (esp_bluedroid_init).
2-The esp_bt_config_file_path_update API is a newly introduced feature, effective only from version 5.2 onwards. The relevant commit is 23a20e4b27633cc6a0722abcef758a223fd105dd.