espressif/esp-idf #13243: Linking a pre-existing .a file to esp32-s3 projec

mlaliwala28, 2024-02-23:
It should be the same as in plain old CMake:

```
add_library(<library-name> STATIC <your-source-files>)
```
That should compile your source files and archive them to a <library-name>.a file. You can then link it to the enclosing component:

```
target_link_libraries(COMPONENT_LIB <PUBLIC|PRIVATE> <library-name>)
```

Or am I missing something with your question?

_Originally posted by @renzbagaporo in https://github.com/espressif/esp-idf/issues/3613#issuecomment-500682359_

I have added library and set property.  However, I don't understand what should go in the place of COMPONENT_LIB in the below statement.

"target_link_libraries(COMPONENT_LIB <PUBLIC|PRIVATE> <library-name>)"

I have tried this"target_link_libraries(lib PUBLIC libalgobsec)  ", but it has built error.

Please look at my device tree below and CMakeList.txt.
            
[image]

Thanks for you help!

igrr, 2024-02-23:
@mlaliwala28 Which IDF version are you using? Also, your screenshot didn't seem to attach properly. Instead of the screenshot, could you please attach you CMakeLists.txt file as text? Thanks.

igrr, 2024-02-24:
I see, in this case your use case should be covered by this example project: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/import_prebuilt

In particular, https://github.com/espressif/esp-idf/blob/5f4249357372f209fdd57288265741aaba21a2b1/examples/build_system/cmake/import_prebuilt/main/CMakeLists.txt#L6-L10

mlaliwala28, 2024-02-24:
Ivan,

Thanks for your response, but I don't think that I understand the CMake
file, since I am new to it.  I understand that the example shows how you
can prebuild *.a file.  However, I don't need a pre-build step since my
file is already existing.  I just need to add_prebuilt_library. I tried to
set up a similar structure with similar folder names, but it is not working
for me.  Please excuse my ignorance!  Below is the screen capture of my
project with CMakeList.txt file.

[image: image.png]

Thanks!
Mehul

dobairoland, 2024-02-26:
Please read the README of the project: https://github.com/espressif/esp-idf/tree/5f4249357372f209fdd57288265741aaba21a2b1/examples/build_system/cmake/import_prebuilt

You can skip the prebuilding part and use your own library.

Also, those screen captures are really not working.

mlaliwala28, 2024-02-26:
Hi Roland,

Thanks for the information!  I had read readme before my previous email.
However, the *.a file is provided by the manufacturer, so I do not have *.c
or *.h files.  Only *.h and *.a files were provided. Hre is the list of *.h
files, datatype.h, bsec_interface.h, and bsec_interface_multi.h.  Sorry for
my ignorance again, but can you recompile *.a file? I thought that I was
already on the second step since I have *.a file already.

I have attached a project.  I am sorry that I may be missing a point, but I
am new to CMakes and ESP.

Thank you so much for your help.

Sincerely,
Mehul

ESP-Marius, 2024-08-08:
@mlaliwala28 as mentioned earlier, the https://github.com/espressif/esp-idf/tree/5f4249357372f209fdd57288265741aaba21a2b1/examples/build_system/cmake/import_prebuilt example should cover this use-case.

If you are still having issues using this, then please re-open this ticket, preferably with an example project that can help us reproduce this issue.

closed 2024-08-08 by ESP-Marius