espressif/esp-idf #13244: LOG COLOR STILL ERROR ON BUILD by wront type INT vs UINT32_T

filzek, 2024-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2

### Espressif SoC revision.

esp32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32

### Power Supply used.

USB

### What is the expected behavior?

C:/Espressif/frameworks/esp-idf-v5.2/components/log/include/esp_log.h:291:27: error: format '%d' expects argument of type 'int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  291 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"

why espressif doenst fix this right away, everyone complain about it but no one takes action to a simple question???

### What is the actual behavior?

work as intent, if the compiler has PROBLEM fix the code to work as the compiler wants it to do, its impossible in 2024 that a simple question couldnt be addressed right away.

### Steps to reproduce.

try to build any log color app will bring error.

C:/Espressif/frameworks/esp-idf-v5.2/components/log/include/esp_log.h:291:27: error: format '%d' expects argument of type 'int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  291 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"

### Debug Logs.

```plain
C:/Espressif/frameworks/esp-idf-v5.2/components/log/include/esp_log.h:291:27: error: format '%d' expects argument of type 'int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  291 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"
```

### More Information.

C:/Espressif/frameworks/esp-idf-v5.2/components/log/include/esp_log.h:291:27: error: format '%d' expects argument of type 'int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  291 | #define LOG_COLOR(COLOR)  "\033[0;" COLOR "m"

igrr, 2024-02-23:
Hi @filzek, please check this section of the migration guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/gcc.html#int32-t-and-uint32-t-for-xtensa-compiler.

If you can't change the application code to use the `inttypes.h` format specifiers (`PRId32` and so on) then you can disable format string warnings in your component CMakeLists.txt file, [like this](https://github.com/espressif/esp-protocols/blob/a363beea6f0e9f051e9e1eb936f216698e7529e1/examples/mqtt/main/CMakeLists.txt#L4).

(I have explained the reason for this change in replies over at https://github.com/espressif/esp-idf/issues/9511, if you need more details.)

filzek, 2024-02-24:
Understood, but somehow if was is to add from the SDK a compiler flag to do not bring this warn/error for it, like to auto add the lib flag in the compiler, and in the menuconfig bring the option to disable it, so could be much more user understandable.

igrr, 2024-02-26:
Well, adding `-Wno-format` flag to the compilation flags of the component is just one line in CMake, and we didn't want to set the precedent of adding a Kconfig option for every compiler warning flag — as that would be an endless number of things to maintain and test. Besides, we support both GCC and Clang toolchains now, and they don't share exactly the same compiler flags in all cases. So adding this Kconfig options is probably something we won't do, sorry.

filzek, 2024-02-27:
> Well, adding `-Wno-format` flag to the compilation flags of the component is just one line in CMake, and we didn't want to set the precedent of adding a Kconfig option for every compiler warning [...]

Sure, I understand. Closing it!

closed 2024-02-27 by filzek