espressif/esp-idf #13245: [Examples/peripherals/usb_serial_jtag] Stack size increase required.

GTCLive, 2024-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Tested on ESP32-S3
ESP-IDF:  v5.3-dev-2087-g5f42493573

Crash loop occurs. The fix/workaround is to revise line 16 in usb_serial_echo_main.c 
From 
`#define ECHO_TASK_STACK_SIZE (2048)`
To
`#define ECHO_TASK_STACK_SIZE (4096)`

Ref:
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb_serial_jtag/usb_serial_jtag_echo/main/usb_serial_echo_main.c

Amazing SDK, appreciate all the efforts guys. Keep it up.

GTCLive, 2024-02-23:
Or remove Line 44. Which doesn't print anything anyway... up to you. 
`Line 44: ESP_LOG_BUFFER_HEXDUMP("Recv str: ", data, len, ESP_LOG_INFO);`

The usb_serial_jtag_write_bytes() works perfectly fine otherwise!

igrr, 2024-02-23:
> print back anything on the JTAG port anyway (only uart ports perhaps?).

Yes, the idea is that it prints to the UART port.

Thanks for the issue report and the suggested fix!

closed 2024-04-04 by espressif-bot