espressif/esp-idf #13248: ESP-IDF vscode extension build warning

Murtaza7745, 2024-02-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using ESP-IDF branch release/v5.1.2 configured in vscode extension.
I have added arduino as a component and have a custom partition as well.

The code is running perfectly till now but when i use build button of vscode extension, after building the code the terminal shows .map file 

There is a warning sign that is bugging me which did not appear when i haven't added arduino as a component.

`WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in esp_idf_size
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in esp_idf_size
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in esp_idf_size
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in esp_idf_size
Total sizes:
Used static DRAM:   44668 bytes (  79912 remain, 35.9% used)
      .data size:   21700 bytes
      .bss  size:   22968 bytes
Used static IRAM:   97598 bytes (  33474 remain, 74.5% used)
      .text size:   96571 bytes
   .vectors size:    1027 bytes
Used Flash size : 1151679 bytes
           .text:  897991 bytes
         .rodata:  253432 bytes
Total image size: 1270977 bytes (.bin may be padded larger)`

Help me find what this issue might be.
Thanks

fhrbata, 2024-02-25:
Hello @Murtaza7745 ,
thank you for the report. Could you please attach the full map file? It should hopefully help to shed some light on what's happening.  The warning means that some of the output sections or memory ranges in the link map file cannot be mapped to the memory type ranges defined by yaml files(e.g. for [esp32](https://github.com/espressif/esp-idf-size/blob/master/esp_idf_size/chip_info/esp32.yaml)) in `esp-idf-size`. 
Thank you!

Murtaza7745, 2024-02-29:
@fhrbata 
Sorry for the delayed response.

Here is the map file for esp32
[map_file.zip](https://github.com/espressif/esp-idf/files/14451910/map_file.zip)

fhrbata, 2024-03-01:
@Murtaza7745 Thank you very much! I will try to take a closer look and report back.

Murtaza7745, 2024-03-01:
@fhrbata Thank you so much. Just for info how soon you can give a reply, as I’m on a very tight schedule for delivering this project

fhrbata, 2024-03-01:
Hello @Murtaza7745 , 

I think you actually  revealed two bugs in the `esp-idf-size`. Meaning, I think there is no problem with your app, just that `esp-idf-size` reports "bogus" warnings. I used the refactored version to get a more verbose output. You can use it as `python -m esp_idf_size --ng jyiot.map` or set `export ESP_IDF_SIZE_NG=1` if interested. Here's screenshot of the output.
[image]

There are two problems/bugs.

1. `warning:  cannot assign memory region iram0_0_seg to chip memory type`

Looking at the map file, I guess you have the `CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM` set. You can check it in the `sdkconfig` file.This option extends the IRAM memory range for portion of Internal SRAM1. Unfortunately this possibility is not reflected in [esp32.yml](https://github.com/espressif/esp-idf-size/blob/master/esp_idf_size/chip_info/esp32.yaml#L10). Currently the IRAM memory type size in `esp32.yaml` is limited to Internal SRAM0 only.
 ```
IRAM:
  primary_address: 0x40070000
  length: 2 * 0x8000 + 16 * 0x2000
```
So `esp-idf-size` considers IRAM only from `0x40070000` to `0x400a0000`. SRAM1 starts at `0x400a0000`. This extension of IRAM for portion of SRAM1 can be seen in the linker script [here](https://github.com/espressif/esp-idf/blob/master/components/esp_system/ld/esp32/memory.ld.in#L41). The memory ranges can be seen in [esp32 TRM](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf), `Table 1­2. Embedded Memory Address Mapping`.

IIUC this should be fixed by setting the IRAM type size to `0x50000`, which should cover the whole possible IRAM memory type.

```diff
diff --git a/esp_idf_size/chip_info/esp32.yaml b/esp_idf_size/chip_info/esp32.yaml                                                   
index a2ee7bcfdaf9..b8eb3c69df86 100644                                                                                              
--- a/esp_idf_size/chip_info/esp32.yaml                                                                                              
+++ b/esp_idf_size/chip_info/esp32.yaml                                                                                              
@@ -7,7 +7,7 @@ DRAM:                                                                                                                
   length: 17 * 0x2000 + 4 * 0x8000 + 4 * 0x4000                                                                                     
 IRAM:                                                                                                                               
   primary_address: 0x40070000                                                                                                       
-  length: 2 * 0x8000 + 16 * 0x2000                                                                                                  
+  length: 0x50000                                                                                                                   
 CACHE_I:                                                                                                                            
   primary_address: 0x400C2000                                                                                                       
   length: 0xB3E000                                
```

IIUC this problem is reported only in the refactored `esp-idf-size` version, which strictly maps memory ranges from the map file into the target's memory types defined in [yaml files](https://github.com/espressif/esp-idf-size/tree/master/esp_idf_size/chip_info). But I think it would show up in the legacy version too, if the output sections accounted to iram overflow `0x400a0000`.

2.   `warning:  Flash Data overflow detected!`. In the legacy version reported simply as `WARNING: Given section not found in any memory region.`

Here the problem is that output sections from the map file are selected based on regular expressions. For example output sections containing `flash` string in their names are considered to be part of the flash memory. The regular expression can be seen [here](https://github.com/espressif/esp-idf-size/blob/master/esp_idf_size/core.py#L154). As you can see all the sections reported in the warnings contain `flash` string in the generated name, but they should not be accounted, because they are actually not loaded into flash.  We will have to harden this somehow.

The new version should select the output sections based on the info in ELF file if available and should not report these warnings.

Thank you very much for this report! Again, I believe there is nothing wrong with your app.

Murtaza7745, 2024-03-01:
@fhrbata Thanks for such a great explanation.

Few doubts:

1. You mentioned,
> This extension of IRAM for portion of SRAM1 can be seen in the linker script

I have attached the memory.ld file  from "build\esp-idf\esp_system\ld\memory.ld"
[memory_ld.zip](https://github.com/espressif/esp-idf/files/14460952/memory_ld.zip)

This file doesn't have below section
```
#if CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM
#define SRAM1_IRAM_LEN 0xA000
#else
#define SRAM1_IRAM_LEN 0x0
#endif
```

2. How do i update the esp32.yaml for IRAM length as 0x50000 ?

3. What do you mean by this ?

>For example output sections containing flash string in their names are considered to be part of the flash memory.

4.  Does this mean in the ESP-IDF v5.2 has this issue fixed or do you mean something else?

>The new version should select the output sections based on the info in ELF file if available and should not report these warnings.

Again thanks @fhrbata for the help.

fhrbata, 2024-03-01:
> @fhrbata Thanks for such a great explanation.

You are welcome of course.

> Few doubts:
>
> 1. You mentioned,
>
>
> I have attached the memory.ld file from "build\esp-idf\esp_system\ld\memory.ld" [memory_ld.zip](https://github.com/espressif/esp-idf/files/14460952/memory_ld.zip) [...]

Note that `memory.ld` in the application's build directory is generated from `memory.ld.in` template, which is part of the `esp-idf`(`components/esp_system/ld/esp32/memory.ld.in`). Meaning the generated file doesn't contain these conditions, because they are processed and replaced during build. You can see the template for` v5.1.2` `esp32` target here [components/esp_system/ld/esp32/memory.ld.](https://github.com/espressif/esp-idf/blob/v5.1.2/components/esp_system/ld/esp32/memory.ld.in#L41)

> 
> also in sdkconfig file
> 
> `CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM=y`

Right, so it's set as expected, based on the `iram0_0_seg` size(`0x2a000`) in the map file.

> 
> 2. How do i update the esp32.yaml for IRAM length as 0x50000 ?

It's part of the `esp-idf-size` python package. You would have to clone it and install it as editable(`pip install -e`) package and change the size there. I will prepare MR for this soon. Note that this problem is actually not reported in your output. It's just something I found out when I used the refactored `esp-idf-size` version on the map file you provided. I will explain the legacy vs refactored version below. I apologize if I caused any confusion because of this.

> 3. What do you mean by this ?
> 
> > For example output sections containing flash string in their names are considered to be part of the flash memory.

This is related to the fact how the output sections are accounted into different memory types and reported by `esp-idf-size`. The link map file contains information, in the `Linker script and memory map` section, how the input sections(presented in object files) are mapped into output sections(presented in the final elf file) during link process, based on the linker script files(`memory.ld`, `sections.ld`, which are generated from templates as noted above and other linker script files).  So for example in your map file there are output sections like `.flash.appdesc `,  `.flash.rodata` and `.flash.text`. Based on the [regex](https://github.com/espressif/esp-idf-size/blob/master/esp_idf_size/core.py#L154) mentioned earlier, output sections containing `flash` string are considered(by `esp-idf-size`) to be loaded and part of the flash memory. In your case there are other sections, like the `.xt.lit._ZN5MB_FS10flashReadyEv` reported in the warning, which contain `flash` string. These are unfortunately also consider by `esp-idf-size` and since the address for this section in the map file is actually `0x0000000000000000`, it doesn't fit into the expected address range for flash defined in the yaml file. Hence the warning is reported. But the real problem is that this output section should not be even considered to be accounted into flash by `esp-idf-size`. I will have to take a closer look to see how to fix this properly.

> 
> 4. Does this mean in the ESP-IDF v5.2 has this issue fixed or do you mean something else?
The `esp-idf-size` is a python package which is installed during install(sh, bat. ps1) into python venv and is used by all `esp-idf` versions. It also contains `ng` subpackage, which is a refactored version, but it's currently by default used only in master branch and should be the default version in the future `esp-idf` major release. In older releases it can be used by explicitly setting the `ESP_IDF_SIZE_NG` environmental variable. I used it to compare the outputs to see if there are any differences and found the problem with the `CONFIG_ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM`.

Hopefully this should make things a little bit more clear.

> 
> > The new version should select the output sections based on the info in ELF file if available and should not report these warnings.
> 
> Again thanks @fhrbata for the help.

You're welcome and thanks again for the report!

Murtaza7745, 2024-03-01:
`I will have to take a closer look to see how to fix this properly.`

Can we start an email thread and connect to solve the issue, i can share the complete project file there.
Let me know if you have some other alternative.

fhrbata, 2024-03-01:
> `I will have to take a closer look to see how to fix this properly.`
> 
> Can we start an email thread and connect to solve the issue, 

I have no problem with this approach, but I think it would be better to keep all the discussion here. At least for future reference.

> i can share the complete project file there. Let me know if you have some other alternative.

The map file you provided should be enough for now. I have some idea/workaround for this. I will prepare a fix and we will see if it breaks something :). 

Thank you for the offer, it very much appreciated! I will definitely reach out to you if needed. Again, thank you very much.

Murtaza7745, 2024-03-01:
Okay, no problem continuing the discussion here.
As mentioned earlier, the only thing i don't have is time, so how soon can you respond with a fix?
I really appreciate you doing this.

fhrbata, 2024-03-02:
Hello @Murtaza7745 , I created internal MRs which should fix the problems we were discussing. I believe that these shouldn't be blockers for you or am I missing something? Unless I've missed something, these are just spurious warnings. Thank you

Murtaza7745, 2024-03-04:
Looking forward to hearing from you about the status of your internal MR.

closed 2024-03-25 by espressif-bot