espressif/esp-idf #13260: DMA PSRAM

vwwwvwww, 2024-02-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

新版本文档https://docs.espressif.com/projects/esp-idf/zh_CN/v5.2/esp32/api-guides/memory-types.html?highlight=psram%20dma

我注意到新版本的idf没有了关于psram的关于dma描述符的这一段:
请问是新版本的esp-idf支持psram使用DMA了吗?

      片外 RAM 不能用于储存 DMA 事务描述符，也不能用作 DMA 读写操作的缓冲区 (Buffer)。因此，当片外 RAM 启用时，与 DMA 搭配使用的 Buffer 必须先使用 heap_caps_malloc(size, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL) 进行分配，之后可以调用标准 free() 回调释放 Buffer。

旧版本文档https://docs.espressif.com/projects/esp-idf/zh_CN/v5.0.6/esp32/api-guides/external-ram.html

suda-morris, 2024-02-26:
esp32's DMA **can't** access the PSRAM region as well. 

Up till now, only ESP32S3 and ESP32P4 support the PSRAM DMA feature.

andylinpersonal, 2024-02-26:
Can the DMA engine of any chip other than S3 and P4 access flash as well?

vwwwvwww, 2024-02-26:
Thank you both, 

this question can be closed now.

closed 2024-02-27 by Alvin1Zhang