espressif/esp-idf #13261: Connect to ESP32 device via standard Bluetooth dialogue using gatt server example

Talha-Babar, 2024-02-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Description:
I am currently facing an issue where I am unable to connect to an ESP32 device via the standard Bluetooth dialogue on both Android and iOS devices. I have been using the GATT server example provided by the ESP-IDF framework to implement Bluetooth Low Energy (BLE) communication between the ESP32 device and mobile devices.

Here are the specific details of the problem:

Current State: I have successfully implemented the GATT server example:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server
I am able to read and write data using third-party apps on both Android and iOS devices. However, when attempting to connect to the ESP32 device via the standard Bluetooth dialogue on Android or iOS, I encounter difficulties.

Issue: The ESP32 device is not discoverable or connectable through the standard Bluetooth dialogue on Android or iOS devices. Instead, the devices show a message indicating that "this device requires special software to operate". This makes it challenging to establish a connection between the ESP32 device and the mobile devices in a user-friendly manner.

Desired Outcome: I want to verify if it is possible to connect to the ESP32 device via the standard Bluetooth dialogue on Android and iOS devices. If it is indeed possible, I would like guidance on what changes need to be made to the ESP32 codebase to facilitate this connection. My ultimate goal is to enable seamless and user-friendly communication between the ESP32 device and mobile devices, utilizing third-party apps for data transmission once the connection is established.

Any insights, suggestions, or guidance on how to resolve this issue and make the ESP32 device discoverable and connectable via the standard Bluetooth dialogue would be greatly appreciated. Thank you for your assistance.

zhp0406, 2024-02-26:
The ESP32 device is not discoverable or connectable through the standard Bluetooth dialogue on Android or iOS devices.

This is because Android or iOS devices do not include the desired service in their broadcast packets（adv data）. If there is a corresponding service, it can be detected in the broadcast.

zhp0406, 2024-02-26:
For instance, advertising data in the following can be discoverable or connectable through the standard Bluetooth interface on Android or iOS devices.

```
static uint8_t raw_adv_data[] = {
    0x02, 0x01, 0x06, 
    0x03, 0x19, 0xC0, 0x03, 0x04, 
    0x09, 0x48, 0x49, 0x44, 0x02, 0x0A, 0x09, 0x03, 0x03, 0x12, 0x18, 
    0x05, 0x12, 0x06, 0x00, 0x10, 0x00,
};
static uint8_t raw_scan_rsp_data[] =
{
    0x02, 0x01, 0x06, 
    0x03, 0x19, 0xC0, 0x03, 0x04, 
    0x09, 0x48, 0x49, 0x44, 0x02, 0x0A, 0x09, 0x03, 0x03, 0x12, 0x18, 
    0x05, 0x12, 0x06, 0x00, 0x10, 0x00,
};
```

zhp0406, 2024-02-27:
You can set the broadcast content yourself. One of the items in the broadcast content is setting the Complete Local Name, which I have set to HID. You can change it to any other name you prefer.
This code represents static advertising data used for Bluetooth Low Energy (BLE) devices. Here's the breakdown of each byte:

0x02: Indicates that the following data length is 2 bytes.

0x01: Indicates the type of advertising data, in this case, Flags, which describe the Bluetooth device type and status.

0x06: Represents the value of the Flags, indicating that the device is in LE General Discoverable Mode and BR/EDR Not Supported.

0x03: Indicates that the following data length is 3 bytes.

0x19: Represents the type of advertising data, specifically, Appearance, which describes the device's appearance category.

0xC0, 0x03: Represents the HID (Human Interface Device) Keyboard appearance.

0x04: Indicates that the following data length is 4 bytes.

**0x09: Represents the type of advertising data, which is the Complete Local Name, indicating the local name of the device.

0x48, 0x49, 0x44: Represents the local name of the device as "HID".**

0x02: Indicates that the following data length is 2 bytes.

0x0A: Represents the type of advertising data, which is the TX Power Level, indicating the transmit power level.

0x09: Represents the transmit power level as -9 dBm.

0x03: Indicates that the following data length is 3 bytes.

0x03: Represents the type of advertising data, which is Service UUIDs, describing the UUIDs of services provided by the device.

0x12, 0x18: Represents the UUID of the first service.

0x05: Indicates that the following data length is 5 bytes.

0x12: Represents the type of advertising data, which is the Slave Connection Interval Range, specifying the range of slave connection intervals.

0x06, 0x00: Represents the minimum (6 units, each unit 1.25 milliseconds) and maximum (10 units) slave connection interval range.

zhp0406, 2024-02-27:
Have you tried with another Android device? I tested it here, and the results show that Android devices can detect the ESP32 device.

Talha-Babar, 2024-02-27:
Yes, i have tried it to different Android devices and it detects the ESP32 device but not able to connect.

zhp0406, 2024-02-28:
What hints does the ESP32 log provide when the connection is not established? I believe this situation shouldn't occur. could you tested the HID discovery and connection using the HID example（[HID example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_hid_device_demo)） with an Android device?

zhp0406, 2024-02-28:
here is my code and log，It works well in my device，
```
/*
 * SPDX-FileCopyrightText: 2021-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */

#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_bt.h"

#include "esp_gap_ble_api.h"
#include "esp_gatts_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_main.h"
#include "esp_bt_device.h"
#include "example_ble_sec_gatts_demo.h"

#define GATTS_TABLE_TAG "SEC_GATTS_DEMO"

#define HEART_PROFILE_NUM                         1
#define HEART_PROFILE_APP_IDX                     0
#define ESP_HEART_RATE_APP_ID                     0x55
#define EXAMPLE_DEVICE_NAME                       "HID"
#define HEART_RATE_SVC_INST_ID                    0

#define GATTS_DEMO_CHAR_VAL_LEN_MAX               0x40

#define ADV_CONFIG_FLAG                           (1 << 0)
#define SCAN_RSP_CONFIG_FLAG                      (1 << 1)

static uint8_t adv_config_done = 0;

static uint16_t heart_rate_handle_table[HRS_IDX_NB];

static uint8_t test_manufacturer[3]={'E', 'S', 'P'};

static uint8_t sec_service_uuid[16] = {
    /* LSB <--------------------------------------------------------------------------------> MSB */
    //first uuid, 16bit, [12],[13] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0x18, 0x0D, 0x00, 0x00,
};

// config adv data
static esp_ble_adv_data_t heart_rate_adv_config = {
    .set_scan_rsp = false,
    .include_txpower = true,
    .min_interval = 0x0006, //slave connection min interval, Time = min_interval * 1.25 msec
    .max_interval = 0x0010, //slave connection max interval, Time = max_interval * 1.25 msec
    .appearance = 0x00,
    .manufacturer_len = 0, //TEST_MANUFACTURER_DATA_LEN,
    .p_manufacturer_data =  NULL, //&test_manufacturer[0],
    .service_data_len = 0,
    .p_service_data = NULL,
    .service_uuid_len = sizeof(sec_service_uuid),
    .p_service_uuid = sec_service_uuid,
    .flag = (ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT),
};
// config scan response data
static esp_ble_adv_data_t heart_rate_scan_rsp_config = {
    .set_scan_rsp = true,
    .include_name = true,
    .manufacturer_len = sizeof(test_manufacturer),
    .p_manufacturer_data = test_manufacturer,
[... 594 lines cut ...]
I (552) main_task: Calling app_main()
I (572) BTDM_INIT: BT controller compile version [0f0c5a2]
I (572) BTDM_INIT: Bluetooth MAC: 08:b6:1f:ed:4e:42
I (582) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (892) SEC_GATTS_DEMO: app_main init bluetooth
I (992) SEC_GATTS_DEMO: The number handle = 8
I (1002) main_task: Returned from app_main()
I (1022) SEC_GATTS_DEMO: advertising start success
I (25852) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
W (28862) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (29092) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC
I (29092) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
I (29092) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID
I (29542) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
I (29602) SEC_GATTS_DEMO: remote BD_ADDR: 54f8fe672588
I (29602) SEC_GATTS_DEMO: address type = 1
I (29612) SEC_GATTS_DEMO: pair status = success
I (29612) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_BOND
I (29622) SEC_GATTS_DEMO: Bonded devices number : 1
I (29622) SEC_GATTS_DEMO: Bonded devices list : 1
I (29632) SEC_GATTS_DEMO: 54 f8 fe 67 25 88 
W (34712) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (34712) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (34732) SEC_GATTS_DEMO: advertising start success
```

Talha-Babar, 2024-02-28:
Would you kindly share your complete project, one of the header files is not included in the HID example and I am facing compiling issues

zhp0406, 2024-02-29:
please replace code in "esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server/main/example_ble_sec_gatts_demo.c"
commit id : 21490894cd

zhp0406, 2024-03-07:
@Talha-Babar 
Have you solved the problem？

Talha-Babar, 2024-03-07:
Hello @zhp0406,

Thank you for following up. I've tested your code and successfully established a connection to the ESP32 using the standard Bluetooth dialogue on both Android and iOS. However, I've encountered an issue where I am unable to send data from a third-party app after connecting via Bluetooth. The third-party app fails to recognize any connected Bluetooth devices, necessitating a reconnection each time data needs to be sent.

I'm unsure whether this issue stems from the third-party app itself or if it's related to the ESP32 code.

zhp0406, 2024-03-08:
@Talha-Babar 
It looks like this issue has been resolved, if there are other issues, please start another github issue and give related log,thanks.

Alvin1Zhang, 2024-03-10:
Thanks for reporting and sharing the updates, feel free to reopen.

closed 2024-03-10 by Alvin1Zhang

Talha-Babar, 2024-03-27:
Hello @zhp0406,

I apologize for the delay as I had temporarily paused the project, but I have resumed work on it now. The code is functioning correctly, and I can successfully establish a connection through the standard Bluetooth dialog. However, I am encountering an issue with sending and receiving serial data. For testing purposes, I am using the Serial Bluetooth Terminal app. Unfortunately, the app is unable to maintain a stable connection and reports that no serial profile is found. Consequently, I am unable to send or receive data through the app.

Thank you for your understanding.

Best regards,