espressif/esp-idf #13264: component/ethernet_init fails whith debug (or verbose) logs

mickeyl, 2024-02-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When debug or verbose logs are enabled, the w5500_task will crash at runtime because the default stack is not enough.

I would suggest to enhance `/examples/ethernet/basic/components/ethernet_init/ethernet_init.c` with `mac_config.rx_task_stack_size = 8192;`, iff debug or verbose logs are enabled.

This is the source of quite a number of questions on the ESP32 forum and elsewhere, where people are looking for problems in their code, when it's actually just a misconfigured stack size.

mickeyl, 2024-03-07:
Alternatively you might want to make this customizable via KConfig?

kostaond, 2024-03-07:
@mickeyl Thanks for the proposed improvement. I'll have a look at it. However, please be patient since we currently have different priorities. Thanks for understanding.

mickeyl, 2024-03-07:
Absolutely, this is completely uncritical. Thanks for responding.

mickeyl, 2025-05-01:
@kostaond Would a pull request of that have a chance to make it upstream?

kostaond, 2025-05-02:
Hi @mickeyl, please consider using https://github.com/espressif/esp-eth-drivers/tree/master/ethernet_init. This version of Ethernet init has stack size configurable via Kconfig.

mickeyl, 2025-05-02:
Oh, I wasn't aware of that. I reckon it gets updated frequently to be in sync with ESP-IDF? Thanks, I'll move to using that component.

closed 2025-05-02 by mickeyl

mickeyl, 2025-08-13:
I just tried that using

```yaml
  ethernet_init:
    git: https://github.com/espressif/esp-eth-drivers.git
    path: ethernet_init
```

in my `idf_component.yml`. However it bails out during dependency resolution:

```
  ERROR: The "override_path" field in the manifest file
  "/private/var/folders/7x/fdswh61d3vq9vzsrx42qyyxh0000gn/T/tmppfs_jqvv/lan865x/idf_component.yml"
  does not point to a directory.  You can safely remove this field from the
  manifest if this project is an example copied from a component repository.
  The dependency will be downloaded from the ESP component registry.
  Documentation:
  https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html#override-path
```