espressif/esp-idf #13268: Interfacing PTX130W NFC frontend with the ESP32C3 MCU for custom application.

rana-roy, 2024-02-27:
### Is your feature request related to a problem?

I am developing a wireless charger with the PTX130W NFC frontend interfaced with the ESP32C3 MCU. The MCU and NFC frontend communicate using SPI on reception of interrupts from the NFC frontend. I have non-OS SDK for the Renesas eval kit with the R7FA4M2AB3CNE#AA0 MCU and PTX130W NFC frontend.

### Describe the solution you'd like.

I want a reference for the PTX130W Zephyr-based SPI driver. Is there any reference available?
Will it be possible to get support for developing a driver on my own if there is no reference available?

sylvioalves, 2024-02-27:
@rana-roy I could not find any PTX130W Zephyr-based SPI driver. Is there any? Anyway, having some code reference like non-OS SDK would be enough to support it on esp-idf. Probably you would need to take care of timing (delay, ticks), interrupt and semaphores.

rana-roy, 2024-02-28:
Thank you @sylvioalves for your quick response. I am trying to port the SDK code into Zephyr based driver. In doing so, I find any roadblock, your support will be very grateful to me.

rftafas, 2024-02-28:
Hi!

If I understood the request, we are talking about a Zephyr Driver for a peripheral connected to an Espressif device.

For sanity check, if you are not familiar with Zephyr, I recommend the [Pull-Up Resistor Channel](https://www.youtube.com/@thepullupresistor). Although they are not affiliated with Espressif, their content is good and they even have the starting information steps about writing a driver.

Once you have that, given this is not necessarily an Espressif / ESP-IDF related issue, you will find a whole [community around zephyr to help you](https://github.com/zephyrproject-rtos/zephyr). We are pretty active there as well. You can also try the [Discord Channel](https://discord.com/channels/720317445772017664/883444902971727882).

If you meet a roadblock, let us know.

In time: since this is preemptive action and not a request/report, I suggest either moving to discussion or closing the topic.

rana-roy, 2024-02-28:
Thank you @rftafas for guiding me!

closed 2024-02-28 by rana-roy

BH4ME, 2026-09-07:
> @rana-roy have you ported PTX130W driver to Zephyr? Have anything to share? 😀 
maybe i need to……