espressif/esp-idf #13269: ESP-IDF : Move autogenerated sdkconfig file to build folder

sukesh-ak, 2024-02-27:
### Is your feature request related to a problem?

Currently, the autogenerated sdkconfig file is created in the root of the project folder.
When we are working on a project targeting multiple ESP32-xx MCU's it becomes difficult to compile for each of them since it gets over-written by commands like `set-target` or `menuconfig`. This also triggers a full compile.

Easiest way to solve it is using different folder for each MCU using `-B` option
```bash
# set target esp32 and build,flash,monitor
idf.py -B build-esp32 set-target esp32 build
idf.py -B build-esp32 -p COM6 flash monitor
```
```bash
# set target esp32-s3 and build,flash,monitor
idf.py -B build-esp32-s3 set-target esp32s3 build
idf.py -B build-esp32-s3 -p COM6 flash monitor
```
But for the above to work, we need to move the autogenerated `sdkconfig` file to the build folder.

### Describe the solution you'd like.

Solution is to move the autogenerated `sdkconfig` file to the build folder.
This can be done by including a line in the root `CMakeLists.txt` file before the `project.cmake` line

```bash
# Make sure sdkconfig is created in the build folder
set(SDKCONFIG "${CMAKE_BINARY_DIR}/sdkconfig")
```

### Additional context.

You can see the change in the following [sample](https://github.com/sukesh-ak/IDF-ESP_LCD-LVGL)
https://github.com/sukesh-ak/IDF-ESP_LCD-LVGL/blob/d2e6b02a6c6962b7d357e1c87b05ba8da2879e9a/CMakeLists.txt#L4

igrr, 2024-02-27:
[The line](https://github.com/sukesh-ak/IDF-ESP_LCD-LVGL/blob/d2e6b02a6c6962b7d357e1c87b05ba8da2879e9a/CMakeLists.txt#L4) you added to your project looks correct! 

We do the same thing in the examples which require such layout: https://github.com/espressif/esp-idf/blob/5f4249357372f209fdd57288265741aaba21a2b1/examples/build_system/cmake/multi_config/CMakeLists.txt#L3-L6

I don't think we will change the default `SDKCONFIG` setting for the entire ESP-IDF, though — for one, this would be a breaking change to the existing projects. You can continue using this line in you project, though!

Another option is to define `idf.py` configuration profiles as illustrated [here](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config#create-configuration-profile-files-via-filename).

igrr, 2024-02-27:
For projects where `sdkconfig` file (rather than `sdkconfig.defaults`) is tracked in git this change would cause the existing `sdkconfig` file to be ignored, and a new file with the default settings will be created instead in the build directory. So upgrading from an older version of IDF where sdkconfig was in the project directory to the newer version where it is in the build directory will not preserve the configuration. Under our current maintenance policy this would be a breaking change.

Not that we never make breaking changes in minor releases, but we are trying to only do this when 1) the change should affect only a small subset of the users / applications, 2) the problem being solved by the change is severe and there is no alternative to that change. In this specific case, the problem is solved by a single line in the project CMakeLists.txt file, so I think it's completely acceptable to solve this on the application side.

sukesh-ak, 2024-02-27:
Thanks for the explanation. I understand where you are coming from.
In my limited testing from last few years, have not caused any issues thus the curiosity.

I have `sdkconfig` in `.gitignore` file always so that manually set credentials etc through `menuconfig` is not pushed to git accidently. Also the same with backup files too `sdkconfig.old`.

I am a moderator on a large `ESP8266/ESP32` group on FB with 90K+ members.
Related question came up yesterday, so reminded me to raise it here.

closed 2024-02-27 by sukesh-ak

igrr, 2024-02-27:
Thank you as well for the useful suggestion! We'll definitely consider it for the future releases.

And thanks for the moderation work you are doing!