espressif/esp-idf #13270: esp32s3 dma

vwwwvwww, 2024-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How to transfer data from a USB_disk/SD_card to a buffer using DMA?     

Do I just need to allocate memory like this:         
 
         void *buffer = heap_caps_aligned_calloc(
           USBH_HAL_FRAME_LIST_MEM_ALIGN, FRAME_LIST_LEN,sizeof(uint32_t), MALLOC_CAP_DMA);

or

        void *buf1 = heap_caps_malloc(BUF_SIZE, MALLOC_CAP_DMA);  

The document only mentions that the host requires DMA, so I came here to ask specifically.
        https://docs.espressif.com/projects/esp-idf/zh_CN/v5.2/esp32s3/api-reference/peripherals/usb_host/usb_host_notes_dwc_otg.html

Dazza0, 2024-02-28:
HAL API is private and is subject to change. Please use either the [USB Host Library API](https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s2/api-reference/peripherals/usb_host.html) or one of the class drivers (listed in the docs). The USB Host stack already utilizes DMA internally.

Dazza0, 2024-03-05:
@vwwwvwww Closing this issue as DMA is already supported in the ESP-IDF USB Host Stack. Please see the [USB Host Library API](https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s2/api-reference/peripherals/usb_host.html) doc for more details.