espressif/esp-idf #13271: RS485 + Wifi. uart_read_bytes() reads less bytes than requested.

tryger, 2024-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I have a custom ESP32 board which polls 4Mbit/s data to other ESP32 custom board via custom RS485 protocol at baudrate of 115200*120. Then it has to send that data via wifi TCP socket.

I am using only core 0, as core 1 is full busy with a task that does SPI transfers to an ADC. I spawn two tasks one that reads the data from RS485 and other that sends that data out via wifi TCP socket. Wifi and TCP/IP tasks are pinned to core 0. UART interrupt is pinned to core 0 with level 3. SPI interrupt is pinned to core 1.

For UART, I setup RX TOUT = 15, and rx full thresh = 55.

This setup works most of the time, i. e. the UART reads correctly all bytes and sends them through wifi with no failure.

### What is the actual behavior?

There are some times (randomly, maybe not happen in minutes) that the data UART read by uart_read_bytes() is less than the data required in length parameter.

Without sending the data through wifi, it works without any failure.

### Steps to reproduce.

To debug, I am using gpio and a logic analyzer. In the uart isr handler uart_rx_intr_handler_default(), inside uart driver i turn on a gpio at the top of the function and turn it off at the end of it. So I can see the exact time it spends inside uart isr. A correct rs485 data transaction would look like this:
[image]
First row (brown) is the packet sent through rs485 asking for data.
Second row (white) is the data that has to be read.
We can see the uart isr (red line) is triggering at a fixed rate, caused by rx full threshold interrupt.

An error transaction look like this:
[image]
We see there is a uart interrupt that delays to rise. This delay is enough to overflow the FIFO. After some more debug I constated that an overflow is what is happening.

So, I supose wifi is somehow preventing UART interruption to rise.

alisitsyn, 2024-03-06:
Hi @tryger ,

You have the tasks that handle intensive communication as high speed RS485 on core 0. This is not recommended because the high priority system tasks work on this core. Especially the WiFi, BLE, LWIP, esp-timer, etc. These tasks have high priority and use medium level interrupts and may prevent your other tasks on the same core to work properly.
It is possible to play with the [interrupt levels](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/intr_alloc.html) and [affinity settings](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-tcpip-task-affinity) for system components. I understand that the core 1 is busy with the SPI transfers but it may be possible to add the RS485 transfer task to this core as well? In spite of the information above I may miss some information about your system and options for optimizations but can advise to do some architectural optimization of active objects in your system. 
The information [here](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s3/api-guides/performance/speed.html) can be helpful to optimize your system. Also try to disable [the power management](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/power_management.html) ([the issue](https://stackoverflow.com/questions/71921533/esp32-hardware-isr-sometimes-not-triggered-when-wifi-is-transmitting)) for your high throughput system.

Let me know the results of this. Thanks.

alisitsyn, 2024-09-06:
@tryger ,

Have you been able to solve your issue? The issue will be closed because of no feedback. Feel free to reopen if you have results or comments.

closed 2024-09-06 by alisitsyn