espressif/esp-idf #13272: On the Fly OTA Update Secure Bootloader V1 on ESP32

filzek, 2024-02-27:
### Is your feature request related to a problem?

On old devices running in field and with the advent of new SDK V5.2, there is a lot of new features that will only work now on, so old boards start to become obsolete if the new approach of memory mapping and IRAM usage become used to develop new firmwares, so, it will be necessary to create somehow an option to update the bootloader on the fly on SECURE BOOT V1 boards.

It could be very important to create an way to write a new signed bootloader with the original pem file and could be used to an OTA update rewrite the old bootloader, so, this could solve all developers problem at once.

### Describe the solution you'd like.

Update Secure Boot V1 on ESP32 over OTA if the original PEM sign file is avaliable.

ESP-Marius, 2024-02-28:
Hi,

Currently we dont any plans to support updating the bootloader over OTA, since if anything were to go wrong during this update you would effectively brick your device. So any such feature would involve considerable risk.

KonstantinKondrashov, 2024-02-28:
Hi,
Yes, we have a task to add a way to update the bootloader (it has not been done yet), but as Marius said it would involve considerable risk. In any case, we want to provide it, even with some caveats.

filzek, 2024-03-08:
That shall be perfect to solve some problems, please let us know when its ready and we can private test in various scenarios as well.

We have done already a repartition encrypted on the fly, thats was not so difficult, but was challenging as well.

KonssnoK, 2024-04-10:
we will also be interested in this feature, following the thread.

KonstantinKondrashov, 2024-12-20:
Hi! The unsafe OTA bootloader update is supported [97d150d](https://github.com/espressif/esp-idf/commit/97d150d69af37a427a5c89ddae040540fba0ccbb). It can be used only for Secure boot V2. 
Sorry but the bootloader update with Secure Boot V1, is not supported.

closed 2024-12-20 by KonstantinKondrashov