espressif/esp-idf #13275: NimBLE msys and mbufs on ESP32C6

spikeyamk, 2024-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I can't seem to figure out how to send packets over BLE with ble_gatts_notify_custom at full speed. When I try and use

int ble_hs_mbuf_to_flat(const struct os_mbuf *om, void *flat, uint16_t max_len,
                        uint16_t *out_copy_len);

this function fails if the msys is full. But when it fails it seems like the whole NimBLE fails and I can't send any more packets over BLE no matter what. I've looked at the example here: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/throughput_app/bleprph_throughput and they use this function there:

if (os_msys_num_free() >= MIN_REQUIRED_MBUF) {

To detect whether there's enough free blocks in the msys for another packet to be fed into msys. But this function doesn't seem to do much on my EPS32C6. It just reports total number of blocks MSYS1 + MSYS2 specified by me in the sdkconfig . The value it returns does in fact decrease after I fill the msys with data but at that point it's already too late and everything fails.

Can anyone suggest a fix for this?

Thanks in advance!

rahult-github, 2024-02-28:
Hi @spikeyamk , 

I think this use case hits the scenario mentioned as part of example scope [here](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/throughput_app/bleprph_throughput#example-scope) 

basically, application is pumping data at a rate faster than the rate at which the buffers are getting freed.  I suggest to put a check in application to wait for some random time in case there are no buffers available and re-check if buffers are available. If not, then continue waiting again. The buffers which were allocated while sending previous notification will eventually get freed up and you will get chance to schedule new packet again.   

A similar implementation can be referred [here](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/throughput_app/blecent_throughput/main/main.c#L120) .

spikeyamk, 2024-02-28:
I'm not sure you read my post through or not so I'm gonna copy this over:

I've looked at the example here: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/throughput_app/bleprph_throughput and they use this function there:

if (os_msys_num_free() >= MIN_REQUIRED_MBUF) {

To detect whether there's enough free blocks in the msys for another packet to be fed into msys. But this function doesn't seem to do much on my EPS32C6. It just reports total number of blocks MSYS1 + MSYS2 specified by me in the sdkconfig . The value it returns does in fact decrease after I fill the msys with data but at that point it's already too late and everything fails.

rahult-github, 2025-08-26:
This [commit](https://github.com/espressif/esp-idf/commit/3f2f57a3cc06029d52033cc877ad68b8c7a30d51) introduced the change to ensure what you get from this API is now a reference on how many msys buffer left.