espressif/esp-idf #13276: High Resolution Timer (ESP Timer) BUG! Timer related migrated from TIMER_GROUP_1/TIMER_0 and TIMER_GROUP_0/TIMER_0 has dispatch problems

filzek, 2024-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

SDK Release V5.2

### Espressif SoC revision.

Esp32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32

### Power Supply used.

USB

### What is the expected behavior?

Timer group has been disable on SDK v5.2, but the High Resolution Timer (ESP Timer) doesnt act exact like the Timer Group on the old SDK V4, so, the new High Resolution Timer (ESP Timer) has dispatch problems for extreme accurate dispatch, it does not act in the behavior of the dispatch of the timer group, so it is not accurate and miss some dispatches make the application impossible to run correcly.

The scenarios running the dispatch is quite simple.

One timer must run after the Zero Crossing act on interrupt being dispatched.

Gpio connected to a Zero Crossing reading and filtered to only enable after minimum of 7500 read filtering any bad dispatch. 

A true wave of in 60hz is 8333us.

So, if a dimer relays onm 20%, it will takes 20% of 8333 to arm, if we are cautelous and have a filter of 7500 read, we can set it to a 7500 do the sum, so 20%of 7500, about 1500us to arm.

A timer is deployed to arm the triac in 1500us.
a second timer id deployed to disarm the triac in 7500us.

The reading from the GPIO ISR DISPATCH and all other functions are in IRAM, so safe to run, and the GPIO enable in the triacs are done direct by the registers to be quite fast, de ISR are complete simple and fast.

What happen is that some of the timers arent being completed, the scenario become worst when ble/wifi is enable, and mucher worst when websocket are enable.

This doesnt happen on SDK V3 or V4(4.4.6), so, the problem after port the fw to the new SDK bring anothers timers problem.

### What is the actual behavior?

run the timer correct dispatched in ISR without miss any dispatchs

### Steps to reproduce.

same as in the main part.

igrr, 2024-02-27:
Please check the migration guide for the timer group driver: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/peripherals.html#timer-group-driver.
You can still use the timer group driver by enabling [CONFIG_GPTIMER_SUPPRESS_DEPRECATE_WARN](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-gptimer-suppress-deprecate-warn). If not using this option, please use the new [gptimer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gptimer.html) driver.

(esp_timer, as you have found, will not meet your requirement for very accurate timing. It is a software timer used by multiple libraries, so the execution time of your esp_timer callbacks will inevitably be affected by the other libraries.)

filzek, 2024-03-08:
GPTIMER works great but the rearm for a next alarm is not working properly, no matter how hard to try, simple doenst work as intent.

unique way to create a new rearm for the timer will be using 

gptimer_stop(gptimer);
gptimer_set_raw_count(gptimer,0);

and now set the new counting as well, 

and start the timer again, 

Docs need some fixes or clarification about it

KonstantinKondrashov, 2024-03-27:
Fill free to open a PR for that. I close this issue for now as there are no actions needed here. 
FYI, recently, the esp_timer doc was rewritten, and added Troubleshooting article that covers this case as well - https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html#troubleshooting.
Thanks.

closed 2024-03-27 by KonstantinKondrashov