espressif/esp-idf #13277: Unable to build the gatt_server_service_table example

ashinempark, 2024-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The gatt_server_service_table example project builds successfully for the ESP32 S3 target

### What is the actual behavior?

The gatt_server_service_table example project is failing to build for the ESP32 S3 target. I am running into linker errors, which complains about undefined references to "r_lld_test_stop", "r_lld_adv_start_update_filter_policy" and "r_rwip_wakeup_end". However, the example builds just fine for the ESP32 C2 target.

### Steps to reproduce.

Build the default gatt_server_service_table example for the ESP32 S3 target without any modifications

### Build or installation Logs.

```plain
/Users/ashish/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/ashish/Documents/ESP/esp-idf/components/bt/controller/lib_esp32c3_family/esp32s3/libbtdm_app.a(ip_funcs.o):(.rodata.r_ip_funcs_ro+0x21c): undefined reference to `r_lld_test_stop'
/Users/ashish/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/ashish/Documents/ESP/esp-idf/components/bt/controller/lib_esp32c3_family/esp32s3/libbtdm_app.a(ip_funcs.o):(.rodata.r_ip_funcs_ro+0x784): undefined reference to `r_lld_adv_start_update_filter_policy'
/Users/ashish/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/ashish/Documents/ESP/esp-idf/components/bt/controller/lib_esp32c3_family/esp32s3/libbtdm_app.a(modules_funcs.o):(.rodata.r_modules_funcs_ro+0x2e4): undefined reference to `r_rwip_wakeup_end'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### More Information.

I tried build using both the VS Code IDE and on command line using idf.py build.
I tried different ESP-IDF releases (v5.2, v5.1, v5.0, v4.4, latest master).
The same issue exists in all cases.

hibfh, 2024-02-28:
check more . I build it without any problem.
[image]

zhp0406, 2024-03-10:
Please use the following command:

```
git submodule update
```

Alvin1Zhang, 2024-03-10:
Thanks for reporting, feel free to reopen.

closed 2024-03-10 by Alvin1Zhang

ashinempark, 2024-03-10:
Thanks for your responses. Yes, I re-updated my workspace and it works now.