espressif/esp-idf #13279: [ble mesh] esp_ble_mesh_init stuck

chegewara, 2024-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5f4249357372f209fdd57288265741aaba21a2b1

### Espressif SoC revision.

esp32-s3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3-devkit

### Power Supply used.

USB

### What is the expected behavior?

To get esp_err_t error

### What is the actual behavior?

`esp_ble_mesh_init` in some cases, when one of parameters is wrong (not sure which one) function stuck and never returns.

### Steps to reproduce.

Im not entirely sure. My library is auto-generating composition elements, so i am guessing if for some reson `composition.elements` is incorrect then it stuck

chegewara, 2024-03-03:
I think its happening when there is a mismatch between number of elements and `element_count` passed in this config, especially when for some reason `el_cnt` is big, like 60 and only 3 elements:
```
    composition = {
        .cid = CID_ESP,
        .element_count = el_cnt,
        .elements = sig_mesh_elements,
    };
```

Ity happened to me when `el_cnt` is not initialized in C++, so it can have random value. 
I know its a bug in my code, but it should not cause this function to stuck forever.

chegewara, 2024-03-14:
Hi,
with vendor model i dont see any of this events when sending/receiving publish
```
ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT
ESP_BLE_MESH_CLIENT_MODEL_RECV_PUBLISH_MSG_EVT
```

forx157, 2024-03-18:
Hi, @chegewara 
In your first question, are you saying that you gave `sig_mesh_elements` a different length than `el_cnt`, which in turn caused the problem?
In your second question, Is it convenient to say what kind of functionality you want to achieve with `esp_ble_mesh_model_publish`? How do you use `esp_ble_mesh_model_publish`?

chegewara, 2024-03-18:
Hi,

> Hi, @chegewara In your first question, are you saying that you gave `sig_mesh_elements` a different length than `el_cnt`, which in turn caused the problem?

Yes, although i understand the problem is on my side, and API is just trying to do its best to build composition data from corrupted information and most likely stuck in endless loop. So, i dont think its a bug anymore

> In your second question, Is it convenient to say what kind of functionality you want to achieve with esp_ble_mesh_model_publish? How do you use esp_ble_mesh_model_publish?

After spending some time with API i figured out its not an issue, i guess, just a publish functionality not implemented in some models yet (generic location in this case).

closed 2024-03-21 by chegewara