espressif/esp-idf #13280: Boot detects that the reset reason is 3 after watchdog in app is triggerred

ouer1224, 2024-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi teams,
I am using **esp32s3 evk** to test my code. 
**The first issue**:
I noticed a strange phenomenon: if I add a `while(1)` code in my app_main() function, the watdog will be triggerred as expected. 
But `esp_rom_get_reset_reason()` in `bootloader_check_wdt_reset()` will return 3. It means the reset is caused by software.
I tested it in "task_watchdog" of example if I enable  "CONFIG_ESP_TASK_WDT_PANIC" . I got the same result. 

What I expect is that boot can detect the reset reason is **watch dog** if watch dog in app is triggerred. I need to process some logic when watch dog is triggerred in APP.
How can I achieve this?
note: I saw someone's reset reason is `TG1WDT_SYS_RESET` if his watchdog is triggerred in the link [TG1WDT_SYS_RESET](https://www.esp32.com/viewtopic.php?f=25&t=36525&p=122712&hilit=%E7%9C%8B%E9%97%A8%E7%8B%97#p122712)
Thank you.
[image]
[image]
[image]

**The second issue**:
I try to feed watchdog in bootloader by calling `bootloader_super_wdt_auto_feed()`.
But the watchdog still be triggerred.
How can I feed the watchdog in bootloader?
Thank you.
note: `S3_bootloader_super_wdt_auto_feed()` is same as `bootloader_super_wdt_auto_feed()` 
[image]
[image]

ESP-Marius, 2024-02-28:
> But esp_rom_get_reset_reason() in bootloader_check_wdt_reset() will return 3. It means the reset is caused by software.

This is as expected. `esp_rom_get_reset_reason` checks the hardware reset reason (see [TRM reset and clock chapter](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf) for these), but when you get a panic from task WDT it is not actually the WDT HW that resets the chip. Rather we catch the panic, and the chip is then restarted by the panic handler (which is why you see the software reset).

The HW reset from the WDT would only trigger if the system locks up and for some reason cannot enter the panic handler. Then the second stage timeout of the WDT HW will trigger a HW reset.

In IDF when we do a SW restart we also store the reason for this restart in an additional RTC registers, you can use this to determine if the restart was due to a WDT. See [esp_reset_reason()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#reset-reason)

> I try to feed watchdog in bootloader by calling bootloader_super_wdt_auto_feed().

Auto-feed is not the function for manually feeding the SWD WDT, but rather a one-time function for activiting the automatic feeding of this WDT (i.e. feeding without any involvement from the CPU). I think my calling this one continuously it might not behave as expected.  This one is actually already called at the begining of `bootloader_init()`, so you should not have to call it again.

ouer1224, 2024-02-28:
> The HW reset from the WDT would only trigger if the system locks up and for some reason cannot enter the panic handler. Then the second stage timeout of the WDT HW will trigger a HW reset.

I disable `CONFIG_ESP_TASK_WDT_PANIC` in the "task_watchdog" project of example. When the watchdog is triggerred, mcu always to output the message of watchdog timeout handler, will not reset into bootloader. See the "wdg in app.txt" for the detail.

> This one is actually already called at the begining of bootloader_init(), so you should not have to call it again.

I analyzed the bootloader code, and noticed  `bootloader_init()` calls `bootloader_super_wdt_auto_feed()`. And I found bootloader will reboot if I add a `while(1)` in `call_start_cpu0()`. And I re-test the code, It still reseted by RTC watchdog. See the "wdg in boot.txt" for the detail.

So can you these two case in your side to check if the result is as you expect?

**Note**:  Please see `task_watchdog_example_main.c` and `bootloader_start.c` for the code modified.
[wdg in boot.txt](https://github.com/espressif/esp-idf/files/14429158/wdg.in.boot.txt)
[wdg in app.txt](https://github.com/espressif/esp-idf/files/14429164/wdg.in.app.txt)
[bootloader_start.c.txt](https://github.com/espressif/esp-idf/files/14429167/bootloader_start.c.txt)
[task_watchdog_example_main.c.txt](https://github.com/espressif/esp-idf/files/14429168/task_watchdog_example_main.c.txt)

ESP-Marius, 2024-02-28:
> I disable CONFIG_ESP_TASK_WDT_PANIC in the "task_watchdog" project of example. When the watchdog is triggerred, mcu always to output the message of watchdog timeout handler, will not reset into bootloader. 

Yes, the task WDT will not reset the chip if this is disabled, it will only print a message.

> And I re-test the code, It still reseted by RTC watchdog.

RTC and Super WDT are two separate WDTs [see TRM chapter about Watchdogs](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf) for more info. If you want to manually feed the RTC WDT you can see an example [here](https://github.com/espressif/esp-idf/blob/5f4249357372f209fdd57288265741aaba21a2b1/components/bootloader_support/src/flash_encryption/flash_encrypt.c#L451) and [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/hardware-abstraction.html#to-reset-the-rtc-wdt-counter).

For more general info about the different WDTs I recommend reading through [WDT - Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/wdts.html) if you havent done so yet.

ouer1224, 2024-02-29:
Thanks for your answer.
I used the following code to feed RTC watchdog successfully.
```
void S3_bootloader_super_wdt_auto_feed(void)
{
    wdt_hal_context_t rtc_wdt_ctx = RWDT_HAL_CONTEXT_DEFAULT();
    wdt_hal_write_protect_disable(&rtc_wdt_ctx);
    wdt_hal_feed(&rtc_wdt_ctx);
    wdt_hal_write_protect_enable(&rtc_wdt_ctx);
}
``` 

But about "[esp_reset_reason()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#reset-reason)", I met an issue.
I can't call `esp_reset_reason()` in bootloader directly. The building log shows "can't find  `esp_system.h` file" when I call `#include "esp_system.h"` in my bootloader code. So I copy a `s3_esp_reset_reason_get_hint()` function as following, and then call it in my bootloader.
```
uint32_t s3_esp_reset_reason_get_hint(void)
{
    uint32_t reset_reason_hint = REG_READ(RTC_RESET_CAUSE_REG);
    uint32_t high = (reset_reason_hint >> RST_REASON_SHIFT) & RST_REASON_MASK;
    uint32_t low = reset_reason_hint & RST_REASON_MASK;
    if ((reset_reason_hint & RST_REASON_BIT) == 0 || high != low) {
        return ESP_RST_UNKNOWN;
    }
    return (uint32_t) low;
}
```
This code can be build successfully. But after the watchdog in  my "task_watchdog" project is triggerred and MCU reset into bootloader, this function return **0**. 
Then I did some other test. I called `esp_reset_reason()` in my app. After watchdog is triggerred, `s3_esp_reset_reason_get_hint()` of bootloader return **6**. And `esp_reset_reason()` of app also returned **6**.

It seems that I must call `esp_reset_reason()` in my app, otherwise `s3_esp_reset_reason_get_hint()` of bootloader will not work correctly.

My bootloader needs to know the reset source is 6.   Are there some errors in my bootloader code?

ESP-Marius, 2024-03-01:
> I can't call esp_reset_reason() in bootloader directly. The building log shows "can't find esp_system.h file" when I call

Ah yeah, not all API is linked into the bootloader

>  So I copy a s3_esp_reset_reason_get_hint() function as following, and then call it in my bootloader.

This looks correct to me, and I also tested in on my end:

```
Hello world!
E (5320) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5320) task_wdt:  - IDLE0 (CPU 0)
E (5320) task_wdt: Tasks currently running:
E (5320) task_wdt: CPU 0: main
E (5320) task_wdt: CPU 1: IDLE1
E (5320) task_wdt: Aborting.
E (5320) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x420089b2:0x3fc97620 0x42017d03:0x3fc97640 0x4037a18d:0x3fc97670
0x420089b2: app_main at /home/marius/esp-idf_2/examples/get-started/hello_world/main/hello_world_main.c:18
0x42017d03: main_task at /home/marius/esp-idf_2/components/freertos/app_startup.c:208
0x4037a18d: vPortTaskWrapper at /home/marius/esp-idf_2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

Rebooting...
��I (23) boot: ESP-IDF v5.3-dev-2298-gc465018bc2-dirty 2nd stage bootloader
I (23) boot: compile time Mar  1 2024 14:35:37
I (24) boot: Multicore bootloader
E (28) boot: reset 6
```

I successfully get 6 as the reset reason when using `s3_esp_reset_reason_get_hint` in the bootloader. So it should work. I suggest double checking your code to see if there is any mistakes. And if you cant find any then I can also take a look if you can   upload an example/patch showing this behavior

ouer1224, 2024-03-04:
I re-tested my product code and other example project. I noticed a strange phenomenon. In my product code, the bootloader can get 6 as the reset reason. But in 'hello world' project and 'watchdog task' project, the same bootloader can only get 0 as the reset reason if the app didn't call `esp_reset_reason()` in app code.
I double check my `s3_esp_reset_reason_get_hint ()` code , and the code which calls `s3_esp_reset_reason_get_hint()`. But I found nothing unusual.

I uploaded my 'hello world' project and the 'bootloader_esp32s3.c'. I don't know whether the problem is the app sdkconfig or the bootloader code.

[hello_world.tar.gz](https://github.com/espressif/esp-idf/files/14477299/hello_world.tar.gz)
[bootloader_esp32s3.c.txt](https://github.com/espressif/esp-idf/files/14477304/bootloader_esp32s3.c.txt)

ESP-Marius, 2024-03-04:
Strange, I just copied your `bootloader_esp32s3.c` and ran your hello world, and I get the expected result: `reset = 0` for the first boot up and then `I (46) reset source:  !!!!reset source = 6` for every WDT reset afterwards.

Which IDF commit/release are you using? I tested with master, so maybe that is what is different?

Could you also post the whole log from your hello world example?

ouer1224, 2024-03-04:
Hi, please see the following log for esp-idf version.
```
virtual-machine ~/c/hello_world> idf.py --version
/home/ouer/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
ESP-IDF v5.1-5-gc43e7aeeba-dirty
```
And the attachment is the full log. I checked my log, I can only get 0 which is different from your side.
[hello.log](https://github.com/espressif/esp-idf/files/14477531/hello.log)

ESP-Marius, 2024-03-04:
Ah, I think I see what is happening. 

We provide a weak version of `esp_reset_reason_set_hint` [here](https://github.com/espressif/esp-idf/blob/master/components/esp_system/panic.c#L477) to explicitly not link in reset reason if the reset reason API is not used.

So when you dont call `esp_reset_reason` the linker just discards the stuff from `reset_reason.c` and the reset reason never gets set.

So the easy work-around is to just make sure you also use this API from your app to force it to get included.

Seems like this behavior is different on master, which is why I didnt see the issue. I'll check why.

ouer1224, 2024-03-04:
Thank you. I will temporarily to call `esp_reset_reason()` in my app.
I did a test. I modified `esp_reset_reason_get_hint()` in panic file.
```
esp_reset_reason_t IRAM_ATTR  __attribute__((weak)) esp_reset_reason_get_hint(void)
{
    return 1;//ESP_RST_UNKNOWN;
}
```
And I still got 0.

ESP-Marius, 2024-03-04:
> I modified esp_reset_reason_get_hint() in panic file.

Yes, modifying the `get` function is not sufficient, the issue is that we call `esp_reset_reason_set_hint` from the panic file isntead of reset reason. So the register which is supposed to contain the reset reason (the one you read in the bootloader) simply never gets set.

ESP-Marius, 2024-05-31:
Marking this one as resolved. Feel free to reopen if you think there is still anything that can be considered a IDF bug here.

closed 2024-05-31 by ESP-Marius