espressif/esp-idf #13281: About flash_download_tool_3.9.4 flash encryption esp8684(esp32-c2)

13248519492a, 2024-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How to set security.conf to do the following
1.Use your own key instead of a software generated key
2.Flash encryption only
3.Can write ciphertext firmware and execute it
My setup is not able to fulfil the requirements and reports errors

13248519492a, 2024-02-28:
[image]

13248519492a, 2024-02-28:
[image]

13248519492a, 2024-02-28:
[image]
[image]
[image]

13248519492a, 2024-02-28:
[image]
[image]

arosyc, 2024-03-07:
Hi,
We will release version 3.9.6 recently (about two weeks ) to fix the ESP32-C2 encryption problem. You can refer to the latest documentation for operation.

mahavirj, 2024-03-11:
@13248519492a 

Interim, you may use the IDF tools for enabling flash encryption on ESP32-C2. Please refer to this guide https://docs.espressif.com/projects/esp-idf/en/latest/esp32c2/security/host-based-security-workflows.html#enable-flash-encryption-externally for externally generated flash encryption key workflow.

I will close this issue for now.