espressif/esp-idf #13283: UART communication error when dynamic frequency scaling enabled

MacWyznawca, 2024-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2, 5.1.2, 5.1.3

### Espressif SoC revision.

ESP32  chip revision: v3.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Sonoff Dual R3 

### Power Supply used.

External 5V

### What is the expected behavior?

Stable communication via UART with BL09x chip

```c
uart_config_t uart_config = {
        .baud_rate = 4800,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1_5,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_DEFAULT, // UART_SCLK_REF_TICK, UART_SCLK_DEFAULT
    };
```

### What is the actual behavior?

With above config and Power Management with freq. scaling enable I have UART Break, and lots of communication error and checksum error.

I Try  **UART_SCLK_REF_TICK** and **UART_SCLK_DEFAULT**, same problems

With PM disabled, all works great.

### Steps to reproduce.

Enable dynamic frequency scaling and try UART with changed CPU frequency.
Things going worse if main CPU freq is higher than 80 MHz.

### Debug Logs.

```plain
E (22607) BL09xx: BL0939 Data WRONG!
I (22639) LuServBlind: Move update cur:97, tar:0, dir:0, startT:2106, curT:2208, endT:4768, Pow 255, P1: 0, P2: 0
I (22639) LuSend: Sending by Line mediun: 15, e: ESP_FAIL, line: -1
I (22654) LuSend: Sending by Air 1, e: ESP_OK
I (22654) LuSend: Send OK, LuID: 2 : 134->10 Z:4 Id: 33 L:ESP_FAIL/A:ESP_OK
I (22825) BL09xx: [UART DATA]: 33
E (22826) BL09xx: BL0939 Data WRONG!
I (23051) BL09xx: [UART DATA]: 35
E (23052) BL09xx: BL0939 Data WRONG!
I (23266) BL09xx: [UART DATA]: 33
E (23267) BL09xx: BL0939 Data WRONG!
I (23484) BL09xx: [UART DATA]: 33
E (23485) BL09xx: BL0939 Data WRONG!
I (23637) LuServBlind: Move update cur:93, tar:0, dir:0, startT:2106, curT:2308, endT:4768, Pow 255, P1: 0, P2: 0
I (23638) LuSend: Sending by Line mediun: 15, e: ESP_FAIL, line: -1
I (23653) LuSend: Sending by Air 1, e: ESP_OK
I (23654) LuSend: Send OK, LuID: 2 : 134->10 Z:4 Id: 34 L:ESP_FAIL/A:ESP_OK
I (23704) BL09xx: [UART DATA]: 34
E (23705) BL09xx: BL0939 Data WRONG!
I (23924) BL09xx: [UART DATA]: 32
E (23925) BL09xx: BL0939 Data WRONG!
I (24365) BL09xx: [UART DATA]: 34
E (24366) BL09xx: BL0939 Data WRONG!
I (24584) BL09xx: [UART DATA]: 34
E (24585) BL09xx: BL0939 Data WRONG!
```

AxelLin, 2024-08-05:
Any update from espressif?

MacWyznawca, 2024-08-05:
I have no new information :-( @AxelLin