espressif/esp-idf #13284: Flash Encryption invalid header: 0x6b6df5f7

srikanthpalvai, 2024-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I am trying to enable the Flash encryption and secure boot feature for my ESP32.
But I ended up with the invalid header: 0x6b6df5f7.

Here are the steps followed.
Steps followed on esp-idf-v5.1.2\examples\security\flash_encryption example

1. idf.py set-target esp32
2. idf.py menuconfig -> Enable Security features -> Bootloader Reflashbale, Flash Encryption Development & Didn't touch the Potentially insecure options.
3. idf.py build
-> Got error  
ninja: error: 'C:/esp/Espressif/frameworks/esp-idf-v5.1.2/examples/security/flash_encryption/secure_boot_signing_key.pem', needed by 'signature_verification_key.bin', missing and no known rule to make it
-> Generated secure_boot_signing_key.pem file
espsecure.py generate_signing_key secure_boot_signing_key.pem
4. idf.py build
5.  espefuse.py --port COM5 burn_efuse FLASH_CRYPT_CONFIG 0x0F
6.  espefuse.py --port COM5 burn_efuse FLASH_CRYPT_CNT
7. As per the build log. Followed the 
esptool.py --chip esp32 --port=(PORT) --baud=(BAUD) --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 C:/esp/Espressif/frameworks/esp-idf-v5.1.2/examples/security/flash_encryption/build/bootloader/bootloader.bin

esptool.py -p (PORT) -b 460800 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0xd000 build\partition_table\partition-table.bin 0x20000 build\flash_encryption.bin

8. idf.py -p COM5 monitor
-> Got error 
Executing action: monitor
Running idf_monitor in directory C:\esp\Espressif\frameworks\esp-idf-v5.1.2\examples\security\flash_encryption
Executing "C:\esp\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\esp\Espressif\frameworks\esp-idf-v5.1.2\tools/idf_monitor.py -p COM5 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 C:\esp\Espressif\frameworks\esp-idf-v5.1.2\examples\security\flash_encryption\build\flash_encryption.elf --force-color -m 'C:\esp\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe' 'C:\esp\Espressif\frameworks\esp-idf-v5.1.2\tools\idf.py' '-p' 'COM5'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM5 instead...
--- esp-idf-monitor 1.3.3 on \\.\COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

    rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7
    ets Jul 29 2019 12:21:46
    
    rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7
    invalid header: 0x6b6df5f7

9. espefuse.py -p COM5 summary

Flash fuses:
FLASH_CRYPT_CNT (BLOCK0)                           Flash encryption is enabled if this field has an o = 127 R/W (0b1111111)
dd number of bits set
FLASH_CRYPT_CONFIG (BLOCK0)                        Flash encryption config (key tweak bits)           = 15 R/W (0xf)

 
[Burn secure boot key to efuse.txt](https://github.com/espressif/esp-idf/files/14432001/Burn.secure.boot.key.to.efuse.txt)
[First time flash command.txt](https://github.com/espressif/esp-idf/files/14432002/First.time.flash.command.txt)
[image]
[idf build.log](https://github.com/espressif/esp-idf/files/14432003/idf.build.log)
[image]
[image]
[image]
[Summary.txt](https://github.com/espressif/esp-idf/files/14432004/Summary.txt)

For reference, I have followed the below links to resolve, but no luck:
https://github.com/espressif/esp-idf/issues/10927
https://github.com/espressif/arduino-esp32/issues/1387
https://medium.com/@kattelroshan1/esp32-firmware-encryption-a53eb1c9bf9c

Tried disabling Flash Encryption.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#disabling-flash-encryption

As It can only be done three times per chip,  So tried like below,

[image]

[image]

@mahavirj Could you please help me here?

AdityaHPatwardhan, 2024-03-01:
Hi @srikanthpalvai 
From looking at the eFuse summary. It seems that the flash encryption key is not burned for the device and FLASH_CRYPT_CNT is not set as well.
But the commands that you have added afterwards show that the FLASH_CRYPT_CNT has been burned to 127 (Max value possible)

If I understand correctly, FLASH_CRYPT_CNT is burned but Flash encryption key is not burned.
Please checkout this host based workflow
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/host-based-security-workflows.html#enable-flash-encryption-externally
It may be the reason why you are getting the invalid header error.

Can you go through the process once again?
You may skip the FLASH_CRYPT_CNT burning process if it is not required for you. Please make sure to sign the binaries before flashing them.

AdityaHPatwardhan, 2024-04-02:
Closing the issue due to lack of response.
@srikanthpalvai Please re-open if the above comment does not help you to solve your issue.
Thanks,
Aditya