espressif/esp-idf #13285: [ESP32S3] v5.1 consuming up to 30kB of internal memory more than same example in v4.4

KonssnoK, 2024-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

this was tested on esp32s3 devkit-C
using 
https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network

Changes on v4.4

```
diff --git a/examples/mesh/ip_internal_network/main/mesh_main.c b/examples/mesh/ip_internal_network/main/mesh_main.c
index 72e4a82d4c..92bbb1d4ce 100644
--- a/examples/mesh/ip_internal_network/main/mesh_main.c
+++ b/examples/mesh/ip_internal_network/main/mesh_main.c
@@ -383,6 +383,8 @@ void ip_event_handler(void *arg, esp_event_base_t event_base,

 void app_main(void)
 {
+    heap_caps_malloc_extmem_enable(0);
+
     ESP_ERROR_CHECK(nvs_flash_init());
     /*  tcpip initialization */
     ESP_ERROR_CHECK(esp_netif_init());
@@ -422,6 +424,6 @@ void app_main(void)
     ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
     /* mesh start */
     ESP_ERROR_CHECK(esp_mesh_start());
-    ESP_LOGI(MESH_TAG, "mesh starts successfully, heap:%d, %s\n",  esp_get_free_heap_size(),
+    ESP_LOGI(MESH_TAG, "mesh starts successfully, internal heap:%" PRId32 ", %s\n", esp_get_free_internal_heap_size(),
              esp_mesh_is_root_fixed() ? "root fixed" : "root not fixed");
 }
```

sdkconfig
```
CONFIG_ESP32S3_SPIRAM_SUPPORT=y

#
# SPI RAM config
#
# CONFIG_SPIRAM_MODE_QUAD is not set
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_TYPE_AUTO=y
# CONFIG_SPIRAM_TYPE_ESPPSRAM64 is not set
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_CLK_IO=30
CONFIG_SPIRAM_CS_IO=26
# CONFIG_SPIRAM_FETCH_INSTRUCTIONS is not set
# CONFIG_SPIRAM_RODATA is not set
# CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY is not set
CONFIG_SPIRAM_SPEED_80M=y
# CONFIG_SPIRAM_SPEED_40M is not set
CONFIG_SPIRAM=y
CONFIG_SPIRAM_BOOT_INIT=y
# CONFIG_SPIRAM_IGNORE_NOTFOUND is not set
# CONFIG_SPIRAM_USE_MEMMAP is not set
# CONFIG_SPIRAM_USE_CAPS_ALLOC is not set
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=2048
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
# end of SPI RAM config
```

[... 304 lines cut ...]

I (1143) phy_init: phy_version 640,cd64a1a,Jan 24 2024,17:28:12
W (1153) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1203) wifi:mode : softAP (7c:df:a1:e1:95:05)
I (1213) wifi:Total power save buffer number: 8
I (1213) wifi:Init max length of beacon: 752/752
I (1213) wifi:Init max length of beacon: 752/752
I (1213) wifi:mode : sta (7c:df:a1:e1:95:04) + softAP (7c:df:a1:e1:95:05)
I (1223) wifi:enable tsf
I (1223) mesh: <nvs>read layer:0, err:0x1102
I (1223) mesh: <nvs>read assoc:0, err:0x1102
I (1783) wifi:Total power save buffer number: 8
I (2333) wifi:mode : sta (7c:df:a1:e1:95:04)
I (2343) mesh: <MESH_NWK_LOOK_FOR_NETWORK>need_scan:0x3, need_scan_router:0x0, look_for_nwk_count:1
I (2343) mesh_main: <MESH_EVENT_MESH_STARTED>ID:77:77:77:77:77:76
I (2343) mesh_main: mesh starts successfully, internal heap:214471, root not fixed

```

See last line of log:
v4.4: 245207 bytes free
v5.1: 214471 bytes free
-> delta ~ 30.7k

FYI @zhangyanjiaoesp

dobairoland, 2024-02-28:
This is just a tip to narrow down the source of the difference: You can use esp-idf-size (https://github.com/espressif/esp-idf-size) (already available in ESP-IDF environments) and with the `--diff` argument to see the difference between the two projects.

```sh
python -m esp_idf_size.ng build/blink_v5.1.map --diff /tmp/blink_v4.4.map
```

You can add arguments to get more details. See `python -m esp_idf_size.ng --help`.

KonssnoK, 2024-02-28:
> This is just a tip to narrow down the source of the difference: You can use esp-idf-size (https://github.com/espressif/esp-idf-size) (already available in ESP-IDF environments) and with the `--diff` argument to see the difference between the two projects.
> 
> ```shell
> python -m esp_idf_size.ng build/blink_v5.1.map --diff /tmp/blink_v4.4.map
> ```
> 
> You can add arguments to get more details. See `python -m esp_idf_size.ng --help`.

hello @dobairoland , we are talking about internal RAM usage, not file size. I'll update the name to reflect that

cbaechler, 2024-03-01:
After quite some debugging I think I have found the difference between `release/v4.4` and `release/v5.1` that is causing the above mentioned behavior. 

While in `v4.4` the FreeRTOS malloc was simply defined as `#define pvPortMalloc malloc`, in `v5.1` it is now defined as 

```c
#define portFREERTOS_HEAP_CAPS    ( MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT )

/*-----------------------------------------------------------*/

void * pvPortMalloc( size_t xWantedSize )
{
    void * pvReturn = NULL;

    /* All dynamic allocation done by FreeRTOS goes through this function. If
     * users need to allocate FreeRTOS objects into external RAM, they should
     * use the "static" equivalents of FreeRTOS API to create FreeRTOS objects
     * (e.g., queues). */
    pvReturn = heap_caps_malloc( xWantedSize, portFREERTOS_HEAP_CAPS );

    return pvReturn;
}
```

which forces the allocations to happen on the internal RAM. 

By placing logs in `esp_wifi/esp32s3/esp_adapter.c` I can observe that calling `esp_mesh_start()` creates a queue of 96 elements with 184 bytes each (resulting in an allocation of 17664 bytes). By changing the code in `queue_create_wrapper` I managed to allocate the storage of the queue in SPIRAM: 

```diff
- return (void *)xQueueCreate(queue_len, item_size);
+ return (void *)xQueueCreateWithCaps(queue_len, item_size, MALLOC_CAP_SPIRAM);
```

I tried to apply a similar modification to `task_create_wrapper`, however it fails on an assertion when creating a task with the name `wifi`:
```
assert failed: spi_flash_disable_interrupts_caches_and_other_cpu cache_utils.c:154 (esp_task_stack_is_sane_cache_disabled())
```
I assume this is related to the explanation in the code, but I am not sure I 100% understand the comment

```c
/*
 * All dynamic allocation done by FreeRTOS should be placed in internal 8-bit
 * accessible RAM (i.e., using the MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT flags).
 * This is due to the fact that FreeRTOS objects (e.g., task stacks, TCBs,
 * queues etc) must be accessible even if the cache is disabled. Therefore, the
 * heap that is made available to FreeRTOS for dynamic allocation is a subset of
 * the ESP-IDF heap (where all MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT memory is
 * made available to FreeRTOS for dynamic allocation).
 * */
```

KonssnoK, 2024-03-01:
what is not clear is why we can write to spi_flash in v4.4 while having tasks allocated in the spi_ram and we cannot do the same in 5.1.

For sure there are no hardware limitations, or we would not have been abled in the first place to put our RTOS stuff on SPI_RAM.

zhangyanjiaoesp, 2024-03-04:
@dobairoland @Dazza0 PTAL

Dazza0, 2024-03-04:
@KonssnoK Please take a look at the FreeRTOS section of the IDF v5.1 [migration guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.1/system.html#dynamic-memory-allocation). All dynamic memory allocated by FreeRTOS will not default to be being internal memory. This change was intentional in to increase run time safety.

For library/user code that need to move memory allocations to external memory. This now needs to be done explicitly (see the `CreateWithCaps()` API mentioned in the migration guide).

KonssnoK, 2024-03-04:
> @KonssnoK Please take a look at the FreeRTOS section of the IDF v5.1 [migration [...]

Hello @Dazza0 , could you please be more specific on what you mean with "increase runtime safety" ?

Does this mean that 4.4 is not runtime safe and the change needs to be backported?

What is the criticality?
Is it related to the HW design?

Thanks.

Dazza0, 2024-03-05:
@KonssnoK Previously, FreeRTOS would simply use `malloc`. Thus, if users enabled `CONFIG_SPIRAM_USE_MALLOC`, it is possible that `malloc()` will return memory in SPIRAM.  Thus, it was possible that `xQueueCreate()` could allocate a queue in SPIRAM. Given that SPIRAM is inaccessible when the cache is disabled (e.g., during an SPI flash write), it was up to the users to ensure that FreeRTOS objects that were allocated in SPIRAM were not access while the cache was disabled. For example, if 
 `xQueueCreate()` allocates a queue in SPIRAM, users must ensure that the queue was not accessed while another task was executing an SPI flash write operation.

However, we encountered numerous users who ran into a `Cache disabled but cached memory region accessed` error as they were not aware of the restrictions of allocating memory from SPIRAM. Thus, the decision was made to make FreeRTOS to allocate all dynamic memory from internal memory by default (which is always accessible). FreeRTOS objects can still be allocated from SPIRAM (or other types of memory as well). However, this must now be done explicitly by the user (via the `...WithCaps()` API) such that they are aware of exactly which type of memory the object is placed in, and the associated restrictions.

KonssnoK, 2024-03-05:
thanks @Dazza0 .

Am I correct saying that the only way to have such a scenario of `Cache disabled but cached memory region accessed` would be when the second core tries to do something on spi ram during spi flash operations?

Because from what I remember on flash operations you always have
- scheduler is suspended on running CPU
- interrupts are disabled

Dazza0, 2024-03-05:
@KonssnoK 

> Am I correct saying that the only way to have such a scenario of Cache disabled but cached memory region accessed would be when the second core tries to do something on spi ram during spi flash operations?

Yes, that's correct

> Because from what I remember on flash operations you always have at least
>
>    scheduler is suspended on running CPU
>    interrupts are disabled

Interrupts created with the `ESP_INTR_FLAG_IRAM` will still run during flash operations. If that interrupt handler then calls `xQueueSendFromISR()` to an SPI RAM queue, we get the `Cache disabled but cached memory region accessed` error.

KonssnoK, 2024-03-05:
our main doubt is about some espressif closed source code (mesh) that we use extensively.

This means that to recover space we'll have to change the esp_adapter file to use `withCaps` but we have no visibility on what happens inside the mesh library.

@zhangyanjiaoesp could you make an evaluation of:
- how many `ESP_INTR_FLAG_IRAM` interrupts there are in the mesh library.
- if there are `xQueueSendFromISR` in the mesh library

If 1 is true (used) it means that we should not move mesh queues to spi ram (even if we currently do).

If 1 is false (not used) it means we can use the mesh library on the same core of spi flash operations independently from 2, which somehow invalidates https://github.com/espressif/esp-idf/issues/11023

If 1 is false and 2 false (not used) it means we can also move mesh to the second core.

@Dazza0 please confirm.

Thanks

zhangyanjiaoesp, 2024-03-05:
> @zhangyanjiaoesp could you make an evaluation of:
> 
> * how many `ESP_INTR_FLAG_IRAM` interrupts there are in the mesh library.
> * if there are `xQueueSendFromISR` in the mesh library

@KonssnoK 

- there are no `ESP_INTR_FLAG_IRAM` interrupts in the mesh library
- there are no `xQueueSendFromISR` in the mesh library

KonssnoK, 2024-03-05:
@zhangyanjiaoesp thanks! 
so it means we should be able to change esp_adapter, add `withCaps `and move it to core1 without any issue.

romocitto88, 2024-03-12:
Hi, I'm facing the same issue: my project in v5.1 is consuming internal RAM more than same code in v4.4 and always the compilation return "out of region..."
Which are the steps to solve this issue?

zhangyanjiaoesp, 2024-03-13:
@romocitto88 Please take a look at https://github.com/espressif/esp-idf/issues/13285#issuecomment-1975891933 and https://github.com/espressif/esp-idf/issues/13285#issuecomment-1978103060

romocitto88, 2024-03-13:
I read the comments and I changed the functions indicated with "...WithCaps() with MALLOC_CAP_SPIRAM" but nothing is changed. I missed something?

Dazza0, 2024-03-13:
@romocitto88 Could you post a copy of the error log you are encountering?

romocitto88, 2024-03-13:
@Dazza0 
Here the error:

"section '.iram0.text' will not fit in region 'iram_0_seg"
"IRAM0 segment data does not fit" 
"region 'iram_0_seg' overflowed by 2528 bytes"

Dazza0, 2024-03-13:
@romocitto88 You are running our of static memory (IRAM to be specific). It's completely unrelated to this issue.

Please refer to this guide regarding [reducing binary size](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/size.html#minimizing-binary-size).

romocitto88, 2024-03-14:
@Dazza0 
The same project with ESP-IDF version 4.4.2 everything compiles without problems, and with ESP-IDF version 5.1.2 runs out the memory. I've attached a file merge of the two section compiled with the different IDF-VERSION
[Merge.pdf](https://github.com/espressif/esp-idf/files/14600223/Merge.pdf)

Dazza0, 2024-03-14:
@romocitto88 Yes, more functions have been placed into IRAM between v4.4.2 and v5.1.2. Again, please refer to the [reducing binary size](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/size.html#minimizing-binary-size) guide to determine what is using up your IRAM and what can be moved to flash to reduce IRAM usage.

closed 2024-03-15 by espressif-bot