espressif/esp-idf #13289: Cannot use i2c1 and i2c2 at the same time

puppy112358, 2024-02-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I want to use i2c to communicate with other devices, 

### What is the actual behavior?

 I find I cannot install two i2c drivers at the same time .  It means that when I config and install both i2c0 and i2c1 ,  my freertos task cannot run .
My project also use TCP, WIFI functions . When I delete one of the i2c ,the task can run normally . 

### Steps to reproduce.

1. Step
[image]

2. Step
[image]

3. Step  build the i2c task ,which is to read the device
...

### More Information.

 I cannot install uart1 and uart2 , too. I think this may be for the same reason .   However, I can use uart1 and uart2 at the same time when I use idf5.0 . When I changed idf to master ,the strange problem appears .

mythbuster5, 2024-03-01:
Currently I didn't reproduce this locally. Could you please 1. Provide a minimum reproduce code? 2 What about disable TCP/WIFI, only run with I2C1 and I2C2? Thanks!

mythbuster5, 2025-01-21:
It cannot be reproduced on our side, is there any more update? If no, I will close this issue, thanks!

mythbuster5, 2025-02-06:
Long time no respond, I will close it now. Feel free to reopen it if you have any more updates, thanks!

closed 2025-02-06 by mythbuster5