espressif/esp-idf #13291: nvs_flash_init_partition triggers idle task watchdog for large number of entries

edeadlk, 2024-02-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-166-ge0944287f3

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

My system includes a partition expected to hold up to 9999 nvs blob entries resembling a 16 byte struct.
Partition is set up in the partition table as follows:
log,        data,   nvs,        ,           0x184000

The more entries are written to said partition, the longer the nvs_flash_init_partition call at system start takes. This is annoying but acceptable. Expected behaviour is that I can fill as many entries as I like and the system runs without error.

### What is the actual behavior?

Actual behaviour is that the system runs into the Idle Task watchdog during nvs_flash_init_partition call after a certain amount of entries were filled. Apparently there is no vTaskDelay() or similar called during nvs_flash_init_partition, resulting in this bug.

### Steps to reproduce.

1. create large enough partition
2. call nvs_flash_init_partition for this partition at every system boot
3. fill partition with 9999 blob entries of 16 byte each 
4. reboot 

### Debug Logs.

```plain
I (26) boot: ESP-IDF v5.2-166-ge0944287f3 2nd stage bootloader
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (29) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : QIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 factory          factory app      00 00 00010000 00200000
[... 247 lines cut ...]
.flash.rodata 0x3c090120 0x5e078 RW A
.ext_ram.bss 0x3c0f0000 0x0 RW  
.iram0.data 0x40392c00 0x0 RW  
.iram0.bss 0x40392c00 0x0 RW  
.dram0.heap_start 0x3fcb5820 0x0 RW  
.coredump.tasks.data 0x3fcb93e8 0x154 RW 
.coredump.tasks.data 0x3fcb8dd0 0x600 RW 
.coredump.tasks.data 0x3fcb9c64 0x154 RW 
.coredump.tasks.data 0x3fcb9a40 0x200 RW 
.coredump.tasks.data 0x3fcba4e0 0x154 RW 
.coredump.tasks.data 0x3fcba200 0x2c0 RW 
.coredump.tasks.data 0x3fcb6afc 0x154 RW 
.coredump.tasks.data 0x3fcb6800 0x2e0 RW 
.coredump.tasks.data 0x3fcb81f4 0x154 RW 
.coredump.tasks.data 0x3fcb7ef0 0x2e0 RW 
.coredump.tasks.data 0x3fcbae5c 0x154 RW 
.coredump.tasks.data 0x3fcbab60 0x2e0 RW 
.coredump.tasks.data 0x3fcb7278 0x154 RW 
.coredump.tasks.data 0x3fcb6f80 0x2e0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump checksum='2e0be8d6'
```

adokitkat, 2024-03-04:
Hello, thanks for the report. Could you please provide an example project in a .zip file here which replicates this issue so I can test the fix easier? Thank you.

edeadlk, 2024-03-04:
Hello,
I created an example based on the nvs blob example. I had to play with ram usage and actual number of items in NVS to reach the same outcome I experience in my project.
The provided sample should result in the reported behaviour at a fill level of around 14k entries. If not please increase the maximum entries or decrease the watchdog timeout time.

[nvs_rw_blob.zip](https://github.com/espressif/esp-idf/files/14487864/nvs_rw_blob.zip)

Thank you for your support!

edeadlk, 2024-03-04:
wrong button 😅

adokitkat, 2024-03-04:
Thank you very much. For the time being, maybe you could try to separate you one big NVS to couple of smaller ones and add `vTaskDelay(0);` between their initialization? (Plus including `"freertos/FreeRTOS.h"` and `"freertos/task.h"`)

rspolyak, 2024-04-22:
Can you try release version of IDF instead of "v5.2-166-ge0944287f3"?

Also I have exactly the same issue with IDF v4.1 or so. Once I have upgraded to v5.1.2, nvs_flash_init_partition() takes up to 2 seconds even when it is almost full (the size is 512kB)

adokitkat, 2024-04-23:
Hi @edeadlk, could you also send me your NVS log partition dumped from the ESP (you can use `esptool.py` for that)? I cannot reproduce your issue just with the example you sent me.

adokitkat, 2024-04-23:
However, you can to apply this patch file to the latest `release/v5.2` branch of ESP-IDF to see if it does help you.

[0001-fix-nvs-Fix-triggering-WDT-when-using-large-NVS-part.patch](https://github.com/espressif/esp-idf/files/15079534/0001-fix-nvs-Fix-triggering-WDT-when-using-large-NVS-part.patch)

adokitkat, 2024-05-24:
Closing this as the user seemingly split their NVS into smaller partitions and this helped with shortening the initialization time.

closed 2024-05-24 by adokitkat