espressif/esp-idf #13295: ERROR at build time : Multiple candidate to satisfy project requirements

juan4java, 2024-02-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

**ESP IDF Version**
5.1.1
**Operating System**
Windows 10

**Condition**
I have clone the project, and cant build it, because the "led strip" component is ambiguos for the builder.
**Expected condition**
Build and run it in my ESP32C6 to use with a zigbee lamp.

**Detail about the error**
Every time who I tried to build the project the console otput is the same:
 
```
 ERROR: Cannot process component requirements.  Multiple candidates to satisfy project requirements:

 requirement: "led_strip" candidates: "espressif__led_strip, led_strip"
```

I checked the _**idf_component.yml**_ file , and his content is:
```
  ## IDF Component Manager Manifest File dependencies:
  espressif/esp-zboss-lib: "1.0.9"
  espressif/esp-zigbee-lib: "1.0.9"
  espressif/led_strip: "~2.0.0"
   ## Required IDF version
   idf:
     version: ">=5.0.0"
```

and the CMakeList.txt file  at project level is:

```
  # The following lines of boilerplate have to be in your project's CMakeLists
  # in this exact order for cmake to work correctly
  cmake_minimum_required(VERSION 3.16)
  include($ENV{IDF_PATH}/tools/cmake/project.cmake)
  project(light_bulb)
```

I had see other similars errors online like [this](https://github.com/espressif/idf-component-manager/issues/25) but mi conditions aparently are well defined.

Also visited the espressif Build pages [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html), and still cant find where is the problem.

Last, I tried another option via deleted the "led strip" dependency, the compiler runs fine (it don't said nothing about the ambiguos dependency) , but finally it throws a lot of errors because the component don't exist and also because it is the expected behaivor.

### What is the actual behavior?

Every time who I tried to build the project the console otput is the same:
 
```
 ERROR: Cannot process component requirements.  Multiple candidates to satisfy project requirements:

 requirement: "led_strip" candidates: "espressif__led_strip, led_strip"
```

### Steps to reproduce.

Clone the esp idf repo, open using VSCode the examples/zigbee/light_sample/HA_on_off_light  set the COM port, ESP32C6 clean and build.

### Build or installation Logs.

```plain
[1/3] espressif/esp-zboss-lib (1.0.9)
[2/3] espressif/esp-zigbee-lib (1.0.9)
[3/3] idf (5.1.1)
CMake Error at C:/Users/juanc/esp-51/esp-idf/tools/cmake/component.cmake:250 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "led_strip" candidates: "led_strip, espressif__led_strip"

Call Stack (most recent call first):
  C:/Users/juanc/esp-51/esp-idf/tools/cmake/build.cmake:574 (__component_get_requirements)
  C:/Users/juanc/esp-51/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:6 (project)
```

### More Information.

I clone another repo, similar to esp-idf, and the problem is the same.

Dazza0, 2024-03-01:
@juan4java Seems like there are two different versions of the same component somehow being build. Could you check...

- the contents of the `manage_components` directory of your example? Does it contain both `espressif__led_strip` and `led_strip`?
- If you do a full clean of the project (`idf.py fullclean`) and then rebuild the project, does the issue get resolved?

igrr, 2024-03-01:
And another thing to try: you can remove the `led_strip` from the manifest, then run `idf.py reconfigure`. There should be a list of `Component paths:` printed. There you can find where the _other_ version of led_strip is located.

juan4java, 2024-03-03:
Hi, thanks for both reply
@Dazza0 
* my  manage_components only contains espressif__led_strip and not another
*  I make clean before all build, also I have clonned another similar repo in a new path and the result was the same.

@igrr 
* I will try to run this option and see the results to reply you.

Updating :
How this was taking too many time, only for test the code, I decided put the files of espressif__led_strip dependency (only the .c and .h files ) in the main folder, and I declared it in the CMake file. Yes!, I know this can be a ugly solution, but this worked fine and finally I did can build it with out any error to test the code. I will try to fix in the correct way, becuase this must to run it using the dependencies.

fromeijn, 2024-09-10:
I also have seen this issue in our pipeline (ESP-IDF 5.2), where we build quite a few apps. Our solutions seems to be to remove any files related to the component manager, like `managed_components` (not only in the current build folder) before starting a new app build. My hunch is that the component manager finds multiple instances and doesn't know which one too use.

kumekay, 2025-02-28:
@fromeijn @juan4java Do you have reliable ways to reproduce the issue? The issue doesn't happen with clean installation of different versions of ESP-IDF (5.1, 5.2, 5.5-dev) and the recent version of the idf-component-manager (2.1.2).