espressif/esp-idf #13299: arduino esp-idf5.0+，当使用wifi功能连接到路由器后，espnow，发送数据没问题但是收不到返回数据，也接收不到其他板发送的espnow数据。

Mitking, 2024-03-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

arduino esp-idf5.0，当使用wifi功能连接到路由器后，espnow，发送数据没问题但是收不到返回数据，也接收不到其他板发送的espnow数据。
之前的4.4版本，连接wifi后，espnow接收和发送都有问题，升级到5.0后，发送没问题，接收不到espnow数据

Mitking, 2024-03-02:
麻烦技术测试看看，模块：esp32-Wrover，谢谢了！

zhangyanjiaoesp, 2024-03-04:
@Mitking  please try to call `esp_now_set_wake_window(65535)` after `esp_now_init()` for all of your esp_now receive devices to see if it could solve your problem.

Mitking, 2024-03-11:
> @Mitking 请尝试致电`esp_now_set_wake_window(65535)`您`esp_now_init()`所有的 esp_now 接收设备，看看是否可以解决您的问题。

试了这个方法，还是原来的效果，不好用

zhangyanjiaoesp, 2024-03-13:
@Mitking I will test it locally and reply you ASAP.

Mitking, 2024-03-15:
arduino开发板管理地址 https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json 下载的开发板包：“by Espressif Systerms 版本 3.0.0-alpha3...”

zhangyanjiaoesp, 2024-03-15:
你可以提供一下测试的demo吗？

zhangyanjiaoesp, 2024-03-18:
@Mitking 我在 Arduino IDE 上， 基于 3.0.0-alpha3 版本测试，当两个 ESP-NOW 设备都未连接到wifi 的时候，可以正常收发；当两个设备都连接到 wifi 的时候，也可以正常收发。你在测试的时候是两个设备都连接到 wifi 吗？请提供一下你的测试 demo。

Mitking, 2024-04-01:
> @Mitking 我在 Arduino IDE 上， 基于 3.0.0-alpha3 版本测试，当两个 ESP-NOW 设备都未连接到wifi 的时候，可以正常收发；当两个设备都连接到 wifi 的时候，也可以正常收发。当其中只有一个设备连接到wifi 的时候，请注意两个设备是不是在同一个信道上。

好的，非常感谢！回头我试试。就是说连接到wifi的时候，esp32的wifi信道会自动变为已连接wifi的信道，这样的话，esp32连接到路由器wifi后，怎么获取当前的信道值？

zhangyanjiaoesp, 2024-04-01:
@Mitking 

调用 `Serial.println(WiFi.channel());` 可以打印当前所在的channel，调用 `esp_wifi_set_channel(channel,second);` 可以设置 channel。

zhangyanjiaoesp, 2024-06-03:
@Mitking 问题解决了吗？

closed 2024-07-17 by Sherry616