espressif/esp-idf #13305: using httpd_req_to_sockfd in http_handler cause Load access fault crash

kerms, 2024-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

ESP32-C3 rev 0.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom (luatos 9.9 esp32c3)

### Power Supply used.

USB

### What is the expected behavior?

`httpd_req_to_sockfd(req->handle)` returns the context fd in the http_handler.

### What is the actual behavior?

in `idf/examples/protocols/http_server/simple/`
```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
Stack dump detected
MEPC    : 0x42018d86  RA      : 0x42008e3e  SP      : 0x3fca95a0  GP      : 0x3fc91800  
0x42018d86: httpd_req_to_sockfd at /data/program/idf/esp-idf5/components/esp_http_server/src/httpd_txrx.c:614
0x42008e3e: hello_get_handler at /tmp/simple/main/main.c:175 (discriminator 1)

```

### Steps to reproduce.

1. copy `idf/examples/protocols/http_server/simple/`
2. set target to esp32-c3
3. add `ESP_LOGI(TAG, "echo fd: %d", httpd_req_to_sockfd(req->handle));` in any http_handler
```
/* An HTTP POST handler */
static esp_err_t echo_post_handler(httpd_req_t *req)
{
    char buf[100];
    int ret, remaining = req->content_len;

	ESP_LOGI(TAG, "echo fd: %d", httpd_req_to_sockfd(req->handle));

    while (remaining > 0) {
        /* Read the data for the request */
        if ((ret = httpd_req_recv(req, buf,
                        MIN(remaining, sizeof(buf)))) <= 0) {
            if (ret == HTTPD_SOCK_ERR_TIMEOUT) {
                /* Retry receiving if timeout occurred */
                continue;
            }
            return ESP_FAIL;
        }

        /* Send back the same data */
        httpd_resp_send_chunk(req, buf, ret);
        remaining -= ret;

        /* Log data received */
        ESP_LOGI(TAG, "=========== RECEIVED DATA ==========");
        ESP_LOGI(TAG, "%.*s", ret, buf);
        ESP_LOGI(TAG, "====================================");
    }

    // End response
    httpd_resp_send_chunk(req, NULL, 0);
    return ESP_OK;
}
```
4. run idf.py flash monitor

### Debug Logs.

```plain
I (5391) example: Starting server on port: '80'
I (5401) example: Registering URI handlers
I (20311) wifi:<ba-add>idx:1 (ifx:0, e8:91:20:fa:a7:f1), tid:6, ssn:0, winSize:64
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
Stack dump detected
MEPC    : 0x42018d86  RA      : 0x42008d34  SP      : 0x3fca9630  GP      : 0x3fc91800  
0x42018d86: httpd_req_to_sockfd at /home/zhuang/data/program/idf/esp-idf5/components/esp_http_server/src/httpd_txrx.c:614
0x42008d34: echo_post_handler at /tmp/zhuang/simple/main/main.c:267 (discriminator 1)

TP      : 0x3fc875c4  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x00000000  
S0/FP   : 0x0004d171  S1      : 0x00000039  A0      : 0x3fca78ec  A1      : 0x3fca7968  
A2      : 0x00000004  A3      : 0x0000006f  A4      : 0x00000004  A5      : 0x00000000  
A6      : 0xffffffff  A7      : 0x00000038  S2      : 0x3fca78ec  S3      : 0x3fca7960  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
0x4005890e: memset in ROM

T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Stack memory:
3fca9630: 0x0000007b 0x3fca9730 0x3fca96e8 0x42017464 0x0000000a 0x3fca9730 0x0000003f 0x420765e4
3fca9650: 0x3fca9718 0x3fca7c06 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca7bee 0x00000033
3fca9670: 0x00000000 0x00000000 0x3fca7b84 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9690: 0x00000004 0x3fca78ec 0x00000001 0x42018e6e 0x00000000 0x00000000 0x00000000 0x3fca7960
3fca96b0: 0x3fca78ec 0x3fca7960 0x3fca78ec 0x420191b8 0x3fca78ec 0x00000000 0x00000000 0x00000000
3fca96d0: 0x3fca78ec 0x3fca7960 0x00000000 0x420179ce 0x00000000 0x00000000 0x00000000 0x420170f6
3fca96f0: 0x00000000 0x420174c2 0x42017068 0x420175d4 0x42017450 0x420173c0 0x00000000 0x00000000
3fca9710: 0x3fca7960 0x00000005 0x00000000 0x00000000 0x00000000 0x00000001 0x00000038 0x000000b7
3fca9730: 0x00150600 0x00000000 0x00000000 0x00000000 0x00010001 0x1f030000 0x3fca96e8 0x00000000
3fca9750: 0x3fca7b84 0x3fca82fc 0x3fca78ec 0x42017a9c 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x3fca97cc
3fca9770: 0x3fca877c 0x3fca82fc 0x3fca78ec 0x4201838a 0x00000000 0x3fca97cc 0x3fca82fc 0x42016294
3fca9790: 0x00000000 0x42016248 0x3fca82fc 0x42017e76 0x00000000 0x00000000 0x00000000 0x00000000
3fca97b0: 0x00000000 0x00000000 0x3fca78ec 0x42016c50 0x00000000 0x00000000 0x00000000 0x3fca97d8
3fca97d0: 0x3fca78ec 0x00000039 0x00000000 0x02000000 0x00000000 0x00000000 0x3fca78ec 0x42016d38
3fca97f0: 0x00000000 0x00000000 0x00000000 0x40386abc 0x00000000 0x00000000 0x00000000 0x00000000
3fca9810: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca9830: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x3fca9510 0x00000000 0x3fc93d28
3fca9850: 0x3fc93d28 0x3fca9844 0x3fc93d20 0x00000014 0x3fca99c4 0x3fca99c4 0x3fca9844 0x00000000
3fca9870: 0x00000005 0x3fca8840 0x70747468 0x00000064 0x00000000 0x00000000 0x3fca9830 0x00000005
3fca9890: 0x00000000 0x3fca82a4 0x4200043e 0x00000000 0x3fc9848c 0x3fc984f4 0x3fc9855c 0x00000000
3fca98b0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x42004d92 0x00000000 0x00000000
3fca98d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca98f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9910: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9930: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9950: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca9970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x00000000 0x3fca99b0
3fca99b0: 0xffffffff 0x3fca99b0 0x3fca99b0 0x00000000 0x3fca99c4 0xffffffff 0x3fca99c4 0x3fca99c4
3fca99d0: 0x00000000 0x00000001 0x00000000 0x3f00ffff 0x00000000 0x0000000c 0x3fca9990 0x00000011
3fca99f0: 0x3fc9be60 0x3fc97cb4 0x00000000 0x3fca99e8 0x00000012 0x3fca9a18 0x00000001 0x42008e22
3fca9a10: 0x3c0a50bc 0x0000000c 0x6c65682f 0x3f006f6c 0x3fca9a10 0x00000010 0x3fca9a3c 0x00000003
```

### More Information.

`httpd_req_to_sockfd` works normally in ws_handler

ESP-YJM, 2024-03-06:
@kerms I think your usage of API `httpd_req_to_sockfd` is mistake. You need use `req` as input parameter instead of `req->handle`. You can use `ESP_LOGI(TAG, "echo fd: %d", httpd_req_to_sockfd(req));` to print the number of socket fd.

kerms, 2024-03-06:
😅 Thanks you.. You are definitely correct. In my ws_handler side, I used `httpd_req_to_sockfd(req);`. I mixed up with "req-handle" used every where. And was so confident that I had use the correct argument as there are **no compilation warning** with `req->handle` (don't know why).

closed 2024-03-06 by kerms

ESP-YJM, 2024-03-06:
> 😅 Thanks you.. You are definitely correct. In my ws_handler side, I used `httpd_req_to_sockfd(req);`. I mixed up with "req-handle" used every where. And was so confident that I had use the correct argument as there are **no compilation warning** with `req->handle` (don't know why).

Yes, the type of  `req->handle` is void *, so the gcc may not regard it as warning or error. Glad to hear that the problem has been found and fixed.